UltimateHackingKeyboard/firmware #1191: UHK 80: Locking and unlocking a layer key with double-tap results in an error and the LED display shows an empty or corrupted keymap name

aminomancer, 2025-03-24:
If I double-tap the Mouse key, it locks my UHK 80 to the Mouse layer, as expected.
Then if I tap the Mouse key again to unlock, it correctly unlocks the Mouse layer.
But at that point, I see a ⚠ icon in the LED display, and the name "QWERTY for PC" is replaced with gibberish.
If the UHK Agent is open when this happens, it shows a banner at the bottom with this log:

```
5636: Invalid abbreviation length: 67
5626: 1 73 1 75 1 74 1 78 1 77
5636: 3 80 1 1 82 3 79 1 1 80
5646: 1 81 1 79 1 40 1 53 1 30
5656: 1 31 1 32 1 33 1 34 1 35
5666: 1 43 1 20 1 26 1 8 1 21
```

It stays like that until I write to the device storage, or change the keymap. Once the keymap has been changed at least once, the bug appears to stop and no longer happens until the device is restarted.

Nothing else seems to happen while the bug is in effect, i.e. the keymap name in the display is bugged and the error banner is showing in UHK Agent. All functionality seems to be normal, other than the keymap name display.

This happened when setting up a brand new UHK 80 out of the box. The first thing I did was flash the latest firmware 13.0.0, so I only noticed the bug after that. I didn't try locking a layer key before flashing the firmware, so I don't know whether it also happens on older firmware.

I can provide my complete logs if that helps.

kareltucek, 2025-03-27:
I can't. 

Would you provide a UserConfig?

mondalaci, 2025-03-27:
Now I can't reproduce it either! I'm on 13.0.0.

@aminomancer Please share your configuration.

aminomancer, 2025-03-27:
> Now I can't reproduce it either! I'm on 13.0.0.
> 
> [@aminomancer](https://github.com/aminomancer?rgh-link-date=2025-03-27T21%3A39%3A59.000Z) Please share your configuration.

How? What do you mean by configuration? `Export device configuration`? Or you mean one of the .bin files in the user-configs folder accessed from `Settings` > `Open configuration folder`?

kareltucek, 2025-03-28:
Export device configuration; You will get a UserConfiguration.json file.

aminomancer, 2025-03-28:
Oh, I was on the default config. [UserConfiguration.json](https://github.com/user-attachments/files/19497573/UserConfiguration.json)

It continued happening after I customized it a bit: [UHK 80 Config.json](https://github.com/user-attachments/files/19497617/UHK.80.Config.json)

The bug stops occurring once you switch keymaps, so maybe that's why you're not seeing it. I can't repro anymore either, I figured it would probably start happening again after a reboot, but since there's no power button I don't know. Maybe resetting the device with the recessed switch on the back would do it.

Now I have it back on the default config, since I was thinking of RMAing it over some other issues.

pcooke9, 2025-03-28:
> I figured it would probably start happening again after a reboot, but since there's no power button I don't know. Maybe resetting the device with the recessed switch on the back would do it.

You can use the reset buttons on the back, or the easier way is to use the `reboot` macro command.😉

mondalaci, 2025-03-28:
I flashed the provided configuration, couldn't reproduce the issue, then restored my configuration, I could reproduce the issue once, and then I couldn't reproduce it again, so it's very sporadic for me.

mondalaci, 2025-03-28:
It may matter that I use my UHK 80 via USB with the bridge cable between the halves. How about you, guys?

kareltucek, 2025-04-02:
I still can't reproduce it. 

What is the exact configuration? Is the bridge cable connected? Are batteries present and connected? Is right side powered via usb? Is left?

pcooke9, 2025-04-02:
I can't reproduce it using both USBs, bridge cable, & batteries.

mondalaci, 2025-04-02:
@aminomancer We need the exact steps to reproduce the issue, starting with loading your configuration, and then, possibly resetting the UHK 80.

aminomancer, 2025-04-03:
I just pulled it out of the box, disconnected the two halves, connected them with the cable, then connected the key cluster module and trackpoint module, then plugged in the USB-C cable (without the USB-A adapter), opened the UHK Agent, and flashed the firmware. I didn't change the configuration until after the bug had already happened.

mondalaci, 2025-04-03:
Thanks for elaborating! I'm on 13.0.1 and can't reproduce it, not even sporadically. This is a hard one.