UltimateHackingKeyboard/firmware #1193: Can't allocate variables after v13

eric-fulfil, 2025-03-24:
Hi,

I upgraded Agent / FW today to v13 and suddenly with no changes my $onInit is crashing no matter how few or many variables I set in it, getting the error "Can't allocate more than 32 variables".  Here's my original $onInit that worked in v12:

```
setVar mRepeat 0
setVar mSpeed 10
setVar mPlaying 0
setVar recMode 0

setVar mediaPlaying 1
setVar soundMuted 0
setVar micMuted 0

setVar trafficKey 64
setVar trafficColor 0

setVar dial -1
setVar mediaVolume 50
setVar blBrightness 100

setVar i 0
setVar j 0
setVar r 0
setVar f 0
setVar g 0
setVar b 0
setVar t 0
setVar mx 0
setVar my 0
setVar addr 0

setVar exKey1 64
setVar exKey2 64
setVar exKey3 64
setVar dKey 1

set leds.brightness $blBrightness
set autoRepeatDelay 100
set autoRepeatRate 0
set keystrokeDelay $mSpeed

setVar thisDialKey 1
setVar semi 0
setVar wiz -1

//0- TS
//101- SH
//201-300 FF
setVar wizType -1

//0 LNX
//1 Droid
//2 Mac
setVar osMode 0
```

Unplugging the board or removing lots of variables, seemed to have no effect, even just a handful of setVars in this macro seems to break, so I'm guessing either a buncha internal variables were added or setVar in $onInit is somehow broken?

eric-fulfil, 2025-03-24:
This is on a UHK v2

eric-fulfil, 2025-03-24:
[image]

kareltucek, 2025-03-27:
Could you provide your UserConfig please? (Of course stripped of sensitive information) 

I haven't managed to reproduce the problem.

kareltucek, 2025-03-27:
Actually I think I see the problem.

mondalaci, 2025-03-28:
Does the https://github.com/UltimateHackingKeyboard/firmware/actions/runs/14104670931 fix this issue?

eric-fulfil, 2025-04-02:
Rolling forward backward in FW/Agent is kinda painful but I'm happy to try latest release if you think this is fixed in latest Agent?  I vaguely remember noobing out trying to flash just the FW alone.

If not I can anonymize my user config and send.

mondalaci, 2025-04-02:
This build should fix the issue, so it's worth a try.