UltimateHackingKeyboard/firmware #1195: Variable limit reached — impacted by BLE connections

tunney, 2025-03-25:
Happily ticking along with my config for a while. Added another BLE device and saved the connection. Results in the below.

Do the saved connections each into the useable variables?

[image]

kareltucek, 2025-03-25:
No. This is a new bug. Will look into it.

mhantsch, 2025-03-26:
> No matter how many variables I remove it finds more to complain about. #borked

Does the error go away if you reboot the keyboard?

kareltucek, 2025-03-27:
Do we know how to reproduce it?

mhantsch, 2025-03-27:
Maybe unreleated, but I remember even on firmware 11.x I got issues with weird errors from macros if (background, forked) macros were still running while saving a new configuration. It would mess up the running macros to a state where the firmware would throw errors, and not be able to reset them other than completely rebooting the keyboard (UHK60: unplugging/replugging).

tunney, 2025-03-27:
> > No matter how many variables I remove it finds more to complain about. #borked
> 
> Does the error go away if you reboot the keyboard?

No, it did not going away with a reset. Took a firmware reinstall to do that.

tunney, 2025-03-27:
> Do we know how to reproduce it?
> 
> (I suspect it is triggered by saving UserConfig, and is independent of pairing.)

I am on client site today so cannot try to repoduce but will try tonight tomorrow.

Yes triggered by saving UserConfig, but I only ever saw it after saving a pairing in the agent software (when that prompted me to do so, which was not always consistent)

kareltucek, 2025-03-27:
@tunney nevermind, we have a fix already ;-).

tunney, 2025-03-27:
Intrigued as to the cause (and solution!)

kareltucek, 2025-03-27:
The cause is simple. The engine isn't wasting the RAM by keeping a copy of the string. The variable records only reference the names stored in the config segment. 

The engine isn't stopped and restarted on UserConfig writes as it should be, and so the variables continue on living... so when a new config is written, the old references suddenly point to different data, and when the variables are referenced, string comparisons fail, resulting in allocation of new variables.

kareltucek, 2025-03-27:
No. That technical debt remains.

mhantsch, 2025-03-27:
> No. That technical debt remains.

**$onConfigReload**:

```
delayUntil 3000
reboot
```

mhantsch, 2025-03-27:
My convoluted brain: 

If I forget to stop running macros before saving a new config, I often end up having to reboot the keyboard. Might as well do that all the time when I save a new config.

Darn, there is no **$onConfigReload** that allows me to automate it!

mhantsch, 2025-03-27:
```
error: reboot loop detected
```

kareltucek, 2025-03-27:
Ah, I see. Signaling via a variable is indeed not possible. Sorry 🤦

pcooke9, 2025-03-28:
> ```
> error: reboot loop detected
> ```

Woah, Déjà vu...😆

mondalaci, 2025-03-28:
Does the https://github.com/UltimateHackingKeyboard/firmware/actions/runs/14104670931 fix this issue?