UltimateHackingKeyboard/firmware #1199: Issue to pair the UHK 80 to Dongle after upgrading to 13.0.1 firmware

PYalandin, 2025-03-27:
UHK can not communicate to dongle anymore after i flashed both halves and dongle to  latest  firmware of 13.0.1.
OLED screen shows that it is connected to dongle but in fact it is not. No responses from any key press or mouse. Also OLED screen shows connection to dongle as active even after i physically disconnect it from the laptop.
Downgraded to version 13.0.0 and connection to Dongle worked nicely.

kareltucek, 2025-03-27:
Please retest with build from https://forum.ultimatehackingkeyboard.com/t/dev-build-testing-request-bluetooth-ble-hid-to-ble-hid-switchovers/2231/2

PYalandin, 2025-03-27:
It did the trick.
Shall i stay on this nightly dev build, or revert back to stable 13.0.0?

kareltucek, 2025-03-27:
Stay ;-).

closed 2025-03-27 by kareltucek