UltimateHackingKeyboard/firmware #1200: Previous and Next HID on uhk-firmware-13.0.1-bluetooth_glitches-v1.tar.gz

tunney, 2025-03-27:
Switching hosts to named hosts works perfectly on uhk-firmware-13.0.1-bluetooth_glitches-v1.tar.gz

[image]

That is the three keys on the left work exactly as expected.

The previous and next only seem to cycle between wired host and last BLE connection.

So if I connect to  MBP BLE  and then swtich to iPhone (BLE) the  named switches that works. Then if I try to do next what i see in sequence on the OLED is

MBP BLE (i hit next)
Macbook Pro -> iPhone
iPhone

Next and previous will then show Macbook Pro and iPhone

Would expect to be able to cycle through all three, or at least the last two connected.

(Not sure if meant to be raising issues against dev builds...)

kareltucek, 2025-03-28:
> Next and previous will then show Macbook Pro and iPhone

That is the expected behavior for the moment. We may allowultiple ble hids in the future, or write something that manages to keep track of available but not connected bles in the future, but we are not there yet

mondalaci, 2025-03-28:
I agree that the desired behavior is the correct one that users expect, me surely included.

mhantsch, 2025-03-28:
I would expect to cycle forward and backward through all the connections that Agent shows as Connections. That includes all the BLE connections that I have established previously.