UltimateHackingKeyboard/firmware #1206: Zephyr logs specification

kareltucek, 2025-04-03:
UHK will expose two new usb variables: 
```
        UsbVariable_ShellEnabled,
        UsbVariable_ShellBuffer,
```

- when UsbVariable_ShellEnabled is false by default. When true, firmware gathers zephyr shell logs in a new buffer.
- UsbVariable_ShellEnabled is set by Agent when Agent/user decide they want to see logs.
- when Agent is disconnected, UsbVariable_ShellEnabled is automatically set to false again. 
- when there is a new log, the 8 bit is set in the byte 2 of get device state command.
- when UsbVariable_ShellBuffer is read, firmware consumes a part of the buffer and returns it to Agent as a null delimited string. (Null delimiter is present only if sending less than maximum payload length.) 

-------------------------------
@ert78gb @mondalaci does this sound sane?

@ert78gb would you like some additional metadata in the UsbVariable_ShellBuffer response?

mondalaci, 2025-04-03:
I wouldn't set UsbVariable_ShellEnabled to false upon Agent disconnection because that way, we can't debug boot issues.

May the Zephyr shell output contain null? If so, the first byte should be the length of the payload.

kareltucek, 2025-04-03:
Logging all the time is expensive. If we need to turn it on during boot, we can do that via wormhole or a custom firmware, but in practice it sounds as a very biche usecase.

Its strings, so it shouldn't.

mondalaci, 2025-04-03:
Users will seldom enable Zephyr logging, so I would keep it enabled to make it as generally useful as possible.

kareltucek, 2025-04-03:
The specification above doesn't require users to enable it. 

The idea is that when there is a sink for the logs then logging is enabled.

mondalaci, 2025-04-03:
There are various use cases when logging would be useful even if a sink is temporarily unavailable, such as boot time logging or when connecting various USB devices one by one and there is a related log entry to be logged. I disagree with the artificial requirement of a sink, as I think it limits the usefulness of this feature.

We could make Zephyr logging explicitly visible via the OLED and Agent, so it'd be easy to notice when enabled.

kareltucek, 2025-04-03:
What is the usecase you are arguing about now?

Just about not disabling logging when Agent is disconnected? 

That may mean that logging will often continue when not needing, slowing down the keyboard, causing mouse jitters and other issues, but sure, that is not that high price.

kareltucek, 2025-04-04:
Ah. Fair enough then!

PR #1142 (2025-04-24): Implement basic battery manager.
Solves the most pressing issues of #974.

Closes #1206.
Closes #1210.
Closes #1011.
Closes #1214.

What it does:
- implement controls that allow easily setting battery limits
- implement a state machine that enables and disables the charger as needed
- implement a charging mode for stationary use that targets 3.8V, and its macro controls `set battery.chargeLimit { full | optimizeHealth }`
- eliminate the step of battery indicators between powered and unpowered states and various brightness states.
- blink battery-low icon on the oled when battery is low
- imple [... cut]
closed 2025-04-24 by mondalaci