UltimateHackingKeyboard/firmware #1207: [BLE/Dongle] No longer can use BLE or Dongle connections on 13.0.0/13.0.1

ndekross, 2025-04-03:
After installing v13, basically all wireless connections stopped working. The Bluetooth and radio symbols show up on the LED panel and in the UHK agent as connections. The dongle shows up as paired. However when I disconnect the USB cable the keyboard doesn't switch over to wireless and says "Disconnected". I deleted the BLE connection in the attempt to re-pair, but the keyboard no longer shows up on the available bluetooth device list. Other Bluetooth devices paired to my PC work just fine, so it's not an issue with my Bluetooth module. Resetting, defaulting the user configuration and re-flashing the firmware to the same version don't work. I didn't try to downgrade since this version is overall more stable.

kareltucek, 2025-04-03:
Please flash the firmware from https://forum.ultimatehackingkeyboard.com/t/dev-build-testing-request-bluetooth-ble-hid-to-ble-hid-switchovers/2231/8 and try to pair again.

Sorry for the inconvenience!

ndekross, 2025-04-03:
Thanks, I'll check later tonight

ndekross, 2025-04-04:
@kareltucek This fixed both the dongle and BLE for me. However, the trackpoint doesn't work on BLE.

closed 2025-04-04 by mondalaci