UltimateHackingKeyboard/firmware #1210: Insead of "charge me", blink a status icon

mondalaci, 2025-04-05:
On the main OLED screen, blink a battery empty icon in the top status bar, such as https://fontawesome.com/icons/battery-slash?f=classic&s=solid

kareltucek, 2025-04-05:
Like where? On the left?

mondalaci, 2025-04-05:
I'd place it leftward of battery percentages because it's battery-related, and this way, the icon wouldn't displace percentage positions.

mhantsch, 2025-04-05:
How about just blinking the battery percentage that is low?

mondalaci, 2025-04-05:
Simply blinking the battery percentages may work, but I'm afraid it's rather inexplicit; Some may interpret it as the battery draining, but don't realize it should be charged.

I agree that battery-slash is probably not the best choice. I think I prefer battery-low because it clearly shows that there's very little charge left in the battery. battery-empty may be interpreted as a battery whose inside is not visible.

pcooke9, 2025-04-05:
You could make it super obvious by alternating the icon and the word "LOW".

mondalaci, 2025-04-05:
I'd try the icon first. The available status bar space is very limited, and we may add additional icons, so we better make the best use of it.

kareltucek, 2025-04-05:
You could make it space efficient by alternating the percentage text with the word "LOW" 😁.

mondalaci, 2025-04-05:
For sure, but as a rule of thumb, I wouldn't cover useful information, such as battery percentage, even for short intervals. I think it's cleaner to leave the information always visible and reveal the blinking icon when relevant.

PR #1142 (2025-04-24): Implement basic battery manager.
Solves the most pressing issues of #974.

Closes #1206.
Closes #1210.
Closes #1011.
Closes #1214.

What it does:
- implement controls that allow easily setting battery limits
- implement a state machine that enables and disables the charger as needed
- implement a charging mode for stationary use that targets 3.8V, and its macro controls `set battery.chargeLimit { full | optimizeHealth }`
- eliminate the step of battery indicators between powered and unpowered states and various brightness states.
- blink battery-low icon on the oled when battery is low
- imple [... cut]
closed 2025-04-24 by mondalaci