UltimateHackingKeyboard/firmware #1214: Weird button release behavior after updating to 13.0.2

Gert-dev, 2025-04-16:
I've recently updated from an older firmware version, 11.2.0, to 13.0.2, and am now experiencing various oddities around keys triggering actions they shouldn't on my UHKv1, usually in combination with the mouse layer.

I can reproduce as follows:

1. **Hold down** the key to trigger the mouse layer (left special key under the spacebar).
2. Move the mouse to the left or right by **holding** the default keys on the mouse layer (e.g. what is visually on the keyboard the 'L' key to move right).
3. Now **press (but don't release)** the Meta/Windows key.
4. Observe how the Meta key press immediately registers as a key up/release as well, despite the key not being released yet. This is not the case when the mouse layer is not active.

In GNOME this is especially noticeable since this results in the overview immediately being opened or closed, despite me not having released the super key. In GNOME a way to move windows is also to hold down the super key and then use the mouse to click on the window and move it afterwards, which this sudden change has broken or at least made unreliable.

**If you switch around 2 and 3 and press the Meta key first, this behaviour doesn't happen and everything works as intended.**

Another oddity is that 3 often results in the mouse movement speeding up drastically, as if the Meta key is some sort of accelerator all of a sudden. This may be a different issue though (perhaps #1186).

kareltucek, 2025-04-16:
I can't reproduce that. Please share your UserConfig (remove anything sensitive first though).

kareltucek, 2025-04-23:
I see. I can reproduce it. Strangely, only with uhk60 (v2 here), not with uhk80. 

Steps to reproduce:

- use above config
- map "a" in mouse layer
- press left case button ("space")
- press j
- tap a
- observe "aa" in a text editor

kareltucek, 2025-04-24:
Fixed on the battery management branch! Feel free to flash the firmware from the branch CI actions.

Gert-dev, 2025-04-24:
That seems to fix it for me, thanks!

PR #1142 (2025-04-24): Implement basic battery manager.
Solves the most pressing issues of #974.

Closes #1206.
Closes #1210.
Closes #1011.
Closes #1214.

What it does:
- implement controls that allow easily setting battery limits
- implement a state machine that enables and disables the charger as needed
- implement a charging mode for stationary use that targets 3.8V, and its macro controls `set battery.chargeLimit { full | optimizeHealth }`
- eliminate the step of battery indicators between powered and unpowered states and various brightness states.
- blink battery-low icon on the oled when battery is low
- imple [... cut]
closed 2025-04-24 by mondalaci