UltimateHackingKeyboard/firmware #1215: USB drops (crashes?) on UHKv1 since update from 11.2.0 to 13.0.2

Gert-dev, 2025-04-18:
As mentioned in another ticket, I've recently updated my UHKv1 from an older firmware version, 11.2.0, to 13.0.2. The update happened a couple of days ago, and with this new version **I'm receiving occasional drops where the keyboard fully disconnects by itself from USB** suddenly, usually when I'm in the middle of typing something. It happens about once or twice a day so far, with the old firmware version this didn't happen.

The LCD display of the keyboard also drops at that point, so **it looks like the keyboard 'crashes' or gets its power fully cut**, if that makes sense. The only fix at that point is to replug the USB connector in my dock to get it to show up again in Linux.

It's unlikely to be related to power saving since I have it disabled for the UHKv1 because it results in momentary stalls when it needs to come back, and it's also happening in the middle of typing.

## Logs
My logs don't show much, unfortunately, so would be happy to try and gather other logs or debug further. On a drop they pretty much just show the drop itself:

```
April 18 09:57:37 hostname kernel: usb 1-1.1.3.3: USB disconnect, device number 12
```

On a replug I also don't really see anything out of the ordinary (most recent log is shown first):

```
April 18 09:57:39 hostname mtp-probe[8433]: bus: 1, device: 15 was not an MTP device
April 18 09:57:39 hostname mtp-probe[8433]: checking bus 1, device 15: "/sys/devices/pci0000:00/0000:00:08.1/0000:06:00.3/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3.3"
April 18 09:57:39 hostname kernel: hid-generic 0003:37A8:0001.0014: hiddev98,hidraw10: USB HID v1.10 Device [Ultimate Gadget Laboratories UHK 60 v1] on usb-0000:06:00.3-1.1.3.3/input4
April 18 09:57:39 hostname kernel: hid-generic 0003:37A8:0001.0013: input,hidraw9: USB HID v1.10 Device [Ultimate Gadget Laboratories UHK 60 v1] on usb-0000:06:00.3-1.1.3.3/input3
April 18 09:57:39 hostname kernel: input: Ultimate Gadget Laboratories UHK 60 v1 as /devices/pci0000:00/0000:00:08.1/0000:06:00.3/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3.3/1-1.1.3.3:1.3/0003:37A8:0001.0013/input/input36
April 18 09:57:39 hostname kernel: hid-generic 0003:37A8:0001.0012: input,hidraw8: USB HID v1.10 Device [Ultimate Gadget Laboratories UHK 60 v1] on usb-0000:06:00.3-1.1.3.3/input2
April 18 09:57:39 hostname kernel: input: Ultimate Gadget Laboratories UHK 60 v1 as /devices/pci0000:00/0000:00:08.1/0000:06:00.3/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3.3/1-1.1.3.3:1.2/0003:37A8:0001.0012/input/input35
April 18 09:57:39 hostname kernel: hid-generic 0003:37A8:0001.0011: input,hidraw7: USB HID v1.10 Mouse [Ultimate Gadget Laboratories UHK 60 v1] on usb-0000:06:00.3-1.1.3.3/input1
April 18 09:57:39 hostname kernel: input: Ultimate Gadget Laboratories UHK 60 v1 as /devices/pci0000:00/0000:00:08.1/0000:06:00.3/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3.3/1-1.1.3.3:1.1/0003:37A8:0001.0011/input/input34
April 18 09:57:39 hostname kernel: hid-generic 0003:37A8:0001.0010: input,hidraw3: USB HID v1.10 Keyboard [Ultimate Gadget Laboratories UHK 60 v1] on usb-0000:06:00.3-1.1.3.3/input0
April 18 09:57:39 hostname kernel: input: Ultimate Gadget Laboratories UHK 60 v1 as /devices/pci0000:00/0000:00:08.1/0000:06:00.3/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3.3/1-1.1.3.3:1.0/0003:37A8:0001.0010/input/input33
April 18 09:57:39 hostname kernel: usb 1-1.1.3.3: SerialNumber: 0148338290
April 18 09:57:39 hostname kernel: usb 1-1.1.3.3: Manufacturer: Ultimate Gadget Laboratories
April 18 09:57:39 hostname kernel: usb 1-1.1.3.3: Product: UHK 60 v1
April 18 09:57:39 hostname kernel: usb 1-1.1.3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
April 18 09:57:39 hostname kernel: usb 1-1.1.3.3: New USB device found, idVendor=37a8, idProduct=0001, bcdDevice= 0.01
April 18 09:57:39 hostname kernel: usb 1-1.1.3.3: new full-speed USB device number 15 using xhci_hcd
```

kareltucek, 2025-04-19:
Ah, yes, thanks for the head ups!

So far, I have not made much progress:

I have been trying to capture these crashes with a gdb, but it looks like the crashes go away when a debugger is connected 😒. 

I guess I will need to modify the firmware to capture traces via the wormhole.

kareltucek, 2025-04-24:
Please:
- flash this firmware: 

[uhk-firmware-13.0.2-crashunt_v1.2.tar.gz](https://github.com/user-attachments/files/19894970/uhk-firmware-13.0.2-crashunt_v1.2.tar.gz)

- after the firmware crashes, please open Agent. Yellow bar should pop up, containing some trace of what happened before the crash. Copy them here.
- bear with me; we will need multiple iterations to nail this down

Gert-dev, 2025-04-25:
I just experienced the drop with the crash firmware active, though I'm not sure if it's the same drop. Here is what happened:

1. Keyboard dropped and wasn't responding, LEDs went out.
2. I opened the agent, no yellow bar popped up.
3. Checked the journal, but USB didn't drop this time.
4. Checked UHK agent again, and it still recognized the keyboard this time.
5. Tried to remap a key and saved.
6. The LEDs came back on, along with the keyboard indicating the 'mouse' layer was active.
7. I tried to toggle the mouse layer as it seemed stuck, no effect.
8. I tried to move the mouse with one of the keys from my mouse layer, seeing as the mouse layer was active according to the keyboard, no effect.
9. Tried to type other keys, none had any effect.
10. Replugged the keboard, everything worked again.
11. Checked UHK agent again, but also no yellow bar.

Gert-dev, 2025-04-27:
I had this a couple more times, and now realise that the USB drop log in the OP was actually me pulling the plug 🤦. Turns out the keyboard doesn't connect from USB at all, as per the previous post, it just 'freezes' every time.

kareltucek, 2025-04-28:
Well, let's sum up what we know:

- The keyboard freezes. 
- All leds go out (?).
- Neither keyboard USB interface nor mouse USB interface works (?).
- Pressing layer keys does nothing (all leds of both halves are out after all).
- Powercycling is necessary to bring it alive again.

Is the above correct?

Questions:
- When it is freezed, does it disconnect from USB?
- While it is freezed, can you connect to it with Agent? (I.e., does your config load ok?)
- Is this happening also when your UHK is connected directly to the PC?

(This looks different from the issues that I had experienced and assumed this topic was about: in my case the keyboard just restarts itself. I don't have to powercycle it.)

kareltucek, 2025-04-28:
Hmm. I see. I will get back to you with another testing firmware.

Gert-dev, 2025-04-28:
Thanks, I will install it and do my best to help debug it 😉.

> I have yet to test this and will get back to you. Currently it's linked to an USB-C dock. In any case before the firmware update this was never a problem, but I'll test it for comparison anyway.

I can now confirm it also happens on a host USB port.

mhantsch, 2025-04-29:
> Here we go: uhk-firmware-13.0.2-crashunt_v1.3.tar.gz
> Instructions: when freeze happens, open Agent, and wait 10 seconds. You should see a log now:

Does this only apply to the UHK60 or also valid for UHK80? I've had a few crashes/hangs of my UHK80 since installing 13.0.2, and I can't really pinpoint under what circumstances they happen. The UHK80 stops responding to keypresses/touchpad. The display still shows connected, but whatever I do (unplug USB, unplug bridge cable, replug etc.) nothing happens. Rebooting both halves (via reset button) recovers it to a working state again.

kareltucek, 2025-04-29:
Should work for uhk80 too!

Gert-dev, 2025-04-29:
Thanks! Is there perhaps an unreleased version of the agent I need to use to flash it? I see this:

[image]

kareltucek, 2025-04-29:
Please try running the Agent with `--ignore-firmware-checksums --disable-agent-update-protection
` arguments.

Gert-dev, 2025-04-30:
I get the yellow banner now after it happens. Here are the logs:

```
Looks like the firmware freezed. If that is the case, please report bellow trace to the devs:
Last EV: 2097666
Trace:
)(l6u1u6u7u8)(l6u1u6u7u8)(l6u1u6u7u8)(l6u1u6u7u8)(l6u1u6u7u8)(l6
u1u6u7u8)(l6u1u6u7u8)(l6u1u6u7u8)(l6u1u6u7u8)(l6u1u2v3v4u3u6u7u8
)(l6u1u6u7u8)(l6u1u6u7u8)(l6u1u6u7u8)(l6u1u6u7u8)(l6u1u6u7u8)(l6
```

kareltucek, 2025-05-01:
Perfect! Will process it in a day or two ;-).

kareltucek, 2025-05-01:
Actually, I think there should be 4 lines of the trace. I might be wrong, but are you sure this is the entire thing?

Gert-dev, 2025-05-01:
Oops, sorry, I had the same output multiple times and stripped too much, it seems:

```
Looks like the firmware freezed. If that is the case, please report bellow trace to the devs:
Last EV: 2097666
Trace:
)(l6u1u6u7u8)(l6u1u6u7u8)(l6u1u6u7u8)(l6u1u6u7u8)(l6u1u6u7u8)(l6
u1u6u7u8)(l6u1u6u7u8)(l6u1u6u7u8)(l6u1u6u7u8)(l6u1u2v3v4u3u6u7u8
)(l6u1u6u7u8)(l6u1u6u7u8)(l6u1u6u7u8)(l6u1u6u7u8)(l6u1u6u7u8)(l6
u1u6u7u8)(l6u1u6u7u8)(l6u1u2u3u6u7u8)(l6u1u2v3u7u8)(l7)(l6u1u2v3
```

kareltucek, 2025-05-02:
[uhk-firmware-13.0.2-crashunt_v1.4.tar.gz](https://github.com/user-attachments/files/20014780/uhk-firmware-13.0.2-crashunt_v1.4.tar.gz)

Here is another version. Same instructions please :-)

Gert-dev, 2025-05-03:
Here is the new freeze trace 🙂 :

```
Looks like the firmware freezed. If that is the case, please report bellow trace to the devs:
Last EV: 2097664
Trace:
u1u6u7u8)(l6u1u6u7u8)(l6u1u6u7u8)(l6u1u6u7u8)(l6u1u2v3w0w1x1x2x3
x4w2w3w4w5w2w3w4w5w6ve3v4u3u6u7u8)(l6u1u6u7u8)(l6u1u6u7u8)(l6u1u
6u7u8)(l6u1u6u7u8)(l6u1u6u7u8)(l6u1u6u7u8)(l6u1u6u7u8)(l6u1u2u3u
6u7u8)(l6u1u2v3w0w1w2w3w4w5w6ve3u7u8)(l7)(l6u1u2v3w0w1x1x2x3x4w2
```

kareltucek, 2025-05-05:
It looks like it is coming from remote wakeup code (the thing that allows waking up a suspended computer by tapping a key). Let's confirm this with this firmware: 

[uhk-firmware-13.0.2-crashunt_v1.5.tar.gz](https://github.com/user-attachments/files/20034041/uhk-firmware-13.0.2-crashunt_v1.5.tar.gz)

This will further log any sleep modes, e.g., caused by suspend and wake of your computer. You can ignore these or remove them after every such event with a `clearStatus` macro.

Gert-dev, 2025-05-06:
Is there anything else I should send as logs this time, or just the info in the yellow bar again? This time I get scrambled output in the yellow bar that displays:

```
␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄ ␞�␟�␄
```

In the UHK logs I see this as well:

```

20:58:08.961 › [DeviceOperation] status buffer segment:  �� �� �� �� �� �� �� �� �� �� 
20:58:08.962 › [SmartMacroDocRendererService] downloadDocumentation
20:58:08.962 › [SmartMacroService] start downloading firmware documentation {
  firmwareGitRepo: 'UltimateHackingKeyboard/firmware',
  firmwareGitTag: 'crashunt_v1.6'
}
20:58:08.962 › [SmartMacroService] firmware documentation downloaded earlier
20:58:08.963 › [DeviceOperation] status buffer segment: �� �� �� �� �� �� �� �� �� �� �
20:58:08.963 › [SmartMacroService] reference manual downloaded earlier
20:58:08.965 › [DeviceOperation] status buffer segment: � �� �� �� �� �� �� �� �� �� ��
20:58:08.967 › [DeviceOperation] status buffer segment:  �� �� �� �� �� �� �� �� �� �� 
20:58:08.968 › [SmartMacroDocRendererService] dispatch action: [smart-macro-doc] download documentation success
20:58:08.969 › [DeviceOperation] status buffer segment: �� �� �� �� �� �� �� �� �� �� �
20:58:08.968 › [SmartMacroDocRendererService] dispatch action: [smart-macro-doc] reference manual changed
20:58:08.971 › [DeviceOperation] status buffer segment: � �� �� �� �� �� �� �� �� �� ��
20:58:08.973 › [DeviceOperation] status buffer segment:  �� �� �� �� �� �� �� �� �� �� 
20:58:08.975 › [DeviceOperation] status buffer segment: �� �� �� �� �� �� �� �� �� �� �
20:58:08.977 › [DeviceOperation] status buffer segment: � �� �� �� �� �� �� �� �� �� ��
20:58:08.979 › [DeviceOperation] status buffer segment:  �� �� �� �� �� �� �� �� �� �� 
20:58:08.981 › [DeviceOperation] status buffer segment: �� �� �� �� �� �� �� �� �� �� �
20:58:08.983 › [DeviceOperation] status buffer segment: � �� �� �� �� �� �� �� �� �� ��
20:58:08.985 › [DeviceOperation] status buffer segment:  �� �� �� �� �� �� �� �� �� �� 
20:58:08.987 › [DeviceOperation] status buffer segment: �� �� �� �� �� �� �� �� �� �� �
20:58:08.989 › [DeviceOperation] status buffer segment: � �� �� �� �� �� �� �� �� �� ��
20:58:08.991 › [DeviceOperation] status buffer segment:  �� �� �� �� �� �� �� �� �� �� 
20:58:08.993 › [DeviceOperation] status buffer segment: �� �� �� �� �� �� �� �� �� �� �
20:58:08.995 › [DeviceOperation] status buffer segment: � �� �� �� �� �� �� �� �� �� ��
20:58:08.997 › [DeviceOperation] status buffer segment:  �� �� �� �� �� �� �� �� �� �� 
20:58:08.999 › [DeviceOperation] status buffer segment: �� �� �� �� �� �� �� �� �� �� �
20:58:09.001 › [DeviceOperation] status buffer segment: � �� �� �� �� �� �� �� �� �� ��
20:58:09.001 › [DeviceRendererService] dispatch action [device] Status buffer changed
```

kareltucek, 2025-05-06:
I see. That's a bug. Will look into it tomorrow.

kareltucek, 2025-05-07:
I hope it should be fixed here: 

[uhk-firmware-14.0.0-crashunt_v1.7.tar.gz](https://github.com/user-attachments/files/20080869/uhk-firmware-14.0.0-crashunt_v1.7.tar.gz)

If you see scrambled output in the yellow bar again, let me know!

Gert-dev, 2025-05-07:
Thanks for the follow-up. I will test for a few days and try to get back to you with whether the issue still occured or not 🙂.

Gert-dev, 2025-05-10:
I had this issue at least once per day, have now been testing for a few days, and haven't had the issue any more with the posted test firmware 🎉.

Feel free to close this, I'll reopen if it does end up coming back but at this point I doubt it 🙂.

Gert-dev, 2025-05-11:
I'm a bit confused by the 😕 emoji response, is it being fixed not good? Or was it not supposed to be fixed yet? Or maybe the wrong emoji was used by accident 😅?

kareltucek, 2025-05-11:
(Only scrambled output was supposed to be fixed )

Gert-dev, 2025-05-11:
Ah, I see. It could indeed be that the bug now just appears less often.

In any case I will keep my eyes open and keep the test firmware installed for the time being in the meantime.

Gert-dev, 2025-05-13:
In case it helps, I tested with the recently released 14.0.0 and it still happens there, so it must be some commit in between.

kareltucek, 2025-05-14:
I don't see there any changes that may be relevant. 

Anyways, let's close this and reopen if it reappears :-(.

Gert-dev, 2025-05-16:
You were right - it's not entirely gone, but it seems to slightly have changed in behaviour. Today I had a full keyboard freeze twice shortly after one another, but this time opening the UHK agent 'fixed' it, i.e. the keyboard became responsive again after opening the agent to see what the problem was.

The agent did show the yellow banner at that point, but there isn't much of note, except:

```
    Skip firmware update
    , and use Agent without saving new configurations.
    Entered Awake power mode, because: received device set configuration event
    Entered LightSleep power mode, because: received device suspend event
    Entered Awake power mode, because: received device resume event
    Entered LightSleep power mode, because: received device suspend event
    Entered Awake power mode, because: received device resume event
    Entered LightSleep power mode, because: received device suspend event
    Entered Awake power mode, because: received device resume event
    Entered LightSleep power mode, because: received device suspend event
    Entered Awake power mode, because: received device resume event
```

Though it would be a slow process, we could also bisect the commits that changed with me installing firmware of each bisection, but if the issue still occurs (and this is the same issue), it's probably moot to suggest that 🙂.

roco72, 2025-05-17:
It might not be related, but this morning when I woke up, my UHK60 (v1) had crashed too. I noticed the text on the LED display was alternating between the tag  I configured between the (brackets), and "OUT", even before I woke up my Mac. Typing was still possible. After waking up my mac with a key stroke and logging in, I saw the yellow banner in the agent (which was already open). Unfortunately I did not make a dump of the yellow banner, but after a usb disconnect and reconnect, it disappeared.
I run firmware 14.0.0 on both halves and the key cluster.

kareltucek, 2025-05-20:
Yes! 🥳

Galathius, 2025-05-30:
He-hey @kareltucek, got this right after flashing [uhk-firmware-14.0.3-debug-v1.8.tar.gz](https://github.com/user-attachments/files/20520683/uhk-firmware-14.0.3-debug-v1.8.tar.gz) you attached above. 

<details>
  <summary>Crash report</summary>
  
```
Printing trace buffer because: Looks like your uhk60 crashed.
Last EV: 0
Trace:
1>(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l
3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l
3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l
3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)Rebooting...######

Printing trace buffer because: Looks like your uhk60 crashed.
Last EV: 0
Trace:
3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l
3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l
3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)1>(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)Rebooting...######

Printing trace buffer because: Looks like your uhk60 crashed.
Last EV: 0
Trace:
3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l
3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l
3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l
3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)(l3)Rebooting...######
```
  
</details>

kareltucek, 2025-05-31:
Those are just reboots.

A bug is that they trigger crash log.

kareltucek, 2025-06-02:
Here is a new firmware with reworked and re-tested crash logging. Please flash it and let's see if we can get any useful log out of it:

[uhk-firmware-14.0.3-debug-v1.11.tar.gz](https://github.com/user-attachments/files/20553033/uhk-firmware-14.0.3-debug-v1.11.tar.gz)

Gert-dev, 2025-06-21:
To give an update from my end: I've been running this debug build about since it was posted and can't reproduce my issue any more.

I've now tried to install 14.2.0 stable to see if it still occurs there since [previously](https://github.com/UltimateHackingKeyboard/firmware/issues/1215#issuecomment-2886905822) I could still reproduce on stable.

Gert-dev, 2025-07-09:
I just encountered a freeze again, but they seem to occur less often than the previous ones and it might be a different one. The keyboard just stays on as if it's working correctly, but no keyboard input is being processed. Checking the Linux system logs shows nothing, so the USB device doesn't seem to have actually dropped.

Opening the UHK agent also shows nothing but magically fixes the problem somehow - after it opens input on the keyboard starts working again. I also tried first unplugging the bridge cable and replugging it, causing the left half to power cycle, but that also didn't fix it.

kareltucek, 2025-07-09:
@Gert-dev so when these new freezes occur, does the keyboard continue to register key presses? E.g., if you press a layer switch, does the corresponding led light up?

Gert-dev, 2025-07-13:
@kareltucek It doesn't seem to - pressing the mouse layer switch for example does not make the LED light up.

Gert-dev, 2025-07-18:
I'm wondering if this new freeze is related to power saving. I had it disabled in the past, but it enabled itself a few days ago because I switched USB ports.

I am using `powertop` to enable power saving. Using the `powertop` command line utility you can turn on power saving for the UHK. In the background it just writes `auto` to e.g. `/sys/bus/usb/devices/1-2.4.3/power/control` (where `1-2.4.3` is the UHK USB device number).

It does not happen all the time with power saving enabled, but with the default `/sys/bus/usb/devices/1-2.4.3/power/autosuspend_delay_ms` set to `2000` (2 seconds) I've had the keyboard freeze about 5 times in short succession, each time 'resetting' it by opening UHK agent. It hasn't occurred yet as long as UHK agent remained open, but that could be coincidence.

I now have the autosuspend delay set much higher and the freezes occur noticeably less often.

kareltucek, 2025-07-22:
> Is the 'UHK 80' just a log thing?

Probably yes. The logs are not familiar, which means that they don't come from the keyboard.

> autosuspend_delay_ms set to 2000 (2 seconds) I've had the keyboard freeze about 5 times in short succession, each time 'resetting' it by opening UHK agent. It hasn't occurred yet as long as UHK agent remained open, but that could be coincidence.

Agent pings UHK every second to retrieve device state, which would explain why you see no freezes with 2 second autosuspend delay. 

This looks like a very likely cause of the issue, as UHK60 is not optimized for low power at all. 

https://claude.ai/chat/f32539e3-16fd-461b-81e8-13540a79a655

Iirc, when in suspend mode, UHK60 doesn't process events (or send the related usb hid reports) and can be awaken only by the relevant usb resume event, which in practice means that it should wake up when you tap a key on another keyboard, given that this is supposed to make powertop wake up all devices, or something like that.

Given this info, do you have any reason to believe that UHK implements the suspend/resume events incorrectly? 

If yes, then it may be worth to have Benedek check the implementation. 

If no, then I am frankly not sure what to do about it.

Gert-dev, 2025-07-22:
Good to know, thanks for the information.

> Given this info, do you have any reason to believe that UHK implements the suspend/resume events incorrectly?

It's hard to say, I suspect it to either be a bug on the UHK side or a kernel power saving bug of some sort. There is certainly a bug _somewhere_ 😅, given that usually power saving works very well on the UHK v1 and it properly goes in and out of suspend. It only seems to happen every so often where it enters suspend but then doesn't come back any more - if there was no bug I would either expect it to never work or to always work 😄 .

Also, it's worth mentioning that I've been using powertop for a very long time, including to suspend the UHK; back on firmware 11.2.0 these freezes never occurred, so there does seem to be some kind of regression here.

sotomski, 2025-10-25:
Hi, I'm having similar crashes a few times a day on stable firmware version 15.0.1:
```
Printing trace buffer because: Looks like your uhk60 crashed.
EV: 0
Tag: v15.0.1
=== K22P121M120SF7 Reset Cause Debug ===
RCM_SRS0:  0x00
RCM_SRS1:  0x04
RCM_SSRS0: 0x82
RCM_SSRS1: 0x04
PMC_LVDSC1: 0x10
PMC_LVDSC2: 0x00
PMC_REGSC:  0x04
SCB_HFSR:  0x00000000
SCB_CFSR:  0x00000000
SCB_BFAR:  0xE000ED38
SCB_MMFAR: 0xE000ED34

=== Reset Cause Analysis ===
Trace:
{(l3)}{(l3)}{(l3)}{(l3)}<i1>{(l3)}{(l3)}{(l3)}{(l3)}{(l3)}<i1>{(
l3)}<i4>{(l3)}{(l3)}{(l3)}{(l3)}{(l3)}{(l3)}{(l3)}{(l3)}{(l3)}<i
1>{(l3)}<i1>{(l3)}<i4>{(l3)}{(l3)}{(l3)}<i1>{(l3)}{(l3)}{(l3)}{(
l3)}{(l3)}<i1>{(l3)}{(l3)}{(l3)}<i4>{(l3)}{(l3)}{(l3)}{(l3)}{###
```

It does seem to happen during typing.

mondalaci, 2025-10-25:
Please try [firmware 15.2.0](https://github.com/UltimateHackingKeyboard/firmware/releases/tag/v15.2.0).

Gert-dev, 2025-10-25:
I think, at least in my case, that this can be closed. I'm indeed on 15.2.0 and haven't seen this happen in a while.