UltimateHackingKeyboard/firmware #1222: FIDO integration with keyboard?

mandrizzle, 2025-05-05:
Hello, I am curious if it would be possible to implement the FIDO standard into the keyboard itself to avoid needing to use a yubi key for passkeys. I was hoping we could support a hotkey to simulate a FIDO touch and essentially invoke the key.

kareltucek, 2025-05-06:
Sure. Just stick a USB hub onto your uhk and plug your ubi key there. 

----------

As for firmware level implementation...

While uhk has flash memory that could store the private keys, as well as sufficient sources of entropy to generate them, it isn't designed as a digital strongbox.

There are good reasons why keys are managed by simple single-purpose hardware devices created by security experts. Messing with it outside of that context would be an incredibly stupid thing to do security-wise.

mhantsch, 2025-05-06:
+1 for a dedicated TPM chip in keyboards 😏

mondalaci, 2025-05-07:
Closing this issue because we don't plan to add USB hubs or dedicated crypto hardware to UHKs. Feel free to continue the conversation regardless.

closed 2025-05-07 by mondalaci