UltimateHackingKeyboard/firmware #1224: USB won't connect when using UHK80 right half reset button

pcooke9, 2025-05-09:
When using the UHK80's reset button (right half), the PC's USB doesn't reconnect after the right half restarts. I have to either use the `reboot` macro command, or unplug the _left USB cable_, and then disconnect/reconnect the _right USB cable_ to resume. (Left USB _**must**_ be unplugged before reconnecting the right USB.🤷‍♂️) LED backlighting and OLED show normal layer changes, etc., just no output to the PC.

If the dongle is plugged in, the dongle will auto connect and work fine after the reset, but then if I try to switch from the dongle to the PC's USB, it won't switch.

I'm not sure when it started, but I checked with FW v13.0.2-0da386a (master branch), as well as the new release v14.0.0. I'm not sure if they're useful, but here's a couple of PuTTY logs via UART adapter using FW v14.0.0 and a _mostly_ default config (added `reboot` macro on Fn2 layer).

<details>
<summary>Log_01</summary>

**UHK80 FW v14.0.0 (Release version)
Both halves connected via USB & bridge cable
Dongle/Bluetooth not connected
User Config mostly default (**`reboot` **macro added to Fn2 layer)**

•press right half's reset button on the back (OLED displays "Disconnected")
•press random keys and move touchpad (no keystroke/mouse output)
•try switch host > 0: My PC (OLED displays "Disconnected -> My PC")
•press random keys and move touchpad (no keystroke/mouse output)
•`reboot` macro command (resumes normal connection & output)

```
=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2025.05.08 03:51:31 =~=~=~=~=~=~=~=~=~=~=~=

uhk80:right$ --- 492 messages dropped ---
Uart, unexpected byte: 0
uhk80:right$ Uart, unexpected byte: 0
uhk80:right$ Uart, unexpected byte: 0
uhk80:right$ --- 1 messages dropped ---
Uart, unexpected byte: 0
uhk80:right$ --- 1 messages dropped ---
Uart, unexpected byte: 0
uhk80:right$ Uart, unexpected byte: 0
uhk80:right$ Uart, unexpected byte: 0
uhk80:right$ Uart, unexpected byte: 0
uhk80:right$ UART_RX_DISABLED
uhk80:right$ [00:00:00.097,290] <inf> fs_nvs: 3 Sectors of 4096 bytes
uhk80:right$ [00:00:00.097,320] <inf> fs_nvs: alloc wra: 2, ea0
uhk80:right$ [00:00:00.097,351] <inf> fs_nvs: data wra: 2, 350
uhk80:right$ [00:00:00.097,930] <inf> bt_sdc_hci_driver: SoftDevice Controller build revision: 
                                            fe 2c f9 6a 7f 36 22 2e  a0 79 c0 40 be 2c 03 20 |.,.j.6". .y.@.,. 
                                            40 c2 f3 32                                      |@..2             
uhk80:right$ [00:00:00.103,485] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 254.63788 Build 573996906
uhk80:right$ [00:00:00.104,339] <inf> bt_hci_core: No ID address. App must call settings_load()
uhk80:right$ Settings: Found peer 'left' with address uhk80:right$ 36uhk80:right$ 7fuhk80:right$ e5uhk80:right$ 8duhk80:right$ a8uhk80:right$ ecuhk80:right$ 
uhk80:right$ [00:00:00.111,358] <inf> bt_hci_core: Identity: D5:13:F6:62:18:21 (random)
uhk80:right$ [00:00:00.111,419] <inf> bt_hci_core: HCI: version 6.0 (0x0e) revision 0x104e, manufacturer 0x0059
uhk80:right$ [00:00:00.111,450] <inf> bt_hci_core: LMP: version 6.0 (0x0e) subver 0x104e
uhk80:right$ Reading hardware config
uhk80:right$ Reading user config
uhk80:right$ Applying user config
uhk80:right$ Flashed User Config version: 9.99.0 (native version: 11.0.0., at v14.0.0 / d63c925ebac51c9f585a8c45c652b0bd)
uhk80:right$ Setting battery mode to 0
uhk80:right$ Clearing bonds
uhk80:right$  - Keeping a known bond.
uhk80:right$  - Keeping a known bond.
uhk80:right$ Resetting left right link! Unidirectional
uhk80:right$ User config applied
uhk80:right$ Scan module initialized
uhk80:right$ Bt: filling scan filters
uhk80:right$ Enabling UART
uhk80:right$ Filters matched: left (n/a, ec:a8:8d:e5:7f:36), connectable:1
uhk80:right$ Scan connecting: left (n/a, ec:a8:8d:e5:7f:36)
uhk80:right$ USB config changing to Hid_NoGamepad
uhk80:right$ Conn state: UartLeft(Ready)
uhk80:right$ Resetting left right link! Bidirectional
uhk80:right$ Resetting left right link! Unidirectional
uhk80:right$ [00:00:00.348,419] <wrn> bt_smp: Unexpected SMP code 0x0b
uhk80:right$ Batteries: 76% 0% (3893 0)
uhk80:right$ Conn state: NusServerLeft(Connected), Peer left
uhk80:right$ Conn state: NusServerLeft(Ready), Peer left
[... 260 lines cut ...]
uhk80:right$ uhk80:right$ uhk80:right$ Can't send report - selected connection is not ready!
uhk80:right$ Giving report to c2usb anyways!
uhk80:right$ Can't send report - selected connection is not ready!
uhk80:right$ Giving report to c2usb anyways!
uhk80:right$ uhk80:right$ [00:00:57.644,317] <inf> Battery:   - Resistances 175 24638 21936, brightness 118, correcting 3872 -> 3902

uhk80:right$ [00:00:57.644,409] <inf> Battery: ... Voltage corrected because of load 3872 (89) -> 3902 (92)
uhk80:right$ [00:00:57.644,439] <inf> Battery:     ... Averaged to 3902 (92)
uhk80:right$ Batteries: 76% 92% (3894 3902)
uhk80:right$ Batteries: 76% 92% (3890 3902)
uhk80:right$ uhk80:right$ [00:01:04.633,911] <wrn> udc_nrf: USB Address incorrect 0x01
uhk80:right$ Conn state: Host0(My PC, Ready)
uhk80:right$ Conn state: Host0(My PC, Ready), Active
uhk80:right$ uhk80:right$ [00:01:05.639,770] <inf> Battery: ... Voltage corrected because of charging 4002 -> 3918 (93) -> 3908 (92)
uhk80:right$ [00:01:05.639,862] <inf> Battery:     ... Averaged to 3905 (92)
uhk80:right$ Batteries: 76% 92% (3890 3905)
uhk80:right$ Batteries: 76% 92% (3891 3905)
uhk80:right$ uhk80:right$ 
```

</details>

___

I also wonder if it would be a good idea to add a `reboot` macro key to the default configs/keymaps for situations like this?

mondalaci, 2025-05-09:
Thanks for reporting!

@kareltucek Can you fix it, or is it c2usb-related?

kareltucek, 2025-05-09:
I don't know. C2usb is a suspect. 

Could you look into git and figure out which commits to test to find out?

benedekkupper, 2025-05-09:
What does pressing the reset button do exactly? Is that handled by the FW or performs a HW reset?

mondalaci, 2025-05-10:
@benedekkupper The reset button of the UHK 80 triggers a hardware reset.

mondalaci, 2025-05-10:
@pcooke9 Thanks for the idea, but I don't want to add a `reboot` macro call to the default configuration for two reasons. In this specific case, I would rather not work around bugs but fix them at their core. From a general standpoint, I avoid adding macro calls to the default configuration because macro commands can change, in which case, all configurations become obsolete and trigger validation warnings for all users.

mondalaci, 2025-05-10:
@benedekkupper Can you reproduce this issue? I can consistently reproduce it. Please make it your highest priority.

benedekkupper, 2025-05-10:
Does it change the behavior if you hold the reset button down for longer?

mondalaci, 2025-05-10:
For me, it doesn't.

pcooke9, 2025-05-11:
No change here either.

closed 2025-05-27 by mondalaci