UltimateHackingKeyboard/firmware #1226: Bricked left side after firmwareupdate to 14.0

lightningRalf, 2025-05-10:
I just updated my agent to 7.0 and the firmware update to 14.0.
Right side went smooth and does still work but the left have is bricked.
I tried every step as mentioned.

The unbricking with the reset button did not work.
After pressing multiple times. The fix button in the agent only appeared once:
[image]

I am on windows 11.
This only worked after separating the left from the right side of the keyboard.

lightningRalf, 2025-05-10:
[DeviceService] stop poll UHK Device
[DeviceService] stopped poll UHK Device
[DeviceService] UHK Device recovery starts: {"id":3,"asCliArg":"uhk80-left","firmwareUpgradeMethod":"mcuboot","logName":"UHK 80 left","name":"UHK 80","keyboard":[{"vid":14248,"pid":7}],"bootloader":[{"vid":14248,"pid":6}],"buspal":[],"reportId":4}
[UhkOperations] Start flashing device firmware with mcuboot
[UhkOperations] Start flashing UHK 80 left firmware with mcumgr
[UhkOperations] Reenumerate bootloader
[UhkHidDevice] Device communication closing.
[UhkHidDevice] Start reenumeration, mode: Bootloader, timeout: 5000ms
[UhkHidDevice] Available devices changed.
[UhkHidDevice] Removed: {"vendorId":"0x37A8","productId":"0x9","path":"\\\\?\\HID#VID_37A8&PID_0009&MI_00#8&14ae0df1&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\\KBD","serialNumber":"0019992159","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 80 Right USB","release":1,"interface":0,"usagePage":1,"usage":6}
[UhkHidDevice] Removed: {"vendorId":"0x37A8","productId":"0x9","path":"\\\\?\\HID#VID_37A8&PID_0009&MI_01#8&f292fd1&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}","serialNumber":"0019992159","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 80 Right USB","release":1,"interface":1,"usagePage":1,"usage":2}
[UhkHidDevice] Removed: {"vendorId":"0x37A8","productId":"0x9","path":"\\\\?\\HID#VID_37A8&PID_0009&MI_03#8&1b3ce14e&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}","serialNumber":"0019992159","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 80 Right USB","release":1,"interface":3,"usagePage":12,"usage":1}
[UhkHidDevice] Removed: {"vendorId":"0x37A8","productId":"0x9","path":"\\\\?\\HID#VID_37A8&PID_0009&MI_02#8&33006d93&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}","serialNumber":"0019992159","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 80 Right USB","release":1,"interface":2,"usagePage":65280,"usage":1}
[UhkHidDevice] Added: {"path":"COM7","manufacturer":"Microsoft","serialNumber":"7&1A80757F&0&0000","pnpId":"USB\\VID_37A8&PID_0006&MI_00\\7&1A80757F&0&0000","locationId":"0009.0000.0003.002.000.000.000.000.000","friendlyName":"Serielles USB-Gerät (COM7)","vendorId":"0x37A8","productId":"0x0006"}
[UhkHidDevice] Device communication closing.
[UhkOperations] Init SerialPeripheral: COM7
[UhkOperations] Read UHK 80 left firmware from file
[UhkOperations] Write memory with mcumgr
[DeviceService] updateFirmware error {"message":"Opening COM7: File not found","stack":"Error: Opening COM7: File not found"}
Opening COM7: File not found

lightningRalf, 2025-05-10:
[image]

A screenshot from the right side after connecting it again

mondalaci, 2025-05-10:
@ert78gb Please look into this.

ert78gb, 2025-05-10:
Please attach the agent logs. It contains timestamps. 

You fint the logs in the `%USERPROFILE%\AppData\Roaming\uhk agent\logs\` folder.

I am typing on mobile so maybe the path is not correct

tunney, 2025-05-12:
Seeing this as well, last week after many attempts to flash 14.0.0 it went on.
Today I tried to reflash to fix some pairing issues, and it bricked it as above. And nothing I do can get it to go on.
have rolled back to agent 6.0 and fw 13.0.2

lightningRalf, 2025-05-12:
> Please attach the agent logs. It contains timestamps.
> 
> You fint the logs in the `%USERPROFILE%\AppData\Roaming\uhk agent\logs\` folder.
> 
> I am typing on mobile so maybe the path is not correct

Thanks for looking into it:
[uhk-agent.log](https://github.com/user-attachments/files/20174017/uhk-agent.log)

nealwashere, 2025-05-12:
Had this same issue on Mac today. Retried on Windows, same thing. I did a final attempt back on my mac using a different cable and the left side finally flashed correctly. Not sure what I did to fix it, but both machines kept seeing the left side disconnect nonstop until my final attempt.

mondalaci, 2025-05-13:
Please test the https://github.com/UltimateHackingKeyboard/agent/actions/runs/14994081429 Agent build, guys. Does it resolve the issue?

tunney, 2025-05-13:
Afraid not Lazlo, still hangs indefinitely at the "connect left half".

tunney, 2025-05-13:
i might be a little slow today but I am doing the below but not seeing the "user config recovery screen"

open -a "UHK Agent 2" --args -- --disable-agent-update-protection

what am I doing wrong?

tunney, 2025-05-13:
no joy

[image]

ert78gb, 2025-05-13:
thank you for the quick response . Could you flash with the bundled 13.0.0 firmware ?

ert78gb, 2025-05-13:
Have your UHK battery? If yes then before you connect the left half to the computer please press the reset button on the bottom of the keyboard and connect the USB cable. If you could press the reset button and the cable connection within 5 seconds that would be great.

ert78gb, 2025-05-13:
And please attach the logs too. thx

tunney, 2025-05-13:
> Have your UHK battery? If yes then before you connect the left half to the computer please press the reset button on the bottom of the keyboard and connect the USB cable. If you could press the reset button and the cable connection within 5 seconds that would be great.

This sequence let me upgrade to 14.0.0

tunney, 2025-05-13:
To clarify:

Using Agent 6 with the command line args
And hitting reset as I connect the cable to the left hand side.

That is the sequence that worked.

ert78gb, 2025-05-13:
Could you repeat the repeat the firmware upgrade process a few (5) times? Downgrade to firmware 13 and after upgrade to firmware 14 both with Agent 6 and 7. With and without press the reset button.

mondalaci, 2025-05-15:
@tunney Thank you! We can't reproduce this issue, so we rely on your further feedback to determine whether it's an Agent or firmware issue.

tunney, 2025-05-16:
Agent 7.0 + 14.0.0 = fail (hangs on left side update and right side eventually displays firmware mismatches, and then in Agent the left side flashes on/off)
Agent 7.0 + 13.0.2 = success

Agent 6.0 + 14.0.0 = fail (hangs on left side update and right side eventually displays firmware mismatches, and then in Agent the left side flashes on/off)
Agent 6.0 + 13.0.2 = success

Any further variations to check?

mondalaci, 2025-05-17:
Please, somebody test and confirm that:
- Flashing [uhk-firmware-12.4.0-0e9c464a-good.tar.gz](https://github.com/user-attachments/files/20264931/uhk-firmware-12.4.0-0e9c464a-good.tar.gz) from Agent succeeds
- Flashing [uhk-firmware-12.4.0-23d23322-bad.tar.gz](https://github.com/user-attachments/files/20264934/uhk-firmware-12.4.0-23d23322-bad.tar.gz) from Agent fails

Relevant commit history:
```
bad         23d23322 Wed Mar 19 12:26:55 2025 +0100 kareltucek@centrum.cz Increase uhk60 noinit segment size to fit logs.
build fails 1b2c9194 Wed Mar 19 11:56:57 2025 +0100 kareltucek@centrum.cz Optimize key polling when no keys are pressed.
build fails 51b0a067 Wed Mar 19 10:49:08 2025 +0100 kareltucek@centrum.cz Undisable sleep mode.
build fails 20cc3cd7 Fri Mar 14 17:41:02 2025 +0100 kareltucek@centrum.cz Use zephyr logging for bluetooth messages.
build fails d1091dcc Thu Mar 13 15:53:23 2025 +0100 kareltucek@centrum.cz Implement our logging backend.
good        0e9c464a Thu Mar 13 16:04:28 2025 +0100 kareltucek@centrum.cz Protect the wormhole.
```

[Git diff](https://github.com/UltimateHackingKeyboard/firmware/compare/0e9c464a..23d23322) which contains the bug

mondalaci, 2025-05-17:
Can you test them with the UHK 60? I used the UHK 60 for testing.

I'll also test them on the UHK 80. Maybe the UHK 60 and UHK 80 differ in this respect.

spunicorn, 2025-05-17:
> Can you test them with the UHK 60? I used the UHK 60 for testing.
> 
> I'll also test them on the UHK 80. Maybe the UHK 60 and UHK 80 differ in this respect.

Dont have a uhk60 around unfortunately

lightningRalf, 2025-05-17:
ON UHK80: With Agent 7.0 and the good tar: 
I get this fireware success. But I also get this:
``` 
5: Invalid abbreviation length: 0
0: 0 0 0 0 0 0 0 0 0 0
10: 0 0 0 0 0 0 0 0 0 0
20: 0 0 0 0 0 0 0 0 0 0
30: 0 0 0 0 0 0 0 0 0 0
40: 0 0 0 0 0 0 0 0 0 0
5: Invalid abbreviation length: 0
0: 0 0 0 0 0 0 0 0 0 0
10: 0 0 0 0 0 0 0 0 0 0
20: 0 0 0 0 0 0 0 0 0 0
30: 0 0 0 0 0 0 0 0 0 0
40: 0 0 0 0 0 0 0 0 0 0
``` 
at the bottom with a yellow background.

 
And now I have the factory default and I have a greyed out 'restore last configuration' 

[image]
:(

Before I tried this, I had Agent 7.0 with uhk-firmware-13.0.2.tar.gz and my configuration working just fine.
Oh man. Why did I do this. :/

mondalaci, 2025-05-18:
@tunney Given that you have a UHK 60, would you please test the good and bad builds of https://github.com/UltimateHackingKeyboard/firmware/issues/1226#issuecomment-2888382605 on it?

Based on your feedback, guys, the offending commit differs for the UHK 60 and the UHK 80. Let's try to find the offending commit for the UHK 80.

```
$ git bisect start
$ git bisect bad v14.0.0
$ git bisect good 23d23322
Bisecting: 70 revisions left to test after this (roughly 6 steps)
[8ad3ba8b71ee04ff3b9099b280e9b46031544491] Implement setLedTxt for uhk80.
# the build fails with display.c:147: undefined reference to `NotificationScreen_NotifyFor'
# based on the output of `git log 23d23322..v14.0.0`, the previous commit is 82d68b9f
$ git checkout 82d68b9f
# this commit builds successfully
```

Please flash [uhk-firmware-13.0.2-82d68b9f.tar.gz](https://github.com/user-attachments/files/20275041/uhk-firmware-13.0.2-82d68b9f.tar.gz) on the UHK 80, guys, and let me know whether it exhibits the issue.

tunney, 2025-05-19:
@mondalaci Not sure if this will help you or not....

**UHK80**
_Agent 7.0_

13.0.2 to Good = Success
13.0.2 to Bad = Success

**UHK60**
_Agent 7.0_

11.2.0 to Good = Success
11.2.0 to Bad = Success

mondalaci, 2025-05-19:
Thanks, every such feedback is useful.

Please also test https://github.com/UltimateHackingKeyboard/firmware/issues/1226#issuecomment-2889218469

tunney, 2025-05-19:
@mondalaci flashing [uhk-firmware-13.0.2-82d68b9f.tar.gz](https://github.com/user-attachments/files/20275041/uhk-firmware-13.0.2-82d68b9f.tar.gz) on UHK80 with Agent 7.0 worked.

The right side didn't display the usual error message that I was getting with 14.0.0 when flashing so I had an idea from there that it would work.

mondalaci, 2025-05-19:
@tunney The OLED error message is not representative in this respect. When you try to flash 14.0.0 in Agent, does it consistently fail to flash the left half of your UHK 80?

mondalaci, 2025-05-19:
@tunney According to the OLED display, the right half is repeatedly trying to connect to the left half. How can you reproduce this?

tunney, 2025-05-19:
I flash the firmware with comes with agent 7, or any official 14.0.0 firmware. 
And that is the state the keyboard gets into even before I connect the USB cable to the left side.

Mac OS (latest OS)
Wired to host
Wired between the sides.

spunicorn, 2025-05-19:
When flashing 14.0 with Agent 7 on win 11 I get a similar outcome, but not quite the same: 

[image]

[uhk-agent.old.log](https://github.com/user-attachments/files/20281373/uhk-agent.old.log)

Still, none of the other firmware packages have ever failed like this, not with agent 7 or 6.

mondalaci, 2025-05-19:
```
$ git bisect start
$ git bisect bad v14.0.0
$ git bisect good 82d68b9f
Bisecting: 54 revisions left to test after this (roughly 6 steps)
[bb992ed691d0ffb2a34a5178f9e1f3988966cf2d] setLedTxt fixes.
```

Please test [uhk-firmware-13.0.2-bb992ed6.tar.gz](https://github.com/user-attachments/files/20284862/uhk-firmware-13.0.2-bb992ed6.tar.gz) on your UHK 80, guys.

tunney, 2025-05-19:
Agent 7.0 + MacOs
This firmware flashed fine @mondalaci

mondalaci, 2025-05-19:
```
$ git bisect good
Bisecting: 32 revisions left to test after this (roughly 5 steps)
[941ebc4590aaaab8489e8d03f13c519de7fb3e69] Fix report resending interactions.
```

Please test [uhk-firmware-13.0.2-941ebc45.tar.gz](https://github.com/user-attachments/files/20287289/uhk-firmware-13.0.2-941ebc45.tar.gz) on the UHK 80.

tunney, 2025-05-19:
@mondalaci this also flashed with agent 7.0

{ 5{ 5{ 5}}} in the error/warnings if relevant.

mondalaci, 2025-05-19:
The mentioned debug information is irrelevant.

```
$ git bisect good
Bisecting: 16 revisions left to test after this (roughly 4 steps)
[4bc59a5a25bee5234c287068a3c44673c7cc7802] Revert user config version bump. We don't need it for testing yet.
```

Please test [uhk-firmware-13.0.2-4bc59a5a.tar.gz](https://github.com/user-attachments/files/20290951/uhk-firmware-13.0.2-4bc59a5a.tar.gz) on the UHK 80.

tunney, 2025-05-19:
@mondalaci  [uhk-firmware-13.0.2-4bc59a5a.tar.gz](https://github.com/user-attachments/files/20290951/uhk-firmware-13.0.2-4bc59a5a.tar.gz) also flashed fine

mondalaci, 2025-05-19:
```
$ git bisect good
Bisecting: 7 revisions left to test after this (roughly 3 steps)
[0da386a78b8419a8a88bdd2b6266c1c61199d41c] Merge pull request #1221 from UltimateHackingKeyboard/fix_layer_stack_reset
```

Please test [uhk-firmware-13.0.2-0da386a7.tar.gz](https://github.com/user-attachments/files/20292356/uhk-firmware-13.0.2-0da386a7.tar.gz) on the UHK 80.

tunney, 2025-05-19:
[uhk-firmware-13.0.2-0da386a7.tar.gz](https://github.com/user-attachments/files/20292356/uhk-firmware-13.0.2-0da386a7.tar.gz) flashed fine @mondalaci

mondalaci, 2025-05-19:
```
$ git bisect good
Bisecting: 3 revisions left to test after this (roughly 2 steps)
[cca97ed6dea299e20f11efc6a38b4b05969fffe1] Merge pull request #1223 from UltimateHackingKeyboard/light_none
```

Please test [uhk-firmware-14.0.0-cca97ed6.tar.gz](https://github.com/user-attachments/files/20293786/uhk-firmware-14.0.0-cca97ed6.tar.gz) on the UHK 80.

tunney, 2025-05-19:
[uhk-firmware-14.0.0-cca97ed6.tar.gz](https://github.com/user-attachments/files/20293786/uhk-firmware-14.0.0-cca97ed6.tar.gz) flashed fine on uhk80 @mondalaci

mondalaci, 2025-05-19:
```
$ git bisect good
Bisecting: 1 revision left to test after this (roughly 1 step)
[7d3314ff2f7dd1c4aa56554d639d953efed108b2] Merge remote-tracking branch 'origin/master'
```

Please test [uhk-firmware-14.0.0-7d3314ff.tar.gz](https://github.com/user-attachments/files/20294973/uhk-firmware-14.0.0-7d3314ff.tar.gz) on the UHK 80.

tunney, 2025-05-19:
Also worked @mondalaci

mondalaci, 2025-05-19:
```
$ git bisect good
Bisecting: 0 revisions left to test after this (roughly 0 steps)
[e3cbb8cd634d5bac033300d93e84a8670649f5f3] Update deviceProtocolVersion and userConfigVersion.
```

Please test [uhk-firmware-14.0.0-e3cbb8cd.tar.gz](https://github.com/user-attachments/files/20296847/uhk-firmware-14.0.0-e3cbb8cd.tar.gz) on the UHK 80.

mondalaci, 2025-05-19:
$ git bisect good
259e54dd4a4eb834e00626f1cd00d59421000661 is the first bad commit

I'm trying to make sense of this. It's the commit tagged with `v14.0.0`, and its content doesn't reveal a bug.

Please test [uhk-firmware-14.0.0.tar.gz](https://github.com/user-attachments/files/20297261/uhk-firmware-14.0.0.tar.gz) on the UHK 80. Despite its filename, its content is not binary identical to the official 14.0.0 firmware release.

tunney, 2025-05-19:
> ```
> $ git bisect good
> Bisecting: 32 revisions left to test after this (roughly 5 steps)
> [941ebc4590aaaab8489e8d03f13c519de7fb3e69] Fix report resending interactions.
> ```
>
> Please test [...]

Interesting...

If I apply uhk-firmware-13.0.2-941ebc45 on top of uhk-firmware-13.0.2-bb992ed6 then it applies.
But if I revert to 13.0.2 then it will not.

The problem seems to be the target AND the source firmware in combination. Which would explain why only a subset see it.

I can go from
13.0.2 -> uhk-firmware-13.0.2-bb992ed6 -> Official 14.0.0 as well

But I cannot go from 13.0.2 -> official 14.0.0

Hope that helps @mondalaci

mondalaci, 2025-05-19:
Thank you for elaborating! According to your findings, the testing procedure should involve flashing firmware version 13.0.2 first, followed by the firmware build to be tested.

However, your findings make me question whether this issue is relevant anymore. Assuming there was a breaking change after which upcoming builds succeed, once one updates to 14.0.0, this issue should not cause any problems anymore.

To test this, please flash 14.0.0 and then flash a newer build, such as https://github.com/UltimateHackingKeyboard/firmware/actions/runs/15049723185 - Does it succeed?

tunney, 2025-05-19:
I agree, once on 14.0.0 then it is a non issue. But the challenge is getting onto 14.0.0 

There is a step in between that is needed, I am fairly sure that I am not the only one affected.

tunney, 2025-05-19:
> Thank you for elaborating! According to your findings, the testing procedure should involve flashing firmware version 13.0.2 first, followed by the firmware build to be tested.
>
> However, your [...]

I have already flashed 14.0.0 and then the bluetooth device fixes patch you posted elsewhere. It flashed (but it didn't let me pair)

mondalaci, 2025-05-19:
Some are surely affected, but I'm unsure if this bug is worth chasing. Maybe it is, so we can avoid it happening in the future.

To find the offending commit, we have to do another bisect session and flash 13.0.2 followed by the actual build in every iteration. If you're up to it, we can start.

```
$ git bisect start
$ git bisect good v13.0.2
$ git bisect bad v14.0.0
Bisecting: 56 revisions left to test after this (roughly 6 steps)
[2574ade3be69f43d791ef944914fb89ea14d6822] Fix setLedTxt dry run validation warnings.
```

[uhk-firmware-13.0.2-2574ade3.tar.gz](https://github.com/user-attachments/files/20299021/uhk-firmware-13.0.2-2574ade3.tar.gz)

(Let's address the pairing issue in its own issue.)

tunney, 2025-05-19:
Unfortunately my day allows it so.....

tunney, 2025-05-19:
> Some are surely affected, but I'm unsure if this bug is worth chasing. Maybe it is, so we can avoid it happening in the future.
>
> To find the offending commit, we have to do another bisect session [...]

13.0.2 to this worked.

tunney, 2025-05-19:
> ```
> $ git bisect good
> Bisecting: 35 revisions left to test after this (roughly 5 steps)
> [941ebc4590aaaab8489e8d03f13c519de7fb3e69] Fix report resending interactions.
> ```
> 
> [uhk-firmware-13.0.2-941ebc45.tar.gz](https://github.com/user-attachments/files/20299505/uhk-firmware-13.0.2-941ebc45.tar.gz)

13.0.2 to this **failed**.

mondalaci, 2025-05-19:
```
$ git bisect bad
Bisecting: 10 revisions left to test after this (roughly 3 steps)
[e25582cc4da1ce572bb7171ded028e9689675929] Adjust battery spacing.
```

[uhk-firmware-13.0.2-e25582cc.tar.gz](https://github.com/user-attachments/files/20299711/uhk-firmware-13.0.2-e25582cc.tar.gz)

tunney, 2025-05-19:
> ```
> $ git bisect bad
> Bisecting: 10 revisions left to test after this (roughly 3 steps)
> [e25582cc4da1ce572bb7171ded028e9689675929] Adjust battery spacing.
> ```
> 
> [uhk-firmware-13.0.2-e25582cc.tar.gz](https://github.com/user-attachments/files/20299711/uhk-firmware-13.0.2-e25582cc.tar.gz)

13.0.2 to this worked.

tunney, 2025-05-19:
> ```
> $ git bisect good
> Bisecting: 5 revisions left to test after this (roughly 3 steps)
> [d9b5002c2af2b230578422fed1b2515938e21d73] Blink battery icon depending on backlight powersaving mode.
> ```
> 
> [uhk-firmware-13.0.2-d9b5002c.tar.gz](https://github.com/user-attachments/files/20299808/uhk-firmware-13.0.2-d9b5002c.tar.gz)

13.0.2 to this **failed**

mondalaci, 2025-05-19:
```
$ git bisect bad
Bisecting: 2 revisions left to test after this (roughly 1 step)
[1bd2bae2ee20c7d4a0e0c7a2f7df095f87c67e18] Fix icon data offsets.
```

[uhk-firmware-13.0.2-1bd2bae2.tar.gz](https://github.com/user-attachments/files/20299970/uhk-firmware-13.0.2-1bd2bae2.tar.gz)

tunney, 2025-05-19:
> ```
> $ git bisect bad
> Bisecting: 2 revisions left to test after this (roughly 1 step)
> [1bd2bae2ee20c7d4a0e0c7a2f7df095f87c67e18] Fix icon data offsets.
> ```
> 
> [uhk-firmware-13.0.2-1bd2bae2.tar.gz](https://github.com/user-attachments/files/20299970/uhk-firmware-13.0.2-1bd2bae2.tar.gz)

Worked from 13.0.2

mondalaci, 2025-05-19:
```
$ git bisect good
Bisecting: 0 revisions left to test after this (roughly 1 step)
[687349499ba7dca1571ce6080c82fe3ae69fde22] Make backlight threshold wider to prevent oscillations.
```

[uhk-firmware-13.0.2-68734949.tar.gz](https://github.com/user-attachments/files/20300071/uhk-firmware-13.0.2-68734949.tar.gz)

mondalaci, 2025-05-19:
Looks like I can also reproduce the issue that causes the [OLED blinking](https://github.com/UltimateHackingKeyboard/firmware/issues/1226#issuecomment-2890285432). The left half has to be reset during the update to succeed.

```
$ git bisect bad
Bisecting: 0 revisions left to test after this (roughly 0 steps)
[77a6ad8711cddad1c5b5f1c1d68314f78ccf24ec] Turn off backlight under certain battery level.
```

[uhk-firmware-13.0.2-77a6ad87.tar.gz](https://github.com/user-attachments/files/20300595/uhk-firmware-13.0.2-77a6ad87.tar.gz)

$ git bisect bad
77a6ad8711cddad1c5b5f1c1d68314f78ccf24ec is the first bad commit

The v13.0.2 -> 77a6ad87 upgrade results in the OLED blinking failure, and the v13.0.2 -> 1bd2bae2 (the commit before 77a6ad87) succeeds as expected.

However, oddly, when upgrading from v13.0.2 to 1bd2bae2 to 77a6ad87, the updates succeeds.

When the OLED oscillates, the left half keeps crashing the right half keeps trying to connect.

Left half Zephyr log:
```
*** Booting nRF Connect SDK v2.8.0-6b415f6ea872 ***
*** Using Zephyr OS v3.7.99-7490142e248f ***
----------
UHK 80 Left USB started
Enabling UART
I: 3 Sectors of 4096 bytes
I: alloc wra: 0, e38
I: data wra: 0, 248
I: SoftDevice Controller build revision: 
I: fe 2c f9 6a 7f 36 22 2e |.,.j.6".
I: a0 79 c0 40 be 2c 03 20 |.y.@.,. 
I: 40 c2 f3 32             |@..2    
I: HW Platform: Nordic Semiconductor (0x0002)
I: HW Variant: nRF52x (0x0002)
I: Firmware: Standard Bluetooth controller (0x00) Version 254.63788 Build 573996906
I: No ID address. App must call settings_load()
Settings: Found peer 'right' with address 9459736334f3
I: Identity: CB:81:6B:E7:52:C9 (random)
I: HCI: version 6.0 (0x0e) revision 0x104e, manufacturer 0x0059
I: LMP: version 6.0 (0x0e) subver 0x104e
Reading hardware config
Loading Factory Config: N/A (native version: 9.0.0., at v13.0.0 / a64fc433da29704a180da11eabb24de8)
Resetting left right link! Unidirectional
Resetting left right link! Bidirectional
Bt: NUS Server module initialized.
Bt: Starting bluetooth services.
Bt: Starting advertising, try 0!
Bt: Adv: Directed NUS advertising successfully started
Conn state: UartRight(Ready)
Resetting left right link! Bidirectional
USB config changing to Hid_NoGamepad
Resetting left right link! Unidirectional

uhk80:left$ Bt: connected right (n/a, f3:34:63:73:59:94), 31 7
E: ***** BUS FAULT *****
E:   Imprecise data bus error
E: r0/a1:  0xf393e0a9  r1/a2:  0x2000cff8  r2/a3:  0x2000d037
E: r3/a4:  0xf393e0a9 r12/ip:  0x2000af48 r14/lr:  0x0003f7f5
E:  xpsr:  0x81000000
E: r4/v1:  0x00000000  r5/v2:  0x00000004  r6/v3:  0x00000021
E: r7/v4:  0x00000000  r8/v5:  0x0006a04a  r9/v6:  0x2000cff5
E: r10/v7: 0x00000000  r11/v8: 0x00000000    psp:  0x20028a68
E: EXC_RETURN: 0xfffffffd
E: Faulting instruction address (r15/pc): 0x0005ea24
E: >>> ZEPHYR FATAL ERROR 26: Unknown error on CPU 0
E: Current thread: 0x2000c260 (main)
```

Right half Zephyr log:
```
Uart: didn't receive ack 225196, resending
Conn state: UartLeft(Disconnected)
Bt: filling scan filters
Scanning failed to start (err -120)
Filters matched: left (n/a, cb:81:6b:e7:52:c9), connectable:1
Scan connecting: left (n/a, cb:81:6b:e7:52:c9)
Resetting left right link! Unidirectional
Conn state: UartLeft(Ready)
Resetting left right link! Bidirectional
Uart: didn't receive ack 226083, resending
Uart: didn't receive ack 226196, resending
Uart: didn't receive ack 226311, resending
Bt: start advertising
Uart: didn't receive ack 226425, resending
Conn state: UartLeft(Disconnected)
Bt: filling scan filters
Resetting left right link! Unidirectional
Conn state: UartLeft(Ready)
Resetting left right link! Bidirectional
Uart: didn't receive ack 227301, resending
Uart: didn't receive ack 227415, resending
Uart: didn't receive ack 227528, resending
Uart: didn't receive ack 227641, resending
Conn state: UartLeft(Disconnected)
Bt: filling scan filters
Scanning failed to start (err -120)
Resetting left right link! Unidirectional
Conn state: UartLeft(Ready)
Resetting left right link! Bidirectional
Uart: didn't receive ack 228522, resending
Uart: didn't receive ack 228636, resending
Uart: didn't receive ack 228750, resending
Bt: start advertising
Uart: didn't receive ack 228865, resending
Conn state: UartLeft(Disconnected)
Bt: filling scan filters
Resetting left right link! Unidirectional
Conn state: UartLeft(Ready)
Resetting left right link! Bidirectional
Uart: didn't receive ack 229743, resending
Uart: didn't receive ack 229857, resending
[00:03:49.953,796] <wrn> Bt: Bt security failed: left (n/a, cb:81:6b:e7:52:c9), level 1, err 9, disconnecting

Uart: didn't receive ack 229971, resending
Bt: start advertising
[00:03:50.008,331] <wrn> bt_hci_core: opcode 0x2005 status 0x0c 
[00:03:50.008,331] <wrn> bt_id: cmd disallowed
Uart: didn't receive ack 230085, resending
Conn state: UartLeft(Disconnected)
Bt: filling scan filters
Resetting left right link! Unidirectional
Conn state: UartLeft(Ready)
```

[5 more replies]

closed 2025-05-20 by mondalaci