UltimateHackingKeyboard/firmware #1228: Unable to pair BLE devices on f/w 14.0.0

tunney, 2025-05-12:
I removed all my BLE devices on my keyboard, then when trying to readd I cannot pair them. Never get the prompt on the keyboard for the pairing key.

Rolling back to 13.0.2 resolves this.

mondalaci, 2025-05-13:
I'm on 14.0.0, I don't have any BLE devices or dongles connected to my UHK 80, I can pair my Android phone, and both the keyboard and mouse functionality work well.

Do you have any dongles connected?

What's the operating system of the host you're trying to connect to?

Please provide the Zephyr log while trying to connect. Click on the Agent logo on the About page 7 times, then go to "Advanced settings", click on "Zephyr logging", then check "Right half".

tunney, 2025-05-13:
MacOs Sequoia 15.4.1
2023 MacBook nPro with M3 Max

No Dongles connected.
No devices paired.

I would provide the logs and all that but https://github.com/UltimateHackingKeyboard/firmware/issues/1226

Last week I managed after a lot of attempts to flash 14.0.0 to the left side, yesterday no joy.

Could not pair with iPhone 15 Pro or the Mac yesterday on 14.0.0. Paired when downgraded to 13.0.2

tunney, 2025-05-13:
now on MacOs 15.5

I used the workaround in the "Bricked" thread to get 14.0.0 back on the uhk80

Removed everything that was paired with the keyboard (this does not work cleanly)
Then tried to pair the MacBook

Saw UHK 80 and clicked "Connect@

[image]

Never ending spinners with no dialog on the uHK80

Turn off BT on the mac and back on and now see "UHK 80 BLE" with no option to pair

[image]

ldriscoll, 2025-05-13:
I run a dongle with two dongles, one for linux, and one for a mac on 15.4.1. When running with the 14.0.0 firmware I'm unable to connect to the dongle on the mac, however going back to 13.0.2 everything works fine. 

On 14.0.0: Initially it connected to the mac dongle ok, but after switching to the linux dongle it was unable to reattach to the mac dongle. I removed the dongles from the connection, and readded them, a number of times, to no avail.

Sorry, I don't have any logs.

mondalaci, 2025-05-15:
Please test the https://github.com/UltimateHackingKeyboard/firmware/actions/runs/15048693377 build, guys.

tunney, 2025-05-16:
Unfortunately cannot flash this firmware.

ldriscoll, 2025-05-16:
Same problem, won't connect to the dongle on my mac

tunney, 2025-05-19:
> Unfortunately cannot flash this firmware.

I can flash this now, it does not help connecting over BLE.

tunney, 2025-05-19:
UHK still only shows up as "UHK 80 BLE"

tlockney, 2025-05-20:
I'm also running into this issue with both BLE and the dongle. I can only connect via USB.

mondalaci, 2025-05-21:
Does the newly released 14.0.1 firmware resolve this issue?

tunney, 2025-05-22:
Able to pair my MacBook Pro over BLE with this firmware.

Dongle failed the first time. Appeared to work the second time however cannot switch to the "Dongle" connection

Stays as "MacBook Pro -> Dongle", I am able to switch to the paired "iPhone" and "MBP BLE".

mondalaci, 2025-05-29:
Please provide debugging information, which is now possible via Agent:

Go to the about page and click the Agent logo 7 times. Then open the newly appearing advanced settings side menu item, choose Zephyr logging, and check the right half. Now reproduce the issue and copy-paste the log.

jwr, 2025-05-29:
Not much there, I'm afraid.

I clicked "Connect" in Mac OS Settings just after the first message.

```
2025-05-29 22:30:55 | UHK 80 right    | Batteries: 98% 100% (4152 4115)
2025-05-29 22:31:11 | UHK 80 right    | Bt: start advertising
bt_l2cap: Ignoring data for unknown channel ID 0x003a
2025-05-29 22:31:35 | UHK 80 right    | Battery: ... Powered, not charging, not correcting: 4112 (99)
Battery:     ... Averaged to 4115 (99)
Step called with 100 99 -> 100
corrected voltage is 4115 100
2025-05-29 22:31:55 | UHK 80 right    | Batteries: 98% 100% (4153 4115)
2025-05-29 22:32:36 | UHK 80 right    | Battery: ... Powered, not charging, not correcting: 4120 (100)
Battery:     ... Averaged to 4116 (99)
Step called with 100 99 -> 100
corrected voltage is 4116 100
Batteries: 98% 100% (4153 4116)
2025-05-29 22:32:56 | UHK 80 right    | Batteries: 98% 100% (4155 4116)
```

kareltucek, 2025-05-29:
Please use this firmware:  [uhk-firmware-14.0.2-bt_debug-v1.3.tar.gz](https://github.com/user-attachments/files/20505631/uhk-firmware-14.0.2-bt_debug-v1.3.tar.gz)

jwr, 2025-05-30:
I flashed the firmware and tried to pair, but could not: "UHK 80 BLE" was in the Settings panel, but the "Connect" button did not appear on hover.

I then noticed that a "BTLEServer" process was consuming some CPU (not a lot, but visible, which was strange), so I killed it. The system restarted the process.

After that, the "Connect" button did appear. Complete logs:

```
2025-05-30 12:30:20 | UHK 80 right    | Batteries: 95% 99% (4161 4113)
2025-05-30 12:30:25 | UHK 80 right    | Bt: connected n/a (n/a, 84:2f:57:5d:7f:dd), 0 10

Bt: Bt connected to unknown. Starting discovery.

Bt: btManager: BtManager_StartScanningAndAdvertisingAsync because StartScanningAndAdvertisingAsync in connected - invalid connection

Bt: btManager: should scanAndAdvertise 0 1

Bt: Starting advertising, try 0!

Bt: start advertising
Bt: Adv: HID "and NUS" advertising successfully started

Bt: Service discovery completed, connection wasn't matched.

bt_l2cap: Ignoring data for unknown channel ID 0x003a
2025-05-30 12:30:27 | UHK 80 right    | Bt: n/a (n/a, 84:2f:57:5d:7f:dd) conn params: interval=15 ms, latency=0, timeout=720 ms

Bt: Received passkey pairing inquiry.

Bt: Type `uhk passkey xxxxxx` to pair, or `uhk passkey -1` to reject
2025-05-30 12:30:34 | UHK 80 right    | Bt: Sending passkey to conn n/a (n/a, 84:2f:57:5d:7f:dd)
2025-05-30 12:30:35 | UHK 80 right    | Bt: Bt connection secured: n/a (n/a, 84:2f:57:5d:7f:dd), level 4. It is auth_conn, so ignoring.

Bt: Pairing completed: n/a (n/a, 84:2f:57:5d:7f:dd), bonded 1

--- Pairing ended, success = 1: Successfuly bonded! ---
Bt: btManager: BtManager_StartScanningAndAdvertisingAsync because StartScanningAndAdvertisingAsync in BtPair_EndPairing

Conn state: Host22(New Bluetooth Device, Connected), Peer host1
Bt: Established HID connection with host1 (New Bluetooth Device, 84:2f:57:5d:7f:dd)

Conn state: Host22(New Bluetooth Device, Ready), Peer host1
Conn state: Host22(New Bluetooth Device, Ready), Peer host1, Active
Bt: btManager: BtManager_StartScanningAndAdvertisingAsync because StartScanningAndAdvertisingAsync in connectHid

Bt: btManager: BtManager_StartScanningAndAdvertisingAsync because StartScanningAndAdvertisingAsync in pairing_complete

Bt: btManager: should scanAndAdvertise 0 1

Bt: Starting advertising, try 0!

Bt: start advertising
Bt: filling adv allow filter
Bt: Adv: NUS advertising successfully started
2025-05-30 12:30:37 | UHK 80 right    | hogp: starting HID app REPORT: 1
bas: BAS Notifications enabled
2025-05-30 12:31:07 | UHK 80 right    | Flashed User Config version: 9.99.0 (native version: 11.0.0., at bt_debug-v1.3 / 11b0c1d6e3c481c090df46065a76bba7)
Setting battery mode to 1
Clearing bonds
 - Keeping a known bond.
 - Keeping a known bond.
Bt: Reallocating peer host1 from connection 23 -> 2

Resetting left right link! Unidirectional

```

I got the pairing prompt and I was able to pair!

mondalaci, 2025-05-30:
@kareltucek Is this a c2usb issue?

kareltucek, 2025-06-02:
Everything looks fine according to the log.

Please put `set bluetooth.directedAdvertisingAllowed false` into your `$onInit` and retest.

kareltucek, 2025-06-02:
> @kareltucek Is this a c2usb issue?

@mondalaci I think it is.

kareltucek, 2025-06-06:
@benedekkupper Given above, 
- 13.0.2 is good
- 14.0.0 broke peripheral connection count, which surely was responsible for some ble issues and might be responsible for pairing issues
- 14.0.1 is bad; This release fixes the peripheral connection count, but issues continue.

kareltucek, 2025-06-06:
I believe I haven't done any bluetooth related changes either.

The most significant and possibly related thing is the change of logging system to deferred logging. My guesses as to the effects is changes in timing (since logs are no longer blocking, but also are processed in larger chunks) and overall cpu demands (which I guess are higher).

Cpu performance is shared, and the failure happens while zephyr is generating security keys, which I guess is a cpu intensive task. 

As to the actual failure, it seems zephyr decides to terminate the connection when 3 or more ll buffer chunks get queued for processing. I guess that is where gatt comes in (but didn't verify), since linux decides to start spamming uhk with gatt requests for some reason....

kareltucek, 2025-06-06:
In the past whenever stack has overflown it resulted in a kernel panic. Not sure what else should be a problem there?

tunney, 2025-06-09:
@kareltucek no prompt to pair on the keyboard for BLE connections from MacBook on latest MacOs

mondalaci, 2025-08-13:
Does the https://github.com/UltimateHackingKeyboard/firmware/actions/runs/16930627109 build fix this issue?

opie4624, 2025-08-13:
This build fails to install for me. After flashing the right half, it reboots, then I see the error below.

```
[UhkOperations] Init SerialPeripheral: /dev/tty.usbmodem14101
[UhkOperations] Read UHK 80 right firmware from file
[UhkOperations] Write memory with mcumgr
[UhkOperations] Reset mcu bootloader
[UhkOperations] Close mcu communication channels
[UhkOperations] UHK 80 right firmware successfully flashed
[DeviceService] Waiting for keyboard
[DeviceService] updateFirmware error {"message":"Cannot find device with vendorId: vendorId: 14248, productId: 9","stack":"Error: Cannot find device with vendorId: vendorId: 14248, productId: 9\n at nn (/Applications/UHK Agent.app/Contents/Resources/app.asar/electron-main.js:5:539718)\n at async ro.updateFirmware (/Applications/UHK Agent.app/Contents/Resources/app.asar/electron-main.js:5:644696)\n at async Ks.process (/Applications/UHK Agent.app/Contents/Resources/app.asar/electron-main.js:5:626600)"}
[DeviceService] start poll UHK Device
[ZephyrLogService | UHK Dongle] Enabling
[ZephyrLogService | UHK 80 left] Enabling
Cannot find device with vendorId: vendorId: 14248, productId: 9
```

pcooke9, 2025-08-14:
> The smaller button

Thanks! I didn't see it because I put some heatshrink over the dongle.🙃

I got everything rolled back to release v 15.1.0 now...

closed 2025-09-17 by kareltucek