UltimateHackingKeyboard/firmware #1230: Getting a "checksum differs" error when upgrading from 13.0.2 to 14.0.0

adandersen, 2025-05-13:
Upgraded Agent to 7 and flashed the firmware that came with it. Right half went fine. Left half error'd out what with what's in the screenshots. Eventually it timed out with a stack trace. Also included.
 
I can't downgrade to an older agent version and re-flashing the firmware because it says i'm on a new configuration version (9) when i'm trying to go back to 8.3 i think. For the same reason they don't show the firmware menu option.

The firmware upgrade text is as follows:

Start flashing firmware
[DeviceService] stop poll UHK Device
[DeviceService] stopped poll UHK Device
[DeviceService] Operating system: Darwin 23.6.0 arm64
[DeviceService] Agent version: 7.0.0
[DeviceService] New firmware version: 14.0.0
[DeviceService] New firmware user config version: 11.0.0
[UhkHidDevice] productId: 9 protocol versions: {"firmwareVersion":"13.0.2","deviceProtocolVersion":"4.14.1","moduleProtocolVersion":"4.3.0","userConfigVersion":"9.0.0","hardwareConfigVersion":"1.0.0","smartMacrosVersion":"3.3.0"}
[DeviceOperation] Read "leftHalf" version information
[DeviceOperation] Read "leftHalf" repo information
[DeviceOperation] Read "leftHalf" firmware checksum
[DeviceOperation] Read "leftModule" version information
[DeviceOperation] Read "leftModule" repo information
[DeviceOperation] Read "leftModule" firmware checksum
[DeviceOperation] Read "rightModule" version information
[DeviceOperation] Read "rightModule" repo information
[DeviceOperation] Read "rightModule" firmware checksum
[DeviceService] UHK Device firmware upgrade starts: {"id":4,"asCliArg":"uhk80-right","firmwareUpgradeMethod":"mcuboot","logName":"UHK 80 right","name":"UHK 80","keyboard":[{"vid":14248,"pid":9}],"bootloader":[{"vid":14248,"pid":8}],"buspal":[],"reportId":4}
[DeviceService] Current Device right built firmware checksum: 7e2cc06b555356f211289c893a4842d7
[DeviceService] New Device right firmware checksum: d63c925ebac51c9f585a8c45c652b0bd
[UhkOperations] Start flashing device firmware with mcuboot
[UhkOperations] Start flashing UHK 80 right firmware with mcumgr
[UhkOperations] Reenumerate bootloader
[UhkHidDevice] Device communication closing.
[UhkHidDevice] Device communication closed.
[UhkHidDevice] Start reenumeration, mode: Bootloader, timeout: 5000ms
[UhkHidDevice] Available devices unchanged
[UhkHidDevice] Reenumeration error. We hope it would not break the process Error: Cannot write to hid device
[UhkHidDevice] closing normal keyboard after reenumeration
[UhkHidDevice] Available devices unchanged
[UhkHidDevice] Could not find reenumerated device: Bootloader. Waiting...
[UhkHidDevice] Available devices changed.
[UhkHidDevice] Removed: {"vendorId":"0x37A8","productId":"0x9","path":"DevSrvsID:4305401869","serialNumber":"3940418563","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 80 Right USB","release":1,"interface":3,"usagePage":12,"usage":1}
[UhkHidDevice] Removed: {"vendorId":"0x37A8","productId":"0x9","path":"DevSrvsID:4305401871","serialNumber":"3940418563","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 80 Right USB","release":1,"interface":2,"usagePage":65280,"usage":1}
[UhkHidDevice] Removed: {"vendorId":"0x37A8","productId":"0x9","path":"DevSrvsID:4305401874","serialNumber":"3940418563","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 80 Right USB","release":1,"interface":1,"usagePage":1,"usage":2}
[UhkHidDevice] Removed: {"vendorId":"0x37A8","productId":"0x9","path":"DevSrvsID:4305401876","serialNumber":"3940418563","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 80 Right USB","release":1,"interface":0,"usagePage":1,"usage":6}
[UhkHidDevice] Available devices changed.
[UhkHidDevice] Added: {"vendorId":"0x37A8","productId":"0x9","path":"DevSrvsID:4305401869","serialNumber":"3940418563","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 80 Right USB","release":1,"interface":3,"usagePage":12,"usage":1}
[UhkHidDevice] Added: {"vendorId":"0x37A8","productId":"0x9","path":"DevSrvsID:4305401871","serialNumber":"3940418563","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 80 Right USB","release":1,"interface":2,"usagePage":65280,"usage":1}
[UhkHidDevice] Added: {"vendorId":"0x37A8","productId":"0x9","path":"DevSrvsID:4305401876","serialNumber":"3940418563","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 80 Right USB","release":1,"interface":0,"usagePage":1,"usage":6}
[UhkHidDevice] Could not find reenumerated device: Bootloader. Waiting...
[UhkHidDevice] Available devices changed.
[UhkHidDevice] Removed: {"vendorId":"0x37A8","productId":"0x9","path":"DevSrvsID:4305401869","serialNumber":"3940418563","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 80 Right USB","release":1,"interface":3,"usagePage":12,"usage":1}
[UhkHidDevice] Removed: {"vendorId":"0x37A8","productId":"0x9","path":"DevSrvsID:4305401871","serialNumber":"3940418563","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 80 Right USB","release":1,"interface":2,"usagePage":65280,"usage":1}
[UhkHidDevice] Removed: {"vendorId":"0x37A8","productId":"0x9","path":"DevSrvsID:4305401874","serialNumber":"3940418563","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 80 Right USB","release":1,"interface":1,"usagePage":1,"usage":2}
[UhkHidDevice] Removed: {"vendorId":"0x37A8","productId":"0x9","path":"DevSrvsID:4305401876","serialNumber":"3940418563","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 80 Right USB","release":1,"interface":0,"usagePage":1,"usage":6}
[UhkHidDevice] Available devices unchanged
[UhkHidDevice] Could not find reenumerated device: Bootloader. Waiting...
[UhkHidDevice] Available devices unchanged
[UhkHidDevice] Could not find reenumerated device: Bootloader. Waiting...
[UhkHidDevice] Available devices unchanged
[UhkHidDevice] Could not find reenumerated device: Bootloader. Waiting...
[UhkHidDevice] Available devices unchanged
[UhkHidDevice] Could not find reenumerated device: Bootloader. Waiting...
[UhkHidDevice] Available devices unchanged
[UhkHidDevice] Could not find reenumerated device: Bootloader. Waiting...
[UhkHidDevice] Available devices unchanged
[UhkHidDevice] Could not find reenumerated device: Bootloader. Waiting...
[UhkHidDevice] Available devices changed.
[UhkHidDevice] Added: {"path":"/dev/tty.usbmodem21401","manufacturer":"Ultimate Gadget Laboratories","serialNumber":"B13289B82FCA216F","locationId":"02140000","vendorId":"0x37A8","productId":"0x0008"}
[UhkHidDevice] Available devices unchanged
[UhkHidDevice] Device communication closing.
[UhkOperations] Init SerialPeripheral: /dev/tty.usbmodem21401
[UhkOperations] Read UHK 80 right firmware from file
[... 4 lines cut ...]
[DeviceService] Waiting for keyboard
[UhkHidDevice] Available devices changed.
[UhkHidDevice] Removed: {"path":"/dev/tty.usbmodem21401","manufacturer":"Ultimate Gadget Laboratories","serialNumber":"B13289B82FCA216F","locationId":"02140000","vendorId":"0x37A8","productId":"0x0008"}
[UhkHidDevice] Added: {"vendorId":"0x37A8","productId":"0x9","path":"DevSrvsID:4305407057","serialNumber":"3940418563","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 80 Right USB","release":1,"interface":2,"usagePage":65280,"usage":1}
[UhkHidDevice] Added: {"vendorId":"0x37A8","productId":"0x9","path":"DevSrvsID:4305407059","serialNumber":"3940418563","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 80 Right USB","release":1,"interface":3,"usagePage":12,"usage":1}
[UhkHidDevice] Added: {"vendorId":"0x37A8","productId":"0x9","path":"DevSrvsID:4305407063","serialNumber":"3940418563","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 80 Right USB","release":1,"interface":0,"usagePage":1,"usage":6}
[UhkHidDevice] Used device: {"vendorId":"0x37A8","productId":"0x9","path":"DevSrvsID:4305407057","serialNumber":"3940418563","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 80 Right USB","release":1,"interface":2,"usagePage":65280,"usage":1}
[UhkHidDevice] productId: 9 protocol versions: {"firmwareVersion":"14.0.0","deviceProtocolVersion":"4.15.0","moduleProtocolVersion":"4.3.0","userConfigVersion":"11.0.0","hardwareConfigVersion":"1.0.0","smartMacrosVersion":"3.4.0"}
[DeviceOperation] Read "rightModule" version information
[DeviceOperation] Read "rightModule" repo information
[DeviceOperation] Read "rightModule" firmware checksum
[DeviceService] User configuration will be saved after right module firmware upgrade
[... cut ...]
[SmartMacroService] reference manual downloaded earlier
[UhkHidDevice] Device communication closing.
[UhkHidDevice] Device communication closed.
[DeviceService] Left module firmware version: null
[DeviceService] Left module remote firmware checksum: null
[DeviceService] Left module built firmware checksum: b216030ef7d9aa0e141eed3a7a7580d6
[DeviceService] To continue the firmware upgrade, now connect the left half via USB. (You can disconnect the right half or use a second USB cable.)

[image]
[image]
[image]

mondalaci, 2025-05-13:
Closing as a duplicate of #1226. Please look for solutions there and feel free to share your findings. We're handling this issue as a priority.

closed 2025-05-13 by mondalaci

adandersen, 2025-05-13:
Oh sorry about that, thanks for letting me know