UltimateHackingKeyboard/firmware #1231: Secondary role issue

rafaelpirolla, 2025-05-15:
Define secondary role in space key to enable Mod layer.
Define a key in the mod layer to have Shift as secondary role.
Define some vowels to have whatever secondary role.

1. Hold space key. Hold the secondary shift key. Type some normal keys.
2. Write some stuff.

You'll find that random keys gets randomly repeated (e.g. teest test test test, [procedure], ,, test)
---
It wasn't so bad until the last firmware upgrade. Now I'm find myself again trying to mess with the typing behavior without much success.

mondalaci, 2025-05-15:
Which UHK are you using?

If you use the advanced resolution strategy, we need your typing behavior settings.

kareltucek, 2025-05-15:
Please post your user config...

rafaelpirolla, 2025-05-15:
[UserConfiguration.json](https://github.com/user-attachments/files/20225064/UserConfiguration.json)

mondalaci, 2025-05-16:
Can't reproduce this one, although this use case melts my mind. I'd never have thought that someone would use recursive secondary roles.

@kareltucek I leave this up to you.

rafaelpirolla, 2025-05-17:
Seems to be fine after not using the USBC adapter. I've checked the last two release notes and just don't understand why it would impact typing behavior that I didn't have issues for years now – it was adjusted fine.

rafaelpirolla, 2025-05-25:
It got better with fine adjustments of the config.

closed 2025-05-25 by rafaelpirolla