UltimateHackingKeyboard/firmware #1235: UHK80 crash when pairing TV

pcooke9, 2025-05-16:
**FW v14.0.0-e70a7ca [master]
Both USBs & bridge cable connected
Dongle plugged into PC, but not in use**

Tried connecting to my TV via BLE. (Samsung Q80T)
•UHK80 showed the pairing code prompt, but the TV didn't offer a pairing code.
•Tapped "ESC" to close pairing code prompt, but UHK80 restarted with the following error appearing in Agent:

```
Bt: Bt security failed: n/a (n/a, bc:7e:8b:d1:ee:19), level 1, err 1, disconnecting

Bt: Pairing of auth conn failed because of 1

Bt: Pairing failed: n/a (n/a, bc:7e:8b:d1:ee:19), reason 1

ASSERTION FAIL @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/conn.c:1522
os: r0/a1:  0x00000004  r1/a2:  0x000005f2  r2/a3:  0x00000004
os: r3/a4:  0x00000004 r12/ip:  0x00000010 r14/lr:  0x0003bf5b
os:  xpsr:  0x01000000
os: r4/v1:  0xffffffff  r5/v2:  0x200155d4  r6/v3:  0x2002ccbe
os: r7/v4:  0x00000001  r8/v5:  0x00066fb9  r9/v6:  0x20001744
os: r10/v7: 0x20002414  r11/v8: 0x20002418    psp:  0x200337a8
os: EXC_RETURN: 0x0
os: Faulting instruction address (r15/pc): 0x00075072
os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
os: Current thread: 0x2000f3e0 (main)
fatal_error: Resetting system
```

pcooke9, 2025-05-16:
A second pairing attempt was "successful"...

•I was able to enter the pairing code and added the TV to Agent's connections list. The TV showed it was paired to "UHK 80 Right NUS", but no UHK output was sent to the TV.
•After rebooting the TV and the UHK80, I couldn't switch from the PC's USB, to the TV's BLE connection.

<details>
<summary>UART_Log_(2nd attempt)</summary>

```
[2025-05-16 01:45:12] Batteries: 71% 80% (3838 3794)
[2025-05-16 01:45:12] Bt: start advertising
[2025-05-16 01:45:12] Batteries: 71% 80% (3862 3794)
[2025-05-16 01:45:14] [00:01:04.826,019] <wrn> Bt: Refusing authenticated connenction 2 (this is not a selected connection)
[2025-05-16 01:45:14]
[2025-05-16 01:45:14] [00:01:04.826,354] <wrn> Bt: Refusing connenction n/a (DONGLE, e7:8a:76:d1:1a:50) (this is not a selected connection)
[2025-05-16 01:45:14]
[2025-05-16 01:45:14] Bt: start advertising
[2025-05-16 01:45:14] Batteries: 71% 80% (3868 3794)
[2025-05-16 01:45:15] [00:01:05.975,189] <wrn> Bt: Refusing authenticated connenction 2 (this is not a selected connection)
[2025-05-16 01:45:15]
[2025-05-16 01:45:15] Bt: start advertising
[2025-05-16 01:45:15] [00:01:06.997,467] <wrn> Bt: Refusing authenticated connenction 2 (this is not a selected connection)
[2025-05-16 01:45:16]
[2025-05-16 01:45:16] Bt: start advertising
[2025-05-16 01:45:16] [00:01:07.974,578] <wrn> Bt: Refusing authenticated connenction 2 (this is not a selected connection)
[2025-05-16 01:45:16]
[2025-05-16 01:45:16] Bt: start advertising
[2025-05-16 01:45:18] [00:01:08.996,124] <wrn> Bt: Refusing authenticated connenction 2 (this is not a selected connection)
[2025-05-16 01:45:18]
[2025-05-16 01:45:18] Bt: start advertising
[2025-05-16 01:45:18] [00:01:09.921,966] <wrn> Bt: Refusing authenticated connenction 2 (this is not a selected connection)
[2025-05-16 01:45:19]
[2025-05-16 01:45:19] Bt: start advertising
[2025-05-16 01:45:19] [00:01:10.968,505] <wrn> Bt: Refusing authenticated connenction 2 (this is not a selected connection)
[2025-05-16 01:45:20]
[2025-05-16 01:45:20] Bt: start advertising
[2025-05-16 01:45:22] [00:01:12.640,930] <wrn> Bt: Refusing authenticated connenction 2 (this is not a selected connection)
[2025-05-16 01:45:22]
[2025-05-16 01:45:22] Bt: start advertising
[2025-05-16 01:45:22] [00:01:13.671,905] <wrn> Bt: Refusing authenticated connenction 2 (this is not a selected connection)
[2025-05-16 01:45:23]
[2025-05-16 01:45:23] Bt: start advertising
[2025-05-16 01:45:23] [00:01:14.705,261] <wrn> Bt: Refusing authenticated connenction 2 (this is not a selected connection)
[2025-05-16 01:45:24]
[2025-05-16 01:45:24] Bt: start advertising
[2025-05-16 01:45:24] [00:01:15.631,896] <wrn> Bt: Refusing authenticated connenction 2 (this is not a selected connection)
[2025-05-16 01:45:24]
[2025-05-16 01:45:24] Bt: start advertising
[2025-05-16 01:45:24] [00:01:16.356,414] <wrn> Bt: Refusing authenticated connenction 2 (this is not a selected connection)
[2025-05-16 01:45:24]
[2025-05-16 01:45:24] Bt: start advertising
[2025-05-16 01:45:25] Bt: start advertising
[2025-05-16 01:45:25] [00:01:17.342,437] <wrn> Bt: Refusing authenticated connenction 2 (this is not a selected connection)
[2025-05-16 01:45:27]
[2025-05-16 01:45:27] Bt: start advertising
[2025-05-16 01:45:27] [00:01:18.591,979] <wrn> Bt: Refusing authenticated connenction 2 (this is not a selected connection)
[2025-05-16 01:45:28]
[2025-05-16 01:45:28] Bt: start advertising
[2025-05-16 01:45:28] [00:01:19.660,156] <wrn> Bt: Refusing authenticated connenction 2 (this is not a selected connection)
[2025-05-16 01:45:28]
[2025-05-16 01:45:28] Bt: start advertising
[2025-05-16 01:45:28] [00:01:20.387,451] <wrn> Bt: Refusing authenticated connenction 2 (this is not a selected connection)
[2025-05-16 01:45:28]
[2025-05-16 01:45:28] Bt: start advertising
[2025-05-16 01:45:29] [00:01:21.103,668] <wrn> Bt: Refusing authenticated connenction 2 (this is not a selected connection)
[2025-05-16 01:45:29]
[2025-05-16 01:45:29] Bt: start advertising
[2025-05-16 01:45:30] [00:01:22.330,932] <wrn> Bt: Refusing authenticated connenction 2 (this is not a selected connection)
[2025-05-16 01:45:30]
[2025-05-16 01:45:30] Bt: start advertising
[... 212 lines cut ...]
[2025-05-16 01:50:09] Bt: start advertising
[2025-05-16 01:50:09] Conn state: Host3(Bluetooth device, Disconnected), Selected
[2025-05-16 01:50:59] Bt: start advertising
[2025-05-16 01:50:59] [00:01:03.086,059] <inf> Battery: ... Powered, not charging, not correcting: 3788 (81)
[2025-05-16 01:51:01] [00:01:03.086,181] <inf> Battery:     ... Averaged to 3788 (81)
```
</details>

Of course, I don't really know what I'm talking about, but the following section from the "UART_Log_(2nd attempt)" seems weird to me. Shouldn't "**Host22**" be "**Host03**"? (bc:7e:8b:d1:ee:19 is the TV) I only have a few BLE devices and one dongle, definitely not 22 hosts:

```
[2025-05-16 01:45:31] Bt: start advertising
[2025-05-16 01:45:33] [00:01:24.450,408] <wrn> Bt: Refusing authenticated connenction 2 (this is not a selected connection)
[2025-05-16 01:45:33]
[2025-05-16 01:45:33] Bt: start advertising
[2025-05-16 01:45:33] [00:01:25.135,284] <wrn> Bt: Bt security failed: n/a (n/a, bc:7e:8b:d1:ee:19), level 3, err 0, disconnecting
[2025-05-16 01:45:33]
[2025-05-16 01:45:33] --- 1 messages dropped ---
[2025-05-16 01:45:33] [00:01:25.272,003] <wrn> Bt: Pairing completed: n/a (n/a, bc:7e:8b:d1:ee:19), bonded 1
[2025-05-16 01:45:33]
[2025-05-16 01:45:33] --- Pairing ended, success = 1: Successfuly bonded! ---
[2025-05-16 01:45:33] Conn state: Host22(New Bluetooth Device, Connected), Peer host1
[2025-05-16 01:45:33] Conn state: Host22(New Bluetooth Device, Ready), Peer host1
[2025-05-16 01:45:33] Conn state: Host22(New Bluetooth Device, Ready), Peer host1, Active
```

pcooke9, 2025-05-29:
After using "Reset device configuration" in Agent, my phone tried to pair repeatedly, even when I hit cancel on the phone. It kept trying, so I also tapped "ESC" on the UHK80, and it crashed with another error message in Agent:

```
Bt: Bt security failed: n/a (n/a, 66:29:a3:df:ae:ef), level 1, err 1, disconnecting

Bt: Pairing of auth conn failed because of 1

Bt: Pairing failed: n/a (n/a, 66:29:a3:df:ae:ef), reason 1

ASSERTION FAIL @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/conn.c:1522
os: r0/a1:  0x00000004  r1/a2:  0x000005f2  r2/a3:  0x00000004
os: r3/a4:  0x00000004 r12/ip:  0x00000010 r14/lr:  0x0003bf77
os:  xpsr:  0x01000000
os: r4/v1:  0xffffffff  r5/v2:  0x20015624  r6/v3:  0x2002cd10
os: r7/v4:  0x00000001  r8/v5:  0x000670cd  r9/v6:  0x20001750
os: r10/v7: 0x20002420  r11/v8: 0x20002424    psp:  0x200337e8
os: EXC_RETURN: 0x0
os: Faulting instruction address (r15/pc): 0x000751a2
os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
os: Current thread: 0x2000f3f0 (main)
fatal_error: Resetting system
Printing trace buffer because: Looks like the firmware freezed. If that is the case, please report bellow trace to the devs:

Last EV: 0
Trace:
()()()()()()()()I2,1,20,13879()OI2,1,32,15O()()()(l7l8)O()()()()
()()()()()()()()(l7l8)O()()()()()()(l7l8)O()()()()()()()()()()(l
7l8)O()()()()()()()(l7l8)O()()()()()()()()()()()()(l7l8)O()()()(
)()()()()I2,2,6,22229(<>l6u1u2v3w0w1x1x2x3x4w2w3w4w5w6ve3u3u6###
```
___
I also had the UART adapter connected this time:
<details>
<summary>UART_adapter_output</summary>

```
[2025-05-28 21:48:27] Flashed User Config version: 9.99.0 (native version: 11.0.0., at ed422fb / f09ffa0a7d1a890a093720318d9718dd)
[2025-05-28 21:48:27] Setting battery mode to 0
[2025-05-28 21:48:27] Clearing bonds
[2025-05-28 21:48:27]  - Deleting an unknown bond!
[2025-05-28 21:48:27] Unpaired device BC:7E:8B:D1:EE:19 (public)
[2025-05-28 21:48:27]  - Deleting an unknown bond!
[2025-05-28 21:48:27] Unpaired device E7:8A:76:D1:1A:50 (random)
[2025-05-28 21:48:27]  - Deleting an unknown bond!
[2025-05-28 21:48:27] Unpaired device 94:E1:29:C9:80:67 (public)
[2025-05-28 21:48:27] Resetting left right link! Unidirectional
[2025-05-28 21:48:27] Resetting left right link! Unidirectional
[2025-05-28 21:48:28] Batteries: 87% 92% (4085 4008)
[2025-05-28 21:48:29] [00:02:34.594,299] <inf> Battery: ... Powered, not charging, not correcting: 4006 (92)
[2025-05-28 21:48:29] [00:02:34.594,360] <inf> Battery:     ... Averaged to 4007 (92)
[2025-05-28 21:48:29] corrected voltage is 4007 92
[2025-05-28 21:48:30] Batteries: 87% 92% (4085 4007)
[2025-05-28 21:48:30] [00:02:35.796,478] <inf> Battery: ... Voltage corrected because of charging 4050 -> 4004 (92) -> 3999 (92)
[2025-05-28 21:48:30] [00:02:35.796,539] <inf> Battery:     ... Averaged to 4005 (92)
[2025-05-28 21:48:30] corrected voltage is 4005 92
[2025-05-28 21:48:30] Batteries: 87% 92% (4085 4005)
[2025-05-28 21:48:30] Batteries: 87% 92% (4085 4005)
[2025-05-28 21:48:32] Batteries: 87% 92% (4084 4005)
[2025-05-28 21:48:34] --- Entering pairing mode. Going to stop BT and disconnect all connections. ---
[2025-05-28 21:48:34] Settings: Found peer 'left' with address 367fe58da8ec
[2025-05-28 21:48:35] Bt: start advertising
[2025-05-28 21:48:35] Settings: Found peer 'left' with address 367fe58da8ec
[2025-05-28 21:48:35] Bt: start advertising
[2025-05-28 21:48:35] Waiting for central to pair to me.
[2025-05-28 21:48:35] Bt: start advertising
[2025-05-28 21:48:37] Bt: start advertising
[2025-05-28 21:48:37] [00:02:42.544,067] <wrn> Bt: Pairing completed: n/a (n/a, e7:8a:76:d1:1a:50), bonded 1
[2025-05-28 21:48:37]
[2025-05-28 21:48:37] --- Pairing ended, success = 1: Successfuly bonded! ---
[2025-05-28 21:48:37] [00:02:42.547,912] <wrn> Bt: Authenticated connection is not known. Disconnecting n/a (n/a, e7:8a:76:d1:1a:50)
[... 126 lines cut ...]
[2025-05-28 21:49:13] Uart, unexpected byte: 0
[2025-05-28 21:49:13] Uart, unexpected byte: 0
[2025-05-28 21:49:13] UART_RX_DISABLED
[2025-05-28 21:49:13] Scan module initialized
[2025-05-28 21:49:13] Bt: filling scan filters
[2025-05-28 21:49:14] Enabling UART
[2025-05-28 21:49:14] Conn state: UartLeft(Ready)
[2025-05-28 21:49:14] Resetting left right link! Bidirectional
[2025-05-28 21:49:14] Filters matched: left (n/a, ec:a8:8d:e5:7f:36), connectable:1
[2025-05-28 21:49:14] Scan connecting: left (n/a, ec:a8:8d:e5:7f:36)
[2025-05-28 21:49:14] Batteries: 87% 0% (4084 0)
[2025-05-28 21:49:15] [00:00:05.592,681] <wrn> bt_smp: Unexpected SMP code 0x0b
[2025-05-28 21:49:16] [00:00:05.595,733] <wrn> Bt: Bt security failed: left (n/a, ec:a8:8d:e5:7f:36), level 1, err 9, disconnecting
[2025-05-28 21:49:16]
[2025-05-28 21:49:16] Batteries: 87% 0% (4084 0)
[2025-05-28 21:49:16] Batteries: 87% 0% (4084 0)
[2025-05-28 21:49:16] [00:00:05.691,772] <wrn> bt_smp: Unexpected SMP code 0x02
[2025-05-28 21:49:16] [00:00:06.849,090] <inf> Battery: ... Voltage corrected because of charging 4058 -> 4012 (92) -> 4007 (92)
[2025-05-28 21:49:16] [00:00:06.849,456] <inf> Battery:     ... Averaged to 4007 (92)
[2025-05-28 21:49:16] corrected voltage is 4007 92
[2025-05-28 21:49:20] Batteries: 87% 92% (4084 4007)
[2025-05-28 21:49:20] Batteries: 87% 92% (4087 4007)
[2025-05-28 21:49:20] [00:00:10.942,230] <wrn> bt_hci_core: opcode 0x2013 status 0x3a
```
</details>

mhantsch, 2025-05-30:
This is similar to what I experience and described in https://github.com/UltimateHackingKeyboard/firmware/issues/999 although in my case the crash happens automatically if I try to pair through the Pop_OS! Bluetooth UI without me actively cancelling the pairing. I don't even get to the pairing code entry.

I can manually pair through commandline calls to bluetoothctl but it is weird and tedious and doesn't always work.

pcooke9, 2025-05-30:
> 22 (BLE HID) and 23 (USB) are placeholders

Ok, that makes sense.

Speaking of long wait times; I'll have to test that tomorrow. It's just about bedtime for me now.

pcooke9, 2025-05-31:
So I finally got around to trying the debug FW...

A few seconds after flashing FW v14.0.3-debug-v1.8, I got the following error in Agent:

```
Printing trace buffer because: Messenger pool space ran out!
Last EV: 786944
Trace:
2v3w0w1w2w3w4w5w6ve3u3u6u7u8)>I2,
2,6,200I3,2,6,0I4,2,6,0I5,2,6,0I6,2,6,0I7,2,6,0I8,2,6,0I9,2,6,0I
10,2,6,0I11,2,6,0I12,2,6,0I13,2,6,0I14,2,6,0I15,2,6,0I16,2,6,0E0

Printing trace buffer because: Messenger pool space ran out!
Last EV: 786944
Trace:
w4w5w6ve3u3u6u7u8)1>I2,2,6,200I3,2
,6,0I4,2,6,0I5,2,6,0I6,2,6,0I7,2,6,0I8,2,6,0I9,2,6,0I10,2,6,0I11
,2,6,0I12,2,6,0I13,2,6,0I14,2,6,0I15,2,6,0I16,2,6,0E0I16,2,6,0E0

Printing trace buffer because: Messenger pool space ran out!
Last EV: 786944
Trace:
u6u7u8)<
i1>I2,2,6,200I3,2,6,0I4,2,6,
0I5,2,6,0I6,2,6,0I7,2,6,0I8,2,6,0I9,2,6,0I10,2,6,0I11,2,6,0I12,2
,6,0I13,2,6,0I14,2,6,0I15,2,6,0I16,2,6,0E0I16,2,6,0E0I16,2,6,0E0

Printing trace buffer because: Messenger pool space ran out!
Last EV: 786944
Trace:

I2,2,6,200I3,2,6,0I4,2,6,0I5,2,6,0I6
,2,6,0I7,2,6,0I8,2,6,0I9,2,6,0I10,2,6,0I11,2,6,0I12,2

```

___

<details>
<summary>Agent_FLog</summary>

```
Start flashing firmware
[DeviceService] stop poll UHK Device
[DeviceService] stopped poll UHK Device
[ZephyrLogService | UHK Dongle] Disabling
[ZephyrLogService | UHK Dongle] wait until polling
[ZephyrLogService | UHK Dongle] stopped
[ZephyrLogService | UHK Dongle] releaseOperations
[ZephyrLogService | UHK Dongle] no instances to release
[ZephyrLogService | UHK Dongle] Disabled
[ZephyrLogService | UHK 80 left] Disabling
[ZephyrLogService | UHK 80 left] wait until polling
[ZephyrLogService | UHK 80 left] stopped
[ZephyrLogService | UHK 80 left] releaseOperations
[ZephyrLogService | UHK 80 left] no instances to release
[ZephyrLogService | UHK 80 left] Disabled
[DeviceService] Operating system: Windows_NT 10.0.26100 x64
[DeviceService] Agent version: 7.0.1
[DeviceService] New firmware version: 14.0.3
[DeviceService] New firmware user config version: 11.0.0
[DeviceService] UHK Dongle firmware upgrade starts: {"id":5,"asCliArg":"dongle","firmwareUpgradeMethod":"mcuboot","logName":"UHK Dongle","name":"UHK Dongle","keyboard":[{"vid":14248,"pid":5}],"bootloader":[{"vid":14248,"pid":4}],"buspal":[],"reportId":4}
[UhkHidDevice] Available devices unchanged
[UhkHidDevice] Used device: {"vendorId":"0x37A8","productId":"0x5","path":"\\\\?\\HID#VID_37A8&PID_0005&MI_02#8&cbf9a07&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}","serialNumber":"4294967295","manufacturer":"Ultimate Gadget Laboratories","product":"UHK dongle","release":1,"interface":2,"usagePage":65280,"usage":1}
[UhkHidDevice] productId: 5 protocol versions: {"firmwareVersion":"14.0.3","deviceProtocolVersion":"4.15.0","moduleProtocolVersion":"4.3.0","userConfigVersion":"11.0.0","hardwareConfigVersion":"1.0.0","smartMacrosVersion":"3.4.1"}
[DeviceService] Current Dongle built firmware checksum: 8dffece54d67d40dcdb8847c62881d10
[DeviceService] New Dongle firmware checksum: b69fb490677f2af27459defe80abb481
[UhkOperations] Start flashing device firmware with mcuboot
[UhkOperations] Start flashing UHK Dongle firmware with mcumgr
[UhkOperations] Reenumerate bootloader
[... 516 lines cut ...]
[2025-05-31 16:42:02]     - MPSL Work (806): 0
[2025-05-31 16:42:02]     - BT RX WQ (235): 0
[2025-05-31 16:42:02]     - sysworkq (842): 0
[2025-05-31 16:42:02]     - logging (938): 0
[2025-05-31 16:42:02]     - BT LW WQ (221): 0
[2025-05-31 16:42:02]     - flash (869): 0
[2025-05-31 16:42:02]     - BT CTLR ECDH (794): 0
[2025-05-31 16:42:02]     - state_sync_dongle_right (881): 0
[2025-05-31 16:42:02] Messenger message pool space ran out!
[2025-05-31 16:42:02] UART_RX_DISABLED
[2025-05-31 16:42:02] Enabling UART
[2025-05-31 16:42:02] Batteries: 91% 97% (4106 4088)
[2025-05-31 16:42:33] Batteries: 91% 97% (4094 4088)
[2025-05-31 16:43:03] Batteries: 91% 97% (4094 4089)
[2025-05-31 16:43:33] Batteries: 91% 97% (4098 4089)
[2025-05-31 16:44:03] Batteries: 91% 97% (4098 4090)
[2025-05-31 16:44:34] Batteries: 91% 97% (4096 4090)
[2025-05-31 16:45:04] Batteries: 91% 97% (4096 4091)
[2025-05-31 16:45:34] Batteries: 91% 97% (4099 4091)
```
</details>

___

I flashed it again, and it seemed to flash properly the second time.

pcooke9, 2025-05-31:
Another crash.
This time I intentionally initiated pairing my phone (android), and then cancelled it on the phone. It automatically tried pairing on its own again, so I tapped "ESC" on the UHK80, and it crashed with the following error in Agent:

```
Printing trace buffer because: crash/panic
Last EV: 598528
Trace:
i1>I2,2,6,18776(<>l6u1u2v3w0w1x1x2x3
x4w2w3w4w5w6ve3u3u6<
i1>bc3bu7>bu1

Bt: n/a (n/a, 51:20:ef:56:4a:27) conn params: interval=11 ms, latency=0, timeout=1000 ms

Bt: Bt security failed: n/a (n/a, 51:20:ef:56:4a:27), level 1, err 1, disconnecting

Bt: Pairing of auth conn failed because of 1

Bt: Pairing failed: n/a (n/a, 51:20:ef:56:4a:27), reason 1

Bt: Reject pairing to conn n/a (n/a, 51:20:ef:56:4a:27)

ASSERTION FAIL @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/conn.c:1522
os: r0/a1:  0x00000004  r1/a2:  0x000005f2  r2/a3:  0x00000006
os: r3/a4:  0x00000004 r12/ip:  0x00000010 r14/lr:  0x0003c06b
os:  xpsr:  0x01000000
os: r4/v1:  0x20033818  r5/v2:  0x2000a330  r6/v3:  0xffffffff
os: r7/v4:  0x20033818  r8/v5:  0x20015628  r9/v6:  0x20001750
os: r10/v7: 0x20002420  r11/v8: 0x20002424    psp:  0x200337e0
os: EXC_RETURN: 0x0
os: Faulting instruction address (r15/pc): 0x0007622a
os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
os: Current thread: 0x2000f3f0 (main)
fatal_error: Resetting system

```

___

<details>
<summary>UART_Log</summary>

```
[2025-05-31 18:20:29] Batteries: 90% 97% (4097 4082)
[2025-05-31 18:20:31] Batteries: 90% 97% (4097 4087)
[2025-05-31 18:20:32] Batteries: 90% 97% (4097 4088)
[2025-05-31 18:20:33] Batteries: 90% 97% (4095 4088)
[2025-05-31 18:20:33] Batteries: 90% 97% (4093 4088)
[2025-05-31 18:20:49] [00:02:51.496,093] <inf> Bt: btManager: BtManager_StartScanningAndAdvertisingAsync because StartScanningAndAdvertisingAsync in BtManager_EnterMode - start advertising
[2025-05-31 18:20:49]
[2025-05-31 18:20:49] [00:02:51.496,185] <inf> Bt: btManager: BtManager_StartScanningAndAdvertisingAsync because StartScanningAndAdvertisingAsync in set_command - alwaysAdvertiseHid changed
[2025-05-31 18:20:49]
[2025-05-31 18:20:49] [00:02:51.545,349] <inf> Bt: btManager: should scanAndAdvertise 0 1
[2025-05-31 18:20:49]
[2025-05-31 18:20:49] [00:02:51.545,379] <inf> Bt: Starting advertising, try 0!
[2025-05-31 18:20:49]
[2025-05-31 18:20:49] Bt: start advertising
[2025-05-31 18:20:49] [00:02:51.545,471] <inf> Bt: Adv: advertise nus+hid.
[2025-05-31 18:20:49]
[2025-05-31 18:20:49] [00:02:51.548,767] <inf> Bt: Adv: HID "and NUS" advertising successfully started
[2025-05-31 18:20:49]
[2025-05-31 18:20:49] [00:02:51.877,502] <inf> Bt: connected n/a (Dongle, e7:8a:76:d1:1a:50), 2 4
[2025-05-31 18:20:49]
[2025-05-31 18:20:50] [00:02:52.322,052] <inf> Bt: NUS peripheral connection is ready.
[2025-05-31 18:20:50]
[2025-05-31 18:20:50] Conn state: Host1(Dongle, Connected), Peer host1
[2025-05-31 18:20:50] [00:02:52.322,784] <inf> Bt: connected to host1 (Dongle, e7:8a:76:d1:1a:50)
[2025-05-31 18:20:50]
[2025-05-31 18:20:50] [00:02:52.673,614] <inf> Bt: host1 (Dongle, e7:8a:76:d1:1a:50) conn params: interval=7 ms, latency=0, timeout=1000 ms
[2025-05-31 18:20:50]
[2025-05-31 18:20:50] Conn state: Host1(Dongle, Ready), Peer host1
[... 186 lines cut ...]
[2025-05-31 18:21:58] [00:00:05.229,339] <inf> Bt: Starting advertising and scanning, try 0!
[2025-05-31 18:21:58]
[2025-05-31 18:21:58] Bt: filling scan filters
[2025-05-31 18:21:58] Filters matched: left (n/a, ec:a8:8d:e5:7f:36), connectable:1
[2025-05-31 18:21:58] Scan connecting: left (n/a, ec:a8:8d:e5:7f:36)
[2025-05-31 18:21:58] Conn state: UartLeft(Ready)
[2025-05-31 18:21:58] Resetting left right link! Bidirectional
[2025-05-31 18:21:58] [00:00:05.359,680] <inf> Bt: connected left (n/a, ec:a8:8d:e5:7f:36), 29 6
[2025-05-31 18:21:58]
[2025-05-31 18:21:58] [00:00:05.411,621] <wrn> bt_smp: Unexpected SMP code 0x0b
[2025-05-31 18:21:58] [00:00:05.415,252] <wrn> Bt: Bt security failed: left (n/a, ec:a8:8d:e5:7f:36), level 1, err 9, disconnecting
[2025-05-31 18:21:58]
[2025-05-31 18:21:58] Batteries: 90% 0% (4094 0)
[2025-05-31 18:21:58] [00:00:05.510,650] <wrn> bt_smp: Unexpected SMP code 0x02
[2025-05-31 18:22:00] Batteries: 90% 97% (4094 4094)
[2025-05-31 18:22:02] Batteries: 90% 97% (4096 4094)
[2025-05-31 18:22:04] [00:00:10.710,540] <inf> Bt: left (n/a, ec:a8:8d:e5:7f:36) conn params: interval=50 ms, latency=0, timeout=420 ms
[2025-05-31 18:22:04]
[2025-05-31 18:22:04] [00:00:10.761,199] <wrn> bt_hci_core: opcode 0x2013 status 0x3a
[2025-05-31 18:23:02] [00:00:11.060,516] <inf> Bt: left (n/a, ec:a8:8d:e5:7f:36) conn params: interval=11 ms, latency=0, timeout=1000 ms
[2025-05-31 18:23:02]
[2025-05-31 18:23:02] Batteries: 90% 97% (4096 4106)
```

</details>

pcooke9, 2025-06-02:
Turns out, I can reproduce this 100% of the time by only tapping "ESC" on the UHK80 to exit the pairing code prompt...

Here's another crash log. I used a completely default config this time. I've also re-flashed the debug-v1.8 FW a couple times, just in case. Am I correct to assume I should be using the FW **v14.0.3-debug-v1.8**, or the **14.0.3-bt_debug-v1.6** linked above?

Steps to reproduce:
•initiate pairing on android phone
•when the phone and UHK80 show code prompts, tap "ESC" on the UHK80
•observe the UHK80 crash/restart and produce log in Agent

A couple of noteworthy mentions:
•Sometimes Agent's log info shows "Last EV:" and nothing else (blank), or is misaligned/unreadable after a crash.
•Sometimes the UHK80 doesn't automatically advertise BLE after it restarts. And of course, there's no way to start/stop BLE advertising when using a default config/keymap unless you press the reset button or add a new macro.

```
Printing trace buffer because: crash/panic
Last EV: 598528
Trace:
i1>I2,2,6,27959(<>l6u1u2v3w0w1x1x2x3x4w2w3w4w5w6
ve3u3u6<
i1>bc3bu7>bu1

Bt: Bt security failed: n/a (n/a, 68:ac:73:8f:e0:23), level 1, err 1, disconnecting

Bt: Pairing of auth conn failed because of 1

Bt: Pairing failed: n/a (n/a, 68:ac:73:8f:e0:23), reason 1

Bt: Reject pairing to conn n/a (n/a, 68:ac:73:8f:e0:23)

ASSERTION FAIL @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/conn.c:1522
os: r0/a1:  0x00000004  r1/a2:  0x000005f2  r2/a3:  0x00000005
os: r3/a4:  0x00000004 r12/ip:  0x00000010 r14/lr:  0x0003c06b
os:  xpsr:  0x01000000
os: r4/v1:  0x20033818  r5/v2:  0x2000a330  r6/v3:  0xffffffff
os: r7/v4:  0x20033818  r8/v5:  0x20015628  r9/v6:  0x20001750
os: r10/v7: 0x20002420  r11/v8: 0x20002424    psp:  0x200337e0
os: EXC_RETURN: 0x0
os: Faulting instruction address (r15/pc): 0x0007622a
os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
os: Current thread: 0x2000f3f0 (main)
fatal_error: Resetting system

```

___

<details>
<summary>UART_Log</summary>

```
[2025-06-01 22:03:06] Conn state: Host0(My PC, Ready), Active
[2025-06-01 22:03:06] [00:00:52.948,089] <inf> Bt: connected n/a (n/a, 68:ac:73:8f:e0:23), 0 10
[2025-06-01 22:03:06]
[2025-06-01 22:03:06] [00:00:52.948,089] <inf> Bt: Bt connected to unknown. Starting discovery.
[2025-06-01 22:03:06]
[2025-06-01 22:03:06] [00:00:52.948,333] <inf> Bt: btManager: BtManager_StartScanningAndAdvertisingAsync because StartScanningAndAdvertisingAsync in connected - invalid connection
[2025-06-01 22:03:06]
[2025-06-01 22:03:06] [00:00:52.996,368] <inf> Bt: btManager: should scanAndAdvertise 0 1
[2025-06-01 22:03:06]
[2025-06-01 22:03:06] [00:00:52.996,398] <inf> Bt: Starting advertising, try 0!
[2025-06-01 22:03:06]
[2025-06-01 22:03:06] Bt: start advertising
[2025-06-01 22:03:06] [00:00:52.996,490] <inf> Bt: Adv: advertise nus+hid.
[2025-06-01 22:03:06]
[2025-06-01 22:03:06] [00:00:53.000,793] <inf> Bt: Adv: HID "and NUS" advertising successfully started
[2025-06-01 22:03:06]
[2025-06-01 22:03:06] [00:00:53.247,192] <inf> Bt: Service discovery completed, connection wasn't matched.
[2025-06-01 22:03:06]
[2025-06-01 22:03:06] [00:00:53.443,298] <inf> Bt: Received passkey pairing inquiry.
[2025-06-01 22:03:06]
[... 112 lines cut ...]
[2025-06-01 22:03:20] [00:00:07.411,163] <inf> Bt: host1 (Dongle, e7:8a:76:d1:1a:50) conn params: interval=7 ms, latency=0, timeout=1000 ms
[2025-06-01 22:03:20]
[2025-06-01 22:03:20] [00:00:07.596,618] <inf> Bt: connected to host1 (Dongle, e7:8a:76:d1:1a:50)
[2025-06-01 22:03:20]
[2025-06-01 22:03:20] Conn state: Host1(Dongle, Ready), Peer host1
[2025-06-01 22:03:20] [00:00:07.971,710] <inf> Bt: btManager: BtManager_StartScanningAndAdvertisingAsync because StartScanningAndAdvertisingAsync in SetConnectionConfigured
[2025-06-01 22:03:20]
[2025-06-01 22:03:20] [00:00:08.022,430] <inf> Bt: btManager: should scanAndAdvertise 0 1
[2025-06-01 22:03:20]
[2025-06-01 22:03:20] [00:00:08.022,460] <inf> Bt: Starting advertising, try 0!
[2025-06-01 22:03:23]
[2025-06-01 22:03:23] Bt: start advertising
[2025-06-01 22:03:23] [00:00:08.022,583] <inf> Bt: Adv: advertise nus+hid.
[2025-06-01 22:03:23]
[2025-06-01 22:03:23] [00:00:08.025,604] <inf> Bt: Adv: HID "and NUS" advertising successfully started
[2025-06-01 22:03:23]
[2025-06-01 22:03:23] [00:00:10.362,945] <inf> Bt: left (n/a, ec:a8:8d:e5:7f:36) conn params: interval=50 ms, latency=0, timeout=420 ms
[2025-06-01 22:03:23]
[2025-06-01 22:03:23] [00:00:10.378,417] <wrn> bt_hci_core: opcode 0x2013 status 0x3a
[2025-06-01 22:03:23] [00:00:10.627,929] <inf> Bt: left (n/a, ec:a8:8d:e5:7f:36) conn params: interval=7 ms, latency=0, timeout=1000 ms
[2025-06-01 22:03:23]

```

</details>

mondalaci, 2025-06-02:
@kareltucek Is this c2usb-related?

kareltucek, 2025-06-02:
No

kareltucek, 2025-06-03:
Much better. You can tell because this part is nicely justified into a block 😊: 

```
i1][i1][i1][i1][i1]I2,2,6,41543[i1]([]l6u1u2v3w0w1x1x2x3x4w2w3w4
w5w6ve3u3u6[i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][
i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1]bc3[i1]bu7[i1][i1
][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1]bu1
```

kareltucek, 2025-06-03:
Crash should be fixed by [uhk-firmware-14.0.3-debug-v1.12.tar.gz](https://github.com/user-attachments/files/20568350/uhk-firmware-14.0.3-debug-v1.12.tar.gz). That authentication failure unfortunately not though.

kareltucek, 2025-06-03:
Yes, sorry, I have made one more typo inbetween. Try this one please: 

[uhk-firmware-14.0.3-debug-v1.13.tar.gz](https://github.com/user-attachments/files/20575894/uhk-firmware-14.0.3-debug-v1.13.tar.gz)

pcooke9, 2025-06-04:
Should this issue be closed now?

I'm pretty sure the crashes are resolved with debug-v1.13, and I think the rest of the connection issues mentioned are being addressed elsewhere.