UltimateHackingKeyboard/firmware #1237: Unable to perform bluetooth pairing on v14.0.0

opie4624, 2025-05-19:
When I try to pair with my macbook, I see a keyboard labeled Bluetooth Device in the nearby devices.

When I click the connect button on that, the name changes to UHK 80 but spins indefinitely.
At the same time, I see this in the logs from the keyboard:

```
2025-05-09 11:09:43 | UHK 80 right | Bt: start advertising
2025-05-09 11:09:59 | UHK 80 right | Bt: start advertising bt_l2cap: Ignoring data for unknown channel ID 0x003a
2025-05-09 11:10:00 | UHK 80 right | Bt: start advertising Conn state: NusServerLeft(Disconnected), Peer left Bt: filling scan filters Bt: filling scan filters Scanning failed to start (err -120) Filters matched: left (n/a, ea:1f:32:08:61:26), connectable:1 Scan connecting: left (n/a, ea:1f:32:08:61:26) bt_smp: Unexpected SMP code 0x0b Conn state: NusServerLeft(Connected), Peer left
2025-05-09 11:10:01 | UHK 80 right | Bt: filling scan filters Conn state: NusServerLeft(Ready), Peer left Resetting left right link! Bidirectional Resetting left right link! Unidirectional Bt: start advertising bt_hci_core: opcode 0x2005 status 0x0c bt_id: cmd disallowed Batteries: 100% 100% (4027 4127)
2025-05-09 11:10:02 | UHK 80 right | Bt: start advertising
2025-05-09 11:10:05 | UHK 80 right | bt_hci_core: opcode 0x2013 status 0x3a
2025-05-09 11:10:28 | UHK 80 right | Battery: ... Powered, not charging, not correcting: 4126 (100) Battery: ... Averaged to 4127 (100)
2025-05-09 11:11:29 | UHK 80 right | Battery: ... Powered, not charging, not correcting: 4132 (100) Battery: ... Averaged to 4128 (100)
2025-05-09 11:11:43 | UHK 80 right | --- Pairing ended, success = 0: Pairing timeout --- Bt: start advertising
```

I've also been unable to pair with an additional dongle, though I _can_ pair successfully with a Samsung ZFold 6.

Attempting to pair a new dongle results in the right-half rebooting. After it reboots, it can no longer communicate over the USB cable until I push the reset button on the bottom.

opie4624, 2025-05-19:
Possibly duplicates #1228

opie4624, 2025-05-19:
I'm unable to get zephyr logs for the dongle pairing either from the dongle or uhk

mondalaci, 2025-05-21:
Please try the newly released 14.0.1 firmware.

opie4624, 2025-05-22:
14.0.1 does not resolve this. I can now pair Dongles, but can not connect to them. Keyboard stays in "Disconnected" state when paired dongle's are present until the reset button on the bottom is pressed.

Still unable to pair to Macbook while existing bluetooth pairings want a new Pairing code every time the keyboard is reset.

opie4624, 2025-05-22:
Dongle is Purple until I use the rest button under the keyboard. Then it's Blue. Dongle firmware is on 14.0.1

Repairing the dongle, logs:

```
2025-05-22 11:08:45 | UHK Dongle      | Deleting all bonds!
Unpaired device E4:4C:21:71:B0:EB (random)
Conn state: NusServerRight(Disconnected), Peer right
Bt: filling scan filters
Filters matched: right (n/a, e4:4c:21:71:b0:eb), connectable:1
Scan connecting: right (n/a, e4:4c:21:71:b0:eb)
bt_conn: conn 0x2000bfb0 failed to establish. RF noise?
Bt: Bt security failed: right (n/a, e4:4c:21:71:b0:eb), level 1, err 9, disconnecting

Bt: filling scan filters
2025-05-22 11:08:46 | UHK 80 right    | Unpaired device FF:C6:1C:BB:DF:30 (random)
Not deleting bond for 24:24:B7:99:1C:3F (public)
Not deleting bond for EA:1F:32:08:61:26 (random)
Settings: Found peer 'left' with address 266108321fea
bt_hci_core: opcode 0x2027 status 0x12 
bt_id: Failed to add IRK to controller
2025-05-22 11:08:46 | UHK Dongle      | Filters matched: right (n/a, e4:4c:21:71:b0:eb), connectable:1
Scan connecting: right (n/a, e4:4c:21:71:b0:eb)
Bt: Bt security failed: right (n/a, e4:4c:21:71:b0:eb), level 1, err 1, disconnecting

Bt: Pairing of auth conn failed because of 1

Bt: Pairing failed: right (n/a, e4:4c:21:71:b0:eb), reason 1

bt_smp: Unexpected SMP code 0x05
2025-05-22 11:08:48 | UHK 80 right    | Conn state: Host2(Dongle, Disconnected), Peer host1
Bt: start advertising
Bt: Bt security failed: n/a (Dongle, ff:c6:1c:bb:df:30), level 1, err 1, disconnecting

Bt: Pairing of auth conn failed because of 1

Bt: Pairing failed: n/a (Dongle, ff:c6:1c:bb:df:30), reason 1

bt_smp: Unexpected SMP code 0x05
2025-05-22 11:08:48 | UHK 80 right    | Flashed User Config version: 9.99.0 (native version: 11.0.0., at v14.0.1 / faa4eeb38630f30b67e521c435e4fef7)
Setting battery mode to 1
Clearing bonds
 - Keeping a known bond.
 - Keeping a known bond.
Resetting left right link! Unidirectional
2025-05-22 11:08:49 | UHK Dongle      | [UhkHidDevice] Device is not connected
2025-05-22 11:08:53 | UHK Dongle      | [UhkHidDevice] Device is not connected
2025-05-22 11:08:54 | UHK Dongle      | [UhkHidDevice] Device is not connected
2025-05-22 11:08:54 | UHK Dongle      | [UhkHidDevice] Device is not connected
2025-05-22 11:08:55 | UHK Dongle      | [UhkHidDevice] Device is not connected
2025-05-22 11:08:55 | UHK Dongle      | [UhkHidDevice] Device is not connected
2025-05-22 11:08:57 | UHK Dongle      | Bt: filling scan filters
2025-05-22 11:09:03 | UHK Dongle      | Filters matched: right (n/a, e4:4c:21:71:b0:eb), connectable:1
Scan connecting: right (n/a, e4:4c:21:71:b0:eb)
Bt: Bt security failed: right (n/a, e4:4c:21:71:b0:eb), level 1, err 1, disconnecting

Bt: Pairing of auth conn failed because of 1

Bt: Pairing failed: right (n/a, e4:4c:21:71:b0:eb), reason 1

bt_smp: Unexpected SMP code 0x05
2025-05-22 11:09:53 | UHK Dongle      | 
2025-05-22 11:09:58 | UHK Dongle      | 
2025-05-22 11:09:59 | UHK Dongle      | Filters matched: right (n/a, e4:4c:21:71:b0:eb), connectable:1Filters matched: right (n/a, e4:4c:21:71:b0:eb), connectable:1Filters matched: right (n/a, e4:4c:21:71:b0:eb), connectable:1Filters matched: right (n/a, e4:4c:21:71:b0:eb), connectable:1Filters matched: right (n/a, e4:4c:21:71:b0:eb), connectable:1Filters matched: right (n/a, e4:4c:21:71:b0:eb), connectable:1Filters matched: right (n/a, e4:4c:21:71:b0:eb), connectable:1Filters matched: right (n/a, e4:4c:21:71:b0:eb), connectable:1Filters matched: right (n/a, e4:4c:21:71:b0:eb), connectable:1Filters matched: right (n/a, e4:4c:21:71:b0:eb), connectable:1Filters matched: right (n/a, e4:4c:21:71:b0:eb), connectable:1Filters matched: right (n/a, e4:4c:21:71:b0:eb), connectable:1Filters matched: right (n/a, e4:4c:21:71:b0:eb), connectable:1Filters matched: right (n/a, e4:4c:21:71:b0:eb), connectable:1Filters matched: right (n/a, e4:4c:21:71:b0:eb), connectable:1Filters matched: right (n/a, e4:4c:21:71:b0:eb), connectable:1Filters matched: right (n/a, e4:4c:21:71:b0:eb), connectable:1
2025-05-22 11:10:00 | UHK Dongle      | 1:71:b0:eb), connectable:1
Scan connecting: right (n/a, e4:4c1:71:b0:eb), connectable:1
Scan connecting: right (n/a, e4:4c1:71:b0:eb), connectable:1
Scan connecting: right (n/a, e4:4c1:71:b0:eb), connectable:1
Scan connecting: right (n/a, e4:4c1:71:b0:eb), connectable:1
Scan connecting: right (n/a, e4:4c1:71:b0:eb), connectable:1
[... 350 lines cut ...]
2025-05-22 11:11:18 | UHK Dongle      | [UhkHidDevice] Device is not connected
2025-05-22 11:11:18 | UHK Dongle      | [UhkHidDevice] Device is not connected
2025-05-22 11:11:18 | UHK Dongle      | [UhkHidDevice] Device is not connected
2025-05-22 11:11:18 | UHK Dongle      | [UhkHidDevice] Device is not connected
2025-05-22 11:11:18 | UHK Dongle      | [UhkHidDevice] Device is not connected
2025-05-22 11:11:18 | UHK Dongle      | [UhkHidDevice] Device is not connected
2025-05-22 11:11:18 | UHK Dongle      | [UhkHidDevice] Device is not connected
2025-05-22 11:11:18 | UHK Dongle      | [UhkHidDevice] Device is not connected
2025-05-22 11:11:18 | UHK Dongle      | [UhkHidDevice] Device is not connected
2025-05-22 11:11:19 | UHK Dongle      | [UhkHidDevice] Device is not connected
2025-05-22 11:11:19 | UHK Dongle      | [UhkHidDevice] Device is not connected
2025-05-22 11:11:19 | UHK Dongle      | [UhkHidDevice] Device is not connected
2025-05-22 11:11:19 | UHK Dongle      | [UhkHidDevice] Device is not connected
[... cut ...]
2025-05-22 11:11:20 | UHK Dongle      | [UhkHidDevice] Device is not connected
2025-05-22 11:11:20 | UHK Dongle      | [UhkHidDevice] Device is not connected
2025-05-22 11:11:20 | UHK Dongle      | [UhkHidDevice] Device is not connected
2025-05-22 11:11:20 | UHK Dongle      | [UhkHidDevice] Device is not connected
2025-05-22 11:11:20 | UHK Dongle      | [UhkHidDevice] Device is not connected
2025-05-22 11:11:20 | UHK Dongle      | Cannot read properties of undefined (reading 'getDeviceState')
```

kareltucek, 2025-05-23:
Well, the log says that the pairing failed. I have just tried updating and pairing two dongles, and it succeeded without any problem. Is it possible that your mac interferes with the dongle pairing process?

kareltucek, 2025-05-23:
Please try this firmware for all new bluetooth issues (past 3 or so months) and let me know results:

[uhk-firmware-14.0.1-disable_allow_list_v1.1.tar.gz](https://github.com/user-attachments/files/20407868/uhk-firmware-14.0.1-disable_allow_list_v1.1.tar.gz)