UltimateHackingKeyboard/firmware #1272: UHK80: battery charging to 100% despite `optimizeHealth`

mhantsch, 2025-06-13:
In Agent, I have set "extend battery life".
In **$onInit**, I have `set battery.chargeLimit optimizeHealth`.

A day or two ago, I updated to firmware 14.1.0.

Before the update, my battery percentages on the OLED showed something like 67% and 83%.
Now they show 99% and 100%.

Did the calculation change, or is there a bug in the `optimizeHealth` logic?

mhantsch, 2025-06-13:
I have now saved the config again from Agent, and rebooted the UHK. Now it seems to start depleting the battery again. The OLED now shows 96%/99%, so I guess it is optimizing for health again now.

mhantsch, 2025-06-13:
I will keep an eye out over the next few days whether this happens again.

mhantsch, 2025-06-13:
I checked my UserConfig.json for anything that has 'battery' in the name. I only find these two relevant items:

```
"batteryChargingMode": "Stationary"
```

and **$onInit** containing `set battery.chargeLimit optimizeHealth`:

```
 "macros": [
    {
      "isLooped": false,
      "isPrivate": true,
      "name": "$onInit",
      "macroActions": [
        {
          "macroActionType": "command",
          "command": "clearStatus\nsetVar count 0\nsetVar snap_running 0\nsetVar lightinfo 99\n// set emergencyKey $keyId.0\n// set keystrokeDelay\nset battery.chargeLimit optimizeHealth\nset backlight.strategy constantRgb\nset backlight.constantRgb.rgb 64 64 64\nset leds.fadeTimeout 300\nset module.trackpoint.baseSpeed 0\nset module.trackpoint.speed 0.5\nset module.trackpoint.xceleration 2.0\nset module.touchpad.invertScrollDirectionX 1\nset module.touchpad.navigationMode.mouse caret\nset module.touchpad.holdContinuationTimeout 500\nset module.keycluster.navigationMode.mouse caret\n// set secondaryRole.defaultStrategy advanced\n// set secondaryRole.advanced.timeout 250\n// set secondaryRole.advanced.timeoutAction secondary\n// set secondaryRole.advanced.triggerByRelease 1\n// positive values for savetyMargin favour primary role, negative favour secondary role\n// set secondaryRole.advanced.safetyMargin +5\nsetVar safetyMargin $secondaryRole.advanced.safetyMargin\n// call hrm-init\nsetLedTxt 300 abbrev ':::' leftStatus ':::' 'init'\n//delayUntil 2999\n//fork lightinfod\n//set navigationModeAction.caret.left keystroke LA-tab\n//set navigationModeAction.caret.right keystroke LALS-tab\n//set navigationModeAction.caret.left macro window-forward\n//set navigationModeAction.caret.right macro window-backward\n"
        }
      ]
    },
```

mhantsch, 2025-06-13:
But shouldn't the maximum Value be at 3800mV in optimizeHealth mode?

kareltucek, 2025-06-13:
I mean the maximum the battery can actually reach. I.e., the value percentage is calculated from.

mhantsch, 2025-06-13:
Why is it trying to charge to that maximum? I want to `optimizeHealth`!

Any logs that may help? I have a JTAG dongle.

kareltucek, 2025-06-13:
Not when usb is connected

mhantsch, 2025-06-13:
On my Framework laptop, they recently updated the firmware to auto-detect long charging connections. They do something like this:

- if power has been connected for more than 3 days, reduce maximum level to 95%
- if power has been connected for more than 7 days, reduce maximum level to 90%
- if power has been connected for more than 14 days, reduce maximum level to 80%
- if power is disconnected for more than 2 hours, reset maximum level to 100%

For whatever the maximum level is, do the following:
- discharge until battery is at $maxlevel-5
- charge until battery is at $maxlevel

So it fluctuates by 5%-points up and down below the maximum level. The maximum level get automatically reduced if the system is connected to power for longer time.

kareltucek, 2025-06-13:
We don't have that level of control. 

There literally isn't any GPIO pin that I could pull to make the battery discharge while the usb is powered.

mhantsch, 2025-06-13:
> There literally isn't any GPIO pin that I could pull to make the battery discharge while the usb is powered

I see. OK, I will disconnect my USB for a while. Oh, I forgot, my BLE isn't working! Catch 22.

mondalaci, 2026-03-06:
Currently, https://uhk.io/manuals/uhk80 is the best page for this content. If you provide the specifics, I can update it.

I plan to replace our website with a Nuxt-based site built from Markdown files, which will provide a better foundation for this content and will be editable by you, too. I plan to implement this in the following months.

kareltucek, 2026-10-23:
This should be solved with 18.0.0, as it introduces power optimization and no longer reboots for the locked mode.

closed 2026-10-23 by kareltucek