UltimateHackingKeyboard/firmware #1285: Exec macro command over usb issues

kareltucek, 2025-06-26:
From https://forum.ultimatehackingkeyboard.com/t/errors-when-executing-commands-over-usb-on-uhk60v2/2345/8

https://github.com/user-attachments/files/20055222/uhk-firmware-13.0.2-cut_macro_hunt-v1.1.tar.gz

kareltucek, 2025-06-30:
@benedekkupper just a check: does c2usb clear the receiving (hid out) buffer after they are used? 

I.e., when I send `abcd` and then `xx`, do I receive `xx<zeroes>` or `xxcd`? (Turns out that uhk60 stack does the latter.)

benedekkupper, 2025-06-30:
There shouldn't be any modification of transfer data in the USB stack itself, so c2usb doesn't do that (if it does, please tell me and I'll fix it). It would make more sense to flush the OUT buffer, before processing the IN packet (imo).

kareltucek, 2025-07-01:
Ah, nevermind, I haven't realized that it is not c2usb that forces the memory as "const" 🤦. I can easily clear the memory myself if I want.

----------------- 

The behavior is as follows:

Assume this script:
```
#16 = current command app interface as seen in my linux
hidraw=16
echo -en "\x04\x14aaaaaaaa" > /dev/hidraw$hidraw;
echo -en "\x04\x14bbbbbbb" > /dev/hidraw$hidraw;
echo -en "\x04\x14cccccc" > /dev/hidraw$hidraw;
echo -en "\x04\x14ddddd" > /dev/hidraw$hidraw;
echo -en "\x04\x14eeee" > /dev/hidraw$hidraw;
echo -en "\x04\x14fff" > /dev/hidraw$hidraw;
echo -en "\x04\x14gg" > /dev/hidraw$hidraw;
echo -en "\x04\x14h" > /dev/hidraw$hidraw;
```

Furthermore, assume that the 0x14 command just prints the argument it received. 

This will give me this output with c2usb:

```
UsbMacroCommand: aaaaaaaa
UsbMacroCommand: bbbbbbb
UsbMacroCommand: ccccccaa
UsbMacroCommand: dddddbb
UsbMacroCommand: eeeeccaa
UsbMacroCommand: fffddbb
UsbMacroCommand: ggeeccaa
UsbMacroCommand: hffddbb
```

PR #1291 (2025-06-30): Fix things around usb macro commands.
Closes #1285 .

- Uhk60 suffered by buffer underruns on generic hid out reports. As a result, for instance `setLedTxt 500 ABC; sleep 1; uhkcmd noOp` would get mingled into a `noOp dTxt 500 ABC` command and throw error.
- Uhk80 wouldn't be awaken when the command is received. As a result, the command would be started only after some time, when the event loop got awaken by another event.
- Make the usb command return an error code when the macro command cannot be executed.
- Fix the user guide hid raw to explicitly terminate macro commands.

Steps to reprodu [... cut]