UltimateHackingKeyboard/firmware #1295: S+F & J+L to wake/unlock sends output to host

pcooke9, 2025-07-02:
On the UHK60; when using either `powerMode lock` or `powerMode sleep`, and then holding and releasing `s`+`f`+`j`+`l` (in that order), those scancodes are output to the host as `fjls` (in that order). Holding only `j`+`l` will also output `jl`, but nothing happens if you try it with only `s`+`f`.🤷‍♂️

Interestingly, if you continue to hold down the `j`+`l` keys, it will auto-repeat both keys together (similar to how the OS repeats a single key when held, only much faster than normal OS output). Likewise, if you hold `s`+`f`+`j`+`l`, it will auto-repeat all four.

E.g., Instead of `sfjllllllllllllllllll`, you get `fjlsfjlsfjlsfjlsfjlsfjlsfjlsfjls`

Instead of `jllllllllllllllll`, you get `jljljljljljljljljljljljljljljljl`.

Weird.

The UHK80 will output _some_ of the s/f/j/l keys if you hold them for a couple of seconds, and then release. But it takes longer to happen, and they won't auto-repeat like with the UHK60.

___

I would assume _nothing_ should output to the host when the UHK resumes from sleep/lock, until _after_ the first time the s/f/j/l keys are released, and _only then_ should normal output resume _upon new keypresses_...

pcooke9, 2025-07-04:
I just updated to master branch v14.2.1-3ff8688, and now the UHK80 auto-repeats when resuming from sleep mode as well. Holding `s`+`f` & `j`+`l` auto-repeats `fhksfhksfhksfhks` (which is super weird, since the `h` & `k` keys aren't even the ones being held...🤨).

pcooke9, 2025-09-19:
Thanks! No longer immediately outputs `sfjl` upon release. Still repeats `fjls` if held for an unreasonably long time (few seconds), but doing so would be pretty silly, so no longer a real issue IMO.😉

kareltucek, 2025-09-19:
Yes, that is expected ;-). Implementing a forever-suppressing variant would be unreasonably complex!

closed 2025-09-19 by pcooke9