UltimateHackingKeyboard/firmware #1304: UHK 80 - Left side battery drains very within a day

kvickyk, 2025-07-07:
I got my UHK 80 about two weeks ago. It has firmware version 14.2.0 already installed. I charged both sides and started using them wirelessly. I also have all the LEDs and the screen off completely. I noticed that almost every day, the left side completely drains its battery down to zero. The right side still sits around 90% after one day of usage.

Anything I can do to address this issue?

kvickyk, 2025-07-07:
Clicking the logo 7 times does not do anything for me.

gffranco, 2025-07-07:
Hey @kvickyk when clicking 7 times it won't change screens or anything, but it should show `Advanced Settings` just below the `Firmware` option on the left pane under Device.

kareltucek, 2025-07-09:
Hm, I see my left half now in Zephyr logging...

@kvickyk  What exactly do you mean by `It only appears if I connect to the right side and not the left side`? 

You are supposed to have your uhk80 connected by two USB cables - one to right and one to left. Bridge cable should not matter. There should be no other UHK connected to the computer at the same time. Agent should be ran without device-selection parameters (the --vid or --pid commandline arguments), but that is probably not your case (as you would know if you did that).

kvickyk, 2025-07-09:
Ah! Since the keyboard came with just one cable, I was just using that. I got it logging now. I will let it run for some time before sharing the logs. I do not see any warnings yet. This is what is there so far.

```
2025-07-09 08:43:49 | UHK 80 left     | Printing trace buffer because: Looks like the firmware freezed. If that is the case, please report bellow trace to the devs:

EV: 512
Tag: v14.2.0
Trace:
[i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1]
[i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1]
[i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1]
[i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1][i1]
2025-07-09 08:44:23 | UHK 80 left     | Battery: ... Voltage corrected because of charging 4184 -] 4116 (88) -] 4108 (87)
Battery:     ... Averaged to 4093 (84)
Battery: Step called with 83 84 -] 83

Battery: corrected voltage is 4093 83

Batteries: 83% 0% (4093 0)
2025-07-09 08:45:23 | UHK 80 left     | Battery: ... Voltage corrected because of charging 4190 -] 4112 (88) -] 4103 (86)
Battery:     ... Averaged to 4095 (85)
Battery: Step called with 83 85 -] 83

Battery: corrected voltage is 4095 83

Batteries: 83% 0% (4095 0)
2025-07-09 08:46:24 | UHK 80 left     | Battery: ... Voltage corrected because of charging 4204 -] 4114 (88) -] 4103 (86)
Battery:     ... Averaged to 4097 (85)
Battery: Step called with 83 85 -] 83

Battery: corrected voltage is 4097 83

Batteries: 83% 0% (4097 0)
2025-07-09 08:47:24 | UHK 80 left     | Battery: ... Voltage corrected because of charging 4200 -] 4128 (90) -] 4119 (89)
Battery:     ... Averaged to 4101 (86)
Battery: Step called with 83 86 -] 83

Battery: corrected voltage is 4101 83

Batteries: 83% 0% (4101 0)
2025-07-09 08:48:25 | UHK 80 left     | Battery: ... Voltage corrected because of charging 4188 -] 4116 (88) -] 4107 (87)
Battery:     ... Averaged to 4104 (86)
Battery: Step called with 83 86 -] 83

Battery: corrected voltage is 4104 83

Batteries: 83% 0% (4104 0)
```

kvickyk, 2025-07-09:
The same logs just keep repeating about the battery, I do not see any warnings

kareltucek, 2025-07-10:
We suspect a hardware problem, but before recall, lets do one more check. 

Please flash this firmware: [uhk-firmware-15.0.0-eventloop_spin_detection_v1.1.tar.gz](https://github.com/user-attachments/files/21164075/uhk-firmware-15.0.0-eventloop_spin_detection_v1.1.tar.gz)

Set the oled display on so that you can monitor the info. (Possibly with some timeout.)  

Then use it as usual, and check the oled from time to time to see if there is a warning triangle icon. If there is, start Agent and check what it says (there will be a big yellow pane in such case), and let us know. 

After a day or so of testing (i.e., after you drain it within a day), let us know.

> I noticed that _almost_ every day

Does that _almost_ mean that there are days when it drains significantly less?

kvickyk, 2025-07-11:
I will try this out and report back.

mondalaci, 2025-07-16:
I think Karel is away for about a week.

kareltucek, 2025-07-22:
@kvickyk these are from right half, not from left, which means that regarding battery drain, we are still in the dark. 

That event mask corresponds to the following: 

```
       EventVector_MouseKeys =                             1 << 3,
       EventVector_Postponer =                             1 << 5,
       EventVector_SendUsbReports =                        1 << 7,
       EventVector_ResendUsbReports =                      1 << 8,
       EventVector_KeyboardLedState =                      1 << 10,
       EventVector_UsbMacroCommandWaitingForExecution =    1 << 11,
       EventVector_ProtocolChanged =                       1 << 12,
       EventVector_LedMapUpdateNeeded =                    1 << 16,
       EventVector_ApplyConfig =                           1 << 17,
```

which is a rather strange combination. I would like to investigate this actually.

Questions:

- Can you post your UserConfig? 
- Are there any additional steps required to reproduce it?
- Did this happen right after saving user config? 
- What modules are you using?

kvickyk, 2025-07-22:
[UserConfiguration.json](https://github.com/user-attachments/files/21371500/UserConfiguration.json)

I have attached by User config. I do have the key cluster and ball mouse modules attached.

To reproduce, I saw these events as soon as I loaded the build you gave me and enabled logging. It did not take much use for those logs to appear. I did not use your build for long because it was fixed in the default config, which was hard to use for me..

kareltucek, 2025-07-23:
Ah, after another look, it looks like a false alarm caused by that config version mismatch, which is also why it was stuck on the factory keymap.

I take it that you are not seeing that warning on 15.0.1.? (It contains the same check as the testing firmware above.)

kvickyk, 2025-07-23:
I enabled it today for a few minutes before I had to step away. I got this trace below. Let me know if I need to work on it longer. 

```
2025-07-23 09:27:36 | UHK 80 left | Printing trace buffer because: Looks like the firmware freezed. If that is the case, please report bellow trace to the devs:

EV: 512
Tag: v15.0.1
Trace:
<i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1>
<i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1>
<i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1>
<i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1>
2025-07-23 09:28:22 | UHK 80 left | Battery: ... Powered, not charging, not correcting: 4052 (100)
Battery: ... Averaged to 4048 (100)
Battery: corrected voltage is 4048 100

Batteries: 100% 0% (4048 0)
2025-07-23 09:29:23 | UHK 80 left | Battery: ... Powered, not charging, not correcting: 4048 (100)
Battery: ... Averaged to 4048 (100)
Battery: corrected voltage is 4048 100
2025-07-23 09:30:23 | UHK 80 left | Battery: ... Powered, not charging, not correcting: 4050 (100)
Battery: ... Averaged to 4048 (100)
Battery: corrected voltage is 4048 100
2025-07-23 09:31:24 | UHK 80 left | Battery: ... Powered, not charging, not correcting: 4036 (98)
Battery: ... Averaged to 4046 (100)
Battery: corrected voltage is 4046 100

Batteries: 100% 0% (4046 0)
2025-07-23 09:32:24 | UHK 80 left | Battery: ... Powered, not charging, not correcting: 4046 (100)
Battery: ... Averaged to 4046 (100)
Battery: corrected voltage is 4046 100
2025-07-23 09:33:25 | UHK 80 left | Battery: ... Powered, not charging, not correcting: 4050 (100)
Battery: ... Averaged to 4046 (100)
Battery: corrected voltage is 4046 100
2025-07-23 09:34:25 | UHK 80 left | Battery: ... Powered, not charging, not correcting: 4052 (100)
Battery: ... Averaged to 4047 (100)
Battery: corrected voltage is 4047 100

Batteries: 100% 0% (4047 0)
2025-07-23 09:35:26 | UHK 80 left | Battery: ... Powered, not charging, not correcting: 4050 (100)
Battery: ... Averaged to 4048 (100)
Battery: corrected voltage is 4048 100

Batteries: 100% 0% (4048 0)
2025-07-23 09:36:26 | UHK 80 left | Battery: ... Powered, not charging, not correcting: 4044 (100)
Battery: ... Averaged to 4047 (100)
Battery: corrected voltage is 4047 100

Batteries: 100% 0% (4047 0)
2025-07-23 09:37:27 | UHK 80 left | Battery: ... Powered, not charging, not correcting: 4054 (100)
Battery: ... Averaged to 4047 (100)
Battery: corrected voltage is 4047 100
2025-07-23 09:38:27 | UHK 80 left | Battery: ... Powered, not charging, not correcting: 4044 (100)
Battery: ... Averaged to 4047 (100)
Battery: corrected voltage is 4047 100
2025-07-23 09:39:28 | UHK 80 left | Battery: ... Powered, not charging, not correcting: 4046 (100)
Battery: ... Averaged to 4048 (100)
Battery: corrected voltage is 4048 100

Batteries: 100% 0% (4048 0)
2025-07-23 09:40:28 | UHK 80 left | Battery: ... Powered, not charging, not correcting: 4046 (100)
Battery: ... Averaged to 4048 (100)
Battery: corrected voltage is 4048 100
2025-07-23 09:41:29 | UHK 80 left | Battery: ... Powered, not charging, not correcting: 4038 (99)
Battery: ... Averaged to 4046 (100)
Battery: corrected voltage is 4046 100

Batteries: 100% 0% (4046 0)
```

kvickyk, 2025-08-01:
Ok, I would like to give an update. 

I charged the new keyboard the entire day yesterday and started using it last night. Checking this morning, I am noticing the same issue, but a bit slower on the drain for the left-hand side. After charging to 100% on both sides, it's 85% | 96% currently. I used it for about an hour last night.

Also, I noticed one more difference between the old and the new. The old UHK80 would not display the battery level of the left-hand side unless I connected the spiral cable. But in the new one, I disconnected all cables, and I can see the display show the battery levels of both sides, which I like.

I will continue to use the new version and report back.

Also, I am using a module per side. I am thinking I will disconnect them to ensure they are not the problem.

kareltucek, 2025-08-01:
I am adding a new hypothesis: left and right halves may have been on different firmware versions? 

Always update both halves as well as dongles.

kvickyk, 2025-08-01:
Nope, both were on the same version, `15.0.1`, which was released recently. My hunch right now is that the key-cluster module was the main culprit for the power draw. I will update you again later in the day.

DhruvDh, 2026-01-06:
I have the same issue with firmware 16 and keycluster on 15.0.1 ; even force updating seems to not update the keycluster to 16, I am assuming firmware 16 is not available for it.

Is the only resolution to not use the keycluster?

DhruvDh, 2026-01-21:
To be more precise, it seems using keycluster + not using the bridge cable seems to cause drain.

kareltucek, 2026-10-26:
> To be more precise, it seems using keycluster + not using the bridge cable seems to cause drain.

I have measured the power pins of the bridge cable ports, and don't see any potantial on them (with disconnected batteries), so the reported interference is unlikely, unless right is powered - in that case, left would of course get power from right half. 

-----

Anyways, power consumption has been optimized for uhk80 in 18.0.0 to appx 1mA per half.

Keycluster still draws around 10mA. We can't do much more about that I am afraid.

This means that battery life should:

-   increase ~2x with keycluster - now it should last weeks
-   increase ~10x without keycluster - now it should last months

I think these values are reasonable for now -> closing.

closed 2026-10-26 by kareltucek