UltimateHackingKeyboard/firmware #1309: UHK60v2 not sending ctrl+shift+super+alt when bound to a single key

jcgl17, 2025-07-17:
I just upgraded to firmware 15.0.1 a few days ago (from 13.0.2, I think?), and am now noticing that my key bound to ctrl+shift+super+alt (I think some people call this "meh"?) is no longer working. I've checked in my desktop environment's shortcut manager, and also in a terminal-based utility. Basically, it looks like the modifiers aren't getting sent if and only if I'm using my meh key; holding the actual modifier keys works just fine.

```command
$ kitten show-key
Press any keys - Ctrl+D will terminate this program
UNIX		send_text
o		o				<----- just pressing "o"
^[[111;16u		\x1b[111;16u		<----- manually holding "ctrl-shift-super-alt-o"
o		o				<----- "meh-o"
```

I'd try to downgrade to a previous firmware version to confirm that the issue lies there and not with some other factor. But I see that v15.0.0 was labeled with `USERCONFIG:MAJOR`, so idk if a downgrade from there is supported.

kareltucek, 2025-07-22:
It works fine for me. Please post your UserConfig.

jcgl17, 2025-07-26:
My computer is broken right now and I can't do anything with my keyboard for now. Will post as soon as I have a working machine!

jcgl17, 2025-07-31:
Hm, I downgraded to both 13.0.2 and 13.0.0 and I'm still seeing this. Maybe it's something on the computer then.

I am puzzled though at the fact that manually holding down the modifiers (rather than using the meh key) seems to work as expected. Does the UHK do anything different in those two cases?

Attached is a stock config with the enter key set to meh in the QWR keymap. The issue is reproducible with this config on both 13.0.2 and 13.0.0 for me.

[stock-with-enter-meh-qwr.json](https://github.com/user-attachments/files/21532603/stock-with-enter-meh-qwr.json)

kareltucek, 2025-07-31:
>  Does the UHK do anything different in those two cases? 

No. 

> [stock-with-enter-meh-qwr.json](https://github.com/user-attachments/files/21532603/stock-with-enter-meh-qwr.json)

Still works fine for me with this config. Checking with https://keyboardchecker.com/

My conclusion is that something in your system must be intercepting it. 

What system and desktop environment are we talking about?

kareltucek, 2025-07-31:
I don't know about Plasma's specific, but on linux, please try to see what showkey says when ran directly from tty (not a terminal emulator!): 

- press Ctrl+Alt+F3 to jump to tty (or another F depending on your mappings)
- log in
- run `sudo showkey -k`
- tap your meh key
- wait 10 second for the program to terminate
- switch back to your desktop by Ctrl+Alt+F2 (or maybe some other F)

kareltucek, 2025-09-17:
Given that I don't think this is a UHK side issue and that it has been without a reply for 6 weeks, closing for now. 

Please follow up if you have any new info!