UltimateHackingKeyboard/firmware #1322: UHK80 - left part's LEDs turns off and on depending on halves' connection type

nkoshcheev, 2025-07-31:
For some reason LED stays on regardless of LED timeout settings when both halves are interconnected with a spiral cable. My use-case is most of the time I use the keyboard in a wired mode. 

When I disconnect the spiral cable, the left half backlight works properly. 
Also when the halves are connected, mod / fn / mouse layer only change the backlight on the right half of my UHK. 

Other than that the left half seems to work correctly, e.g. mod layer combinations I have on the left half work fine. Halves' link status changes correctly if I unplug the cable.

Firmware version - 15.0.1.

I'm also attaching my current configuration.

[currentconfig.json](https://github.com/user-attachments/files/21529157/currentconfig.json)

kareltucek, 2025-08-01:
Let's try to debug this:

- Flash this firmware: [uhk-firmware-15.0.1-backlight_hunt_v1.0.tar.gz](https://github.com/user-attachments/files/21533831/uhk-firmware-15.0.1-backlight_hunt_v1.0.tar.gz)
- Go to Agent logging: 
  - Open About page
  - Click Agent logo 7x
  - Scroll left pane up
  - Click the new "Advanced settings" entry
  - Click Zephyr logging
  - Check right. 
- Pick up stopwatch, then: 
  - tap a key
  - hands off keyboard
  - start the stopwatch
  - wait for your timeout
  - now remember where the log ended.
- Copy and paste the log here, ideally using another keyboard (to avoid polluting the log)

nkoshcheev, 2025-08-01:
```
2025-08-01 18:05:39 | UHK 80 right | LedManager_UpdateSleepModes: elapsedTime: 0, keyBacklightTimeout: 10000, KeyBacklightSleepModeActive: 0
LedManager_UpdateSleepModes: elapsedTime: 0, keyBacklightTimeout: 10000, KeyBacklightSleepModeActive: 0
recalculateLedBrightness: gSleep: 0, gAO: 0, KBSleep: 0, Powersaving: 0, BMode: 0, KBBB: 158, DB: 255Uart: didn't receive ack 238551, resending
Uart: didn't receive ack 238664, resending
Uart: didn't receive ack 238777, resending
Uart: didn't receive ack 238890, resending
Uart: didn't receive ack 239003, resending
Repeatedly failed to send a message! 84 86 13 86 119 4 3 55 1 15 0 158
2025-08-01 18:05:39 | UHK 80 right | 255 255 255 255 0 0 0 0 0 0 0 0 0 0 0
2025-08-01 18:05:39 | UHK 80 right | 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
2025-08-01 18:05:40 | UHK 80 right | 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
2025-08-01 18:05:40 | UHK 80 right | 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 85
Uart: didn't receive ack 240337, resending
Uart: didn't receive ack 240450, resending
2025-08-01 18:05:40 | UHK 80 right | 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
2025-08-01 18:05:41 | UHK 80 right | 0 0 0 0 0 0 0 85
Uart: didn't receive ack 240908, resending
2025-08-01 18:05:41 | UHK 80 right | 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
2025-08-01 18:05:41 | UHK 80 right | 0 0 0 0 0 85
Uart: didn't receive ack 241479, resending
2025-08-01 18:05:41 | UHK 80 right | 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
2025-08-01 18:05:42 | UHK 80 right | 0 0 0 0 0 0 85
Uart: didn't receive ack 242050, resending
2025-08-01 18:05:42 | UHK 80 right | 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
2025-08-01 18:05:42 | UHK 80 right | 0 0 0 0 0 0 0 85
Uart: didn't receive ack 242621, resending
2025-08-01 18:05:43 | UHK 80 right | 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
2025-08-01 18:05:43 | UHK 80 right | 0 0 0 0 0 85
Uart: didn't receive ack 243193, resending
2025-08-01 18:05:43 | UHK 80 right | 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
2025-08-01 18:05:43 | UHK 80 right | 0 0 0 0 0 85
2025-08-01 18:05:49 | UHK 80 right | LedManager_UpdateSleepModes: elapsedTime: 9944, keyBacklightTimeout: 10000, KeyBacklightSleepModeActive: 0
LedManager_UpdateSleepModes: elapsedTime: 10000, keyBacklightTimeout: 10000, KeyBacklightSleepModeActive: 1
LedManager_UpdateSleepModes: elapsedTime: 10001, keyBacklightTimeout: 10000, KeyBacklightSleepModeActive: 1
recalculateLedBrightness: gSleep: 0, gAO: 0, KBSleep: 1, Powersaving: 0, BMode: 0, KBBB: 0, DB: 255Uart: didn't receive ack 248606, resending
Uart: didn't receive ack 248719, resending
Uart: didn't receive ack 248833, resending
Uart: didn't receive ack 248947, resending
2025-08-01 18:05:49 | UHK 80 right | Uart: didn't receive ack 249061, resending
Repeatedly failed to send a message! 84 86 113 86 250 4 3 64 1 15 0 0
2025-08-01 18:05:49 | UHK 80 right | 255 255 255 255 0 85
```

nkoshcheev, 2025-08-05:
Additional observation: it seems that there is some sort of sync issue - if I reconnect the halves when the left side is dark, it stays that way. And in this case also, from functional perspective, everything works as expected (layers, keystrokes, etc), only the backlight logic is faulty.

nkoshcheev, 2025-08-07:
I actually have an older spare UHK60 cable - I guess as an experiment I could also rewire it and try again.

mondalaci, 2025-08-07:
Ah, gotcha! See https://ultimatehackingkeyboard.com/modding/can-i-have-a-longer-bridge-cable

nkoshcheev, 2025-08-15:
I have tried 12.x, 13.x and 14.x firmware versions - the issue reproduces on all of them.

However, what I also noticed is that regardless of the rollback / upgrade path (e.g. I've been downgrading from 15 to 12, and in the end from 13 to 15), the log shows a weird message:
```

[UhkHidDevice] Device communication closed.
[DeviceService] Left module firmware version: 0.0.0
[DeviceService] Left module remote firmware checksum:
[DeviceService] Left module built firmware checksum: 41ce0c320816a305af8ad9e3708e03c3
[DeviceService] To continue the firmware upgrade, now connect the left half via USB. (You can disconnect the right half or use a second USB cable.)
[UhkOperations] Start flashing UHK 80 left firmware with mcumgr
```

So it looks like the checksum can't be calculated. Also there was a similar warning in the UI when I did a 12->13 upgrade.

Sheldon-Cole, 2025-08-19:
My apologies for the late reply, I haven't had time to sit down and work this out. 

# kareltucek's Questions
 
>Is typing rock solid, ir is the keyboard getting dropped or stuck keys from time to time too?

Typing is rock solid. 

>can you reproduce it with a factory config?

Yes

>does it - failing led sync - happen every time you reconnect the cable?

Every time, without fail. 

>is there any unusual about your user configs? (esp., are you using macro commands in them?)

Nothing that stands out in my mind. I will attach a copy. The only macro command I have programmed is the keyboard `reboot` macro. 

>are you using functional backlighting or per-key-rgb?

Functional backlighting

>is there any specific about your environment? (Like high elmag interference or many bluetooth connections around)

Nothing stands out in my mind. My office at work might have a little more wireless/bluetooth traffic but home is pretty sparse. 

>anything comes to mind that could spin the mcu a lot?

Nothing. 

>what firmware versions are affected? Would you give a try to the latest firmware of each majir version and report back? 13.x, 14.x, 15x

Posted findings below with Agent update logs. 

>does led sync fail only when the cable is reconnected, or every time it is supposed to take place - say when tapping a layer switch?

Only fails when the bridge cable is reconnected. 

---

# Checking major versions

## 13.0.0
### Update Log Files
[13.0.0 - Bridge Cable Connected.txt](https://github.com/user-attachments/files/21847870/13.0.0.-.Bridge.Cable.Connected.txt)
[13.0.0 - Bridge Cable Disconnected.txt](https://github.com/user-attachments/files/21847871/13.0.0.-.Bridge.Cable.Disconnected.txt)

### Notes
The problem persisted in this version. However, a secondary problem arose. The keyboard would completely lock up if the *Fn2* key was tapped several times in a row requiring both halves to be reset via the reset button. I also had it lock up twice without any button presses and Agent couldn't find the right-half of the keyboard on a couple of these lockups. Problems stopped after disconnecting bridge cable.

I also got a similar error as **nkoshcheev** during the 13.0.0 update but I didn't get a copy of the output. 

## 14.0.0
### Update Log Files
[14.0.0 - Bridge Cable Connnected.txt](https://github.com/user-attachments/files/21847918/14.0.0.-.Bridge.Cable.Connnected.txt)
[14.0.0 - Bridge Cable Disconnnected.txt](https://github.com/user-attachments/files/21847919/14.0.0.-.Bridge.Cable.Disconnnected.txt)

### Notes
Nothing particularly interesting here. LED's continued to lockup on the left-half if the bridge cable was connected, but the keyboard otherwise functioned normally. Problems stopped after disconnecting bridge cable.

## 15.0.0

### Update Log Files
[15.0.0 - Bridge Cable Connected.txt](https://github.com/user-attachments/files/21847928/15.0.0.-.Bridge.Cable.Connected.txt)
[15.0.0 - Bridge Cable Disconnected.txt](https://github.com/user-attachments/files/21847953/15.0.0.-.Bridge.Cable.Disconnected.txt)

### Notes
Again, nothing particularly interesting. LED's continued to lockup on the left-half if the bridge cable was connected, but the keyboard otherwise functioned normally. Problems stopped after disconnecting bridge cable.

## 15.0.1 backlight_hunt_v1.0

### Update Log Files
[15.0.1 backlight_hunt_v1.0 - Bridge Cable Connected.txt](https://github.com/user-attachments/files/21848035/15.0.1.backlight_hunt_v1.0.-.Bridge.Cable.Connected.txt)
[15.0.1 backlight_hunt_v1.0 - Bridge Cable Disconnected.txt](https://github.com/user-attachments/files/21848037/15.0.1.backlight_hunt_v1.0.-.Bridge.Cable.Disconnected.txt)

### Notes
For good measure, I decided to test this as well. The issue persisted, however the left-half completely locked upon updating with the bridge cable connected. The LED's were on but no one was home. Problems stopped after disconnecting bridge cable.

---

# Current Config
[My current config](https://github.com/user-attachments/files/21847662/UserConfiguration.json)

nkoshcheev, 2025-08-20:
>Have you guys tried with another computer? How are your UHKs powered?

I didn't have a chance to try another computer yet; usually I connect the keyboard via my monitor, but for these tests I've used direct connection, both halves (so two USB-C cables, per half).

### Left half log:
```
2025-08-20 15:51:26 | UHK 80 left     | Enabling UART
UART_RX_STOPPED
UART_RX_STOPPED
UART_RX_STOPPED
UART_RX_DISABLED
2025-08-20 15:51:27 | UHK 80 left     | Enabling UART
UART_RX_STOPPED
UART_RX_STOPPED
UART_RX_STOPPED
Uart, unexpected byte: 128
UART_RX_DISABLED
Enabling UART
UART_RX_STOPPED
UART_RX_STOPPED
UART_RX_STOPPED
Uart, unexpected byte: 255
UART_RX_DISABLED
Enabling UART
2025-08-20 15:51:27 | UHK 80 left     | UART_RX_STOPPED
UART_RX_STOPPED
Uart, unexpected byte: 0
UART_RX_DISABLED
Enabling UART
UART_RX_STOPPED
UART_RX_STOPPED
UART_RX_STOPPED
Uart, unexpected byte: 255
UART_RX_DISABLED
Enabling UART
UART_RX_STOPPED
UART_RX_STOPPED
UART_RX_STOPPED
UART_RX_STOPPED
Uart, unexpected byte: 255
UART_RX_DISABLED
Enabling UART
UART_RX_STOPPED
UART_RX_STOPPED
Uart, unexpected byte: 248
UART_RX_DISABLED
Enabling UART
Printing trace buffer because: Looks like the firmware freezed. If that is the case, please report bellow trace to the devs:

EV: 512
Tag: uart_led_hunt_v1.1
Trace:
<i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1>
<i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1>
()<i1><i1><i1><i1><i1><i1><i1>()<i1><i1><i1><i1><i1><i1><i1><i1>
<i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1>
UART_RX_STOPPED
UART_RX_STOPPED
UART_RX_STOPPED
UART_RX_STOPPED
UART_RX_STOPPED
UART_RX_DISABLED
2025-08-20 15:51:28 | UHK 80 left     | Enabling UART
UART_RX_STOPPED
UART_RX_STOPPED
UART_RX_STOPPED
UART_RX_STOPPED
UART_RX_STOPPED
UART_RX_STOPPED
UART_RX_STOPPED
[... 377 lines cut ...]
Battery: corrected voltage is 4141 100
```
### Right half log:
```
2025-08-20 15:52:24 | UHK 80 right | Uart: didn't receive ack 237548, resending (delay 235499)
2025-08-20 15:52:27 | UHK 80 right | 0 0 1 29 0 0 0 1 27 0 0 0 1 6 0 0 0 1 25 0 0 0 1 5 0 0 0 65 0 0 0 0 65 0 0 0 0 65 0 0 0 0 3 0 0 0 0 3 0 0 0 0 1 44 0 0 0 85
Sen 27 5 StateSync LayerActionsLayer1
Uart: didn't receive ack 241738, resending (delay 241674)
2025-08-20 15:52:29 | UHK 80 right | Uart: didn't receive ack 241880, resending (delay 241752)
Uart: didn't receive ack 242149, resending (delay 241893)
Uart: didn't receive ack 242675, resending (delay 242163)
2025-08-20 15:52:30 | UHK 80 right | Uart: didn't receive ack 243713, resending (delay 242689)
2025-08-20 15:52:33 | UHK 80 right | Uart: didn't receive ack 245774, resending (delay 243726)
2025-08-20 15:52:35 | UHK 80 right | Batteries: 100% 100% (4141 4158)
2025-08-20 15:52:36 | UHK 80 right | 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 1 76 0 0 0 1 42 0 0 0 1 40 0 0 0 1 44 0 0 0 85
Sen 27 6 StateSync Backlight
Uart: didn't receive ack 249965, resending (delay 249901)
Rec 29 21 KeyStates
2025-08-20 15:52:37 | UHK 80 right | Uart: didn't receive ack 250106, resending (delay 249978)
Rec 29 22 KeyStates
Rec 29 23 KeyStates
Rec 29 24 KeyStates
Uart: didn't receive ack 250376, resending (delay 250119)
Uart: didn't receive ack 250902, resending (delay 250389)
```

kareltucek, 2025-08-20:
Those `UART_RX_STOPPED` seem relevant!

Please take another left half log with this firmware, to find out the reported reason: [uhk-firmware-15.0.1-uart_led_hunt_v1.2.tar.gz](https://github.com/user-attachments/files/21899221/uhk-firmware-15.0.1-uart_led_hunt_v1.2.tar.gz)

nkoshcheev, 2025-08-20:
Sure, here are the left half entries:

```
2025-08-20 17:44:20 | UHK 80 left     | Enabling UART
UART_RX_STOPPED, because 4
UART_RX_STOPPED, because 8
UART_RX_DISABLED
2025-08-20 17:44:21 | UHK 80 left     | Enabling UART
UART_RX_STOPPED, because 4
UART_RX_DISABLED
2025-08-20 17:44:22 | UHK 80 left     | Enabling UART
UART_RX_STOPPED, because 4
UART_RX_DISABLED
2025-08-20 17:44:23 | UHK 80 left     | Enabling UART
UART_RX_STOPPED, because 4
UART_RX_DISABLED
2025-08-20 17:44:24 | UHK 80 left     | Enabling UART
UART_RX_STOPPED, because 4
UART_RX_STOPPED, because 8
UART_RX_DISABLED
2025-08-20 17:44:25 | UHK 80 left     | Enabling UART
UART_RX_STOPPED, because 4
UART_RX_DISABLED
2025-08-20 17:44:26 | UHK 80 left     | Enabling UART
Uart, unexpected byte: 128
UART_RX_DISABLED
Enabling UART
UART_RX_STOPPED, because 4
UART_RX_DISABLED
2025-08-20 17:44:28 | UHK 80 left     | Enabling UART
UART_RX_STOPPED, because 4
UART_RX_DISABLED
Printing trace buffer because: Looks like the firmware freezed. If that is the case, please report bellow trace to the devs:

EV: 512
Tag: uart_led_hunt_v1.2
Trace:
<i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1>
<i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1>
<i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1>
<i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1>
2025-08-20 17:44:28 | UHK 80 left     | Enabling UART
UART_RX_STOPPED, because 4
UART_RX_STOPPED, because 8
UART_RX_DISABLED
2025-08-20 17:44:29 | UHK 80 left     | Enabling UART
UART_RX_STOPPED, because 4
UART_RX_DISABLED
2025-08-20 17:44:30 | UHK 80 left     | Enabling UART
UART_RX_STOPPED, because 4
UART_RX_STOPPED, because 8
Uart, unexpected byte: 252
UART_RX_DISABLED
Enabling UART
UART_RX_STOPPED, because 4
UART_RX_DISABLED
2025-08-20 17:44:31 | UHK 80 left     | Enabling UART
UART_RX_STOPPED, because 4
UART_RX_STOPPED, because 4
Uart, unexpected byte: 224
UART_RX_DISABLED
Enabling UART
UART_RX_STOPPED, because 4
UART_RX_STOPPED, because 4
Uart, unexpected byte: 224
UART_RX_DISABLED
Enabling UART
2025-08-20 17:44:32 | UHK 80 left     | UART_RX_STOPPED, because 4
UART_RX_STOPPED, because 4
UART_RX_STOPPED, because 4
UART_RX_STOPPED, because 8
UART_RX_STOPPED, because 4
UART_RX_STOPPED, because 4
UART_RX_STOPPED, because 4
UART_RX_DISABLED
Enabling UART
2025-08-20 17:44:32 | UHK 80 left     | UART_RX_STOPPED, because 4
UART_RX_STOPPED, because 4
Uart, unexpected byte: 192
UART_RX_DISABLED
Enabling UART
UART_RX_STOPPED, because 4
Uart, unexpected byte: 192
UART_RX_DISABLED
Enabling UART
UART_RX_STOPPED, because 4
UART_RX_STOPPED, because 4
UART_RX_STOPPED, because 4
Uart, unexpected byte: 255
UART_RX_DISABLED
Enabling UART
UART_RX_STOPPED, because 4
UART_RX_STOPPED, because 8
UART_RX_STOPPED, because 4
Uart, unexpected byte: 252
UART_RX_DISABLED
Enabling UART
UART_RX_STOPPED, because 4
UART_RX_STOPPED, because 4
Uart, unexpected byte: 240
UART_RX_DISABLED
Enabling UART
2025-08-20 17:44:32 | UHK 80 left     | UART_RX_STOPPED, because 4
Uart, unexpected byte: 128
UART_RX_DISABLED
Enabling UART
UART_RX_STOPPED, because 4
Uart, unexpected byte: 0
UART_RX_DISABLED
Enabling UART
UART_RX_STOPPED, because 4
Uart, unexpected byte: 0
UART_RX_DISABLED
Enabling UART
2025-08-20 17:44:33 | UHK 80 left     | UART_RX_STOPPED, because 4
UART_RX_STOPPED, because 4
Uart, unexpected byte: 192
UART_RX_DISABLED
Enabling UART
UART_RX_STOPPED, because 4
UART_RX_STOPPED, because 8
UART_RX_STOPPED, because 4
UART_RX_STOPPED, because 4
UART_RX_DISABLED
```

Sheldon-Cole, 2025-08-21:
# kareltucek's Questions

> Have you guys tried with another computer? 

Yes, this originally began when connected to a docking station for my work laptop and is now connected to my home desktop. 

> How are your UHKs powered?

Originally, it was powered by the laptop docking station, then my desktop, but is now powered by a powered USB hub that is only running the keyboard and I bought for troubleshooting this issue. 

---

# Testing

## 15.0.1 uart_led_hunt_v1.1
### Upgrade Logs

[15.0.1 uart_led_hunt_v1.1 - Bridge Cable Connected.txt](https://github.com/user-attachments/files/21909406/15.0.1.uart_led_hunt_v1.1.-.Bridge.Cable.Connected.txt)
[15.0.1 uart_led_hunt_v1.1 - Bridge Cable Disconnected.txt](https://github.com/user-attachments/files/21909407/15.0.1.uart_led_hunt_v1.1.-.Bridge.Cable.Disconnected.txt)

### Connection Logs

[Right Log - Connecting Bridge Cable.txt](https://github.com/user-attachments/files/21909410/Right.Log.-.Connecting.Bridge.Cable.txt)
[Left Log - Connecting Bridge Cable.txt](https://github.com/user-attachments/files/21909411/Left.Log.-.Connecting.Bridge.Cable.txt)

### Notes

- Left-half completely unresponsive until bridge cable was removed; no keypresses or layer changes registered
- After upgrading, the left-half version showed 0.0.0
- After initiating upgrade with the bridge cable disconnected, the version corrected

### Curiosity Test

Since the left-half locked up after the upgrade with the bridge cable connected, I decided to upgrade the device again and capture the disconnection logs from both halves immediately after upgrading, then disconnecting. 

[15.0.1_v1.1 - Bridge Cable Disconnecting - Left log.txt](https://github.com/user-attachments/files/21909610/15.0.1_v1.1.-.Bridge.Cable.Disconnecting.-.Left.log.txt)
[15.0.1_v1.1 - Bridge Cable Disconnecting - Right log.txt](https://github.com/user-attachments/files/21909609/15.0.1_v1.1.-.Bridge.Cable.Disconnecting.-.Right.log.txt)

## 15.0.1 uart_led_hunt_v1.2
### Upgrade Log
Only one upgrade log as it felt a little redundant

[15.0.1 uart_led_hunt_v1.2 - Bridge Cable Connected.txt](https://github.com/user-attachments/files/21909550/15.0.1.uart_led_hunt_v1.2.-.Bridge.Cable.Connected.txt)

### Connection Logs

[15.0.1_v1.2 - Bridge Cable Disconnection Post-Upgrade - Left log.txt](https://github.com/user-attachments/files/21909551/15.0.1_v1.2.-.Bridge.Cable.Disconnection.Post-Upgrade.-.Left.log.txt)
[15.0.1_v1.2 - Bridge Cable Connecting - Left log.txt](https://github.com/user-attachments/files/21909613/15.0.1_v1.2.-.Bridge.Cable.Connecting.-.Left.log.txt)

### Notes

The left-half locked up again after upgrading with the bridge cable connected. After disconnecting the bridge cable, the left-half began working like normal.

nkoshcheev, 2025-08-21:
Thank you very much! I have sent the request via the form.

closed 2025-09-17 by kareltucek

kareltucek, 2025-09-17:
> Were you able to figure out what was going on? 

@mondalaci ? 🙂

mondalaci, 2025-09-17:
My colleagues have replaced the PCB, and I don't know further details.