UltimateHackingKeyboard/firmware #1326: UHK60v1=Reproducible freeze from activateKeyPostponed

GrauLab, 2025-08-03:
Hi, I was able to distill a minimal config that reproduces the freezes from `activateKeyPostponed`:
- I started from the vanilla QWERTY keymap, and replaced each letter on the base row (a,s,d,f,g,h,j,k,l) with the *same one-liner macro*:
- This macro just contains the command `activateKeyPostponed atLayer fn2 prepend $thisKeyId`.
- On `fn2`, the corresponding keys are just mapped to their vanilla keystrokes in Agent.
- (An example config is attached [1], but you need to press `LMod+LCase` first to switch to the simple QWERTY test scenario.)

Expected behavior: 
- No visible change relative to QWERTY typing.

Actual behavior: 
- It works fine for some words, but then my UHK just freezes. (To get the freeze quickly, just slam the base row.)
- Sometimes the freeze occurs before sending the release event of a tap, causing OS key repetition.
- The only way out at this point seems to be power-cycling.
---
Background rationale/why this macro?:
- Imaging you have your own secondary resolver that decides to just play the primary vanilla action after all. With this `activateKeyPostponed` command you could implement the desired vanilla actions conveniently with Agent on `fn2`, and only need one elegant macro line for *any* trigger key. This can save quite some config space (relative to hard-coding and maintaining  all vanilla keystrokes on macro level).
- From a more abstract but equivalent and maybe useful point of view, this is like a `noConsume` modifier for the macro trigger key: Imagine you would use this `activateKeyPostponed prepend` immediately as first line of your resolver macro in postponing mode. Then the resulting queue state should be *identical* to before calling the resolver, making the macro "transparent". Then your resolver just needs to `toggleLayer primaryAction` or `toggleLayer secondaryAction` before opening the postponer for precisely/just the first pending key press that triggered it. In this way, the key press would be "routed" to the right actions layer. And thanks to prepend, everything would be played automatically in the correct/incoming typing order...

Remarks (more like wild guesses...) for debugging:
- Could it be a race/sync/deadlock condition when updating/prepending to the postponer queue?
- I do not fully understand whether releases have their own queue entry. (I recall that I tried once, and was ultimately unable, to just open the postponer for playing pending releases while keeping new presses postponed for screening them in a resolver/lookahead loop first. So, I guess releases are not fully independent first class citizens of the postponer queue.) So, if releases are somehow "linked" to their presses in the queue, then prepending new keys might also need careful consistent updates of these link structures, especially if the same key is pressed again? But just a wild guess...
---

As always, thank you for your help and your great work with the macro language! And have nice weekend first. :)

[1] [20250803=one-liner on QWR reproducing the activateKeyPostponed freezes.json](https://github.com/user-attachments/files/21565622/20250803.simple.one-liner.on.QWR.reproducing.the.activateKeyPostponed.freezes.json)

GrauLab, 2025-09-25:
Hi! Hmm, the "QWR" keymap that I created to reproduce the freezes with minimal macro code shouldn't produce any parentheses...🧐🤷‍♂️

For simplicity, please define a macro with just one line: `activateKeyPostponed atLayer fn2 prepend $thisKeyId`. Assign it to one of your vanilla keys on your base layer, and move the corresponding keystroke action to layer `fn2`.

Maybe, this already reproduces the freezes for you, instead of producing the expected scancode from `fn2.$thisKeyId`? If it does not freeze with just one remapped key, please assign the identical macro to some more base layer keys, moving their primary keystroke actions also to `fn2`, and then "smash" them.

_____

The idea behind this macro for base keys is: If the macro decides "primary", it can just type its own key again on `fn2`, which conveniently defines all primary actions via GUI. If it decides "secondary", it can do whatever it wants (logically "consuming" the user's key press by these actions).

To preserve the typing order, the action must be prepended, i.e., executed before any other queued key is processed. It seems to run more stable without `prepend`, but then the typing order gets wrong. More precisely, whenever the macro's `postponeKeys` lookahead for its primary/secondary decision takes longer than your typing speed...

GrauLab, 2025-09-26:
Okay, I generated a new minimal config with just the one one-liner macro [1]. Tests with firmware 15.0.1 #e007410:
- on my UHKv1:
    - After a power cycle it seems to immediately freeze when hitting any key with this macro.
    - Weirdly, just after saving this config, typing once still worked (before the power cycle).
    - When it freezes, no config saving in Agent is possible any more (error "cannot write to HID device").
- on my UHKv2:
    - After saving + power cycle it also immediately freezes whenever I hit one of the keys assigned to the one-liner macro.

[1] [20250925b=temp=just QWR to reprod the activateKeyPP freezes.json](https://github.com/user-attachments/files/22565913/20250925b.temp.just.QWR.to.reprod.the.activateKeyPP.freezes.json)

In the original config example above, the freeze was only intermittently, not 100% with the first hit. I do not understand why and am not sure if both are the "same freeze". (Maybe a relevant difference is the timer service macro that I fork on init and that always runs in the background? Or it is other macros with `postponeKeys` that are running before the one with `activateKeyPostponed prepend` gets called? Not sure...)

Thank you and have a nice weekend!

GrauLab, 2026-10-07:
Hi, I just upgraded to Agent v11 and firmware 18.0.3, and have retested this issue as you asked. Interestingly, it produces _hard faults now_ and resets the keyboard automatically, instead of just freezing as reported earlier. Reproduced several times today:
- I used the following minimal config to reproduce the issue: [just QWR to reprod the activateKeyPP freezes.json](https://github.com/user-attachments/files/33160587/20261007a.temp.just.QWR.to.reprod.the.activateKeyPP.freezes.json) 
- For surfacing this bug, the minimal config just uses the `fn2` layer to implement identical key actions that would be played if the key was not rerouted on the base layer via `activateKeyPostponed atLayer fn2 prepend $thisKeyId`.
- The hard faults always came on or after macro exit, i.e. after the log entry from `setStatus "E reprodFreeze"`.
- The log contents from Agent copied below* show some example hard faults. Sometimes they happened only after slamming `asdfghjkl...` repeatedly, one time it already happened on the first `a` after mousing around.

I hope this log is more helpful than the older freezes to identify the cause. Can you reproduce it with the attached minimal config?
Thank you again for the UHK and for your help! 

*Log:
```
(...)
pp prepend 78
E reprodFreeze
pp prepend 81
E reprodFreeze
pp prepend 17
E reprodFreeze
pp prepend 78
E reprodFreeze
pp prepend 81
E reprodFreeze
pp prepend 18
E reprodFreeze
pp prepend 17
E reprodFreeze
pp prepend 78
E reprodFreeze
pp prepend 18
E reprodFreeze
=== BusFault Detected ===
CFSR:  0x00000400
HFSR:  0x00000000
MMFAR: 0xE000ED34
BFAR:  0xE000ED38
=== Going to reboot. ===
Printing trace buffer because: Looks like your uhk60 crashed.
ID: 2, EV: 0, Tag: v18.0.3
Uptime: 0 ms, Last main: 0 ms ago
Trace:
4w5w2w3w4w5x3x4w2w3w4w5x3x4w2w3w4w5x1x2x3x4w2w3w4w5w6ve3u3u6z11,
1z12,0u7u8)(l6u1x3x4x3x4x3x4x3x4x3x4<i6>)<F8>(l6u1u2u3u6u7u8)<i6
>(l6u1u2v3w0w1x3x4x1x2x3x4w2w3w4w5x3x4w2w3w4w5w2w3w4w5x3x4w2w3w4
w5x3x4w2w3w4w5x3x4w6ve3u3<i6<U18>>u6u7u8)(l6u1u2v1FaultReboot###

asdfeosasdlk;jlksdlfjas;dl;kfaasdfjkl;asdasdfhkjl;asdfhkl;asdhfkl;asdfhkl;asdfhklasdfhjkalsdfhjklasdfhjklasdfhjkl'asdhfjkl
asdhfjklasdfhjlkasdfhjkalsdfhjalksdfhjasdklfhasdfljkasdhfjklasdhfjklasdfhjlaksdflkjahsdf'asdf
asdfh
asdhfjklasdfhjklasd;lkjfdhsjaklasdlkjhasdfhjklasdhjkalsdfasdfhjklasdfhjkl'
asdfhjklasdfhjlkasdfhjklasdfhjklasdfhjklasdfhjklasdfhjklasdfhjkl
asdfhjklasdfhjklasdfhjklasdfhjklasdflkjhasdfhjklasdfhjkl
asdhfjklasdfhjklasdfhjklasdfhjklasdfhjklasdfhjklasdfhjlkasdfhjkl
asdfhjklasdfhjlkasdfhjklasdfhjklasdfhjklasdfhjklasdfhjklasdfhjkl
asdfhjklasdfhjklasdfhjklasdfhjklasdfhjklasdfhjklasdfhjklasdfhjkl
asdfhjklasdfhjklasdfhjklasdfhjklasdfhjklasdfhjklasdfhjkl
asdfhjklasdhfalksdjfhalsdfkjhasldfkhjasdlfkjh
asdhfjklasdhfalsdkjfhalsdkfjhasldfkjhasdflkjhasdflakjhsdfpp 

(...)
E reprodFreeze
pp prepend 79
E reprodFreeze
pp prepend 80
E reprodFreeze
pp prepend 81
E reprodFreeze
pp prepend 18
E reprodFreeze
pp prepend 17
E reprodFreeze
pp prepend 16
E reprodFreeze
pp prepend 15
E reprodFreeze
pp prepend 78
E reprodFreeze
pp prepend 79
E reprodFreeze
pp prepend 80
E reprodFreeze
pp prepend 81
E reprodFreeze
pp prepend 18
E reprodFreeze
pp prepend 78
E reprodFreeze
pp prepend 17
E reprodFreeze
pp prepend 16
E reprodFreeze
pp prepend 15
E reprodFreeze
pp prepend 79
E reprodFreeze
pp prepend 80
E reprodFreeze
pp prepend 81
E reprodFreeze
pp prepend 78
E reprodFreeze
=== HardFault Detected ===
CFSR:  0x00000400
HFSR:  0x40000000
MMFAR: 0xE000ED34
BFAR:  0xE000ED38
=== Going to reboot. ===
Printing trace buffer because: Looks like your uhk60 crashed.
ID: 2, EV: 0, Tag: v18.0.3
Uptime: 0 ms, Last main: 0 ms ago
Trace:
z11,1z12,0u7u8)(l6u1x3x4x3x4x3x4x3x4x3x4x3x4)<i6><F8>(l6u1u2u3u6
u7u8)<F9>(l6u1u2v3w0w1x3x4x3x4x3x4x3x4w6ve3u7u8)<F9>(l6u1u2v3w0w
1x3x4x3x4x3x4w6ve3u7u8)<F9>(l6u1u2v3w0w1x3x4x3x4w6ve3u7u8)(l6u1u
2v3w0w1x3x4x1x2x3x4w2w3w4w5x3x4w6ve3u7u8)(l6u1u2v1FaultReboot###
pp prepend 18
E reprodFreeze
=== HardFault Detected ===
CFSR:  0x00000400
HFSR:  0x40000000
MMFAR: 0xE000ED34
BFAR:  0xE000ED38
=== Going to reboot. ===
Printing trace buffer because: Looks like your uhk60 crashed.
ID: 2, EV: 0, Tag: v18.0.3
Uptime: 0 ms, Last main: 0 ms ago
Trace:
3w4w5w6ve3u3u6u7u8)<i6><i6<U0>><i6><i6><i6><i6<U0>><i6><i6><i6><
i6<U0>><i6><i6><i6><i6<U0>><i6><i6><i6><i6<U9>><i6><i6><i6><i6<U
14>><i6><i6><i6><i6<U14>><i6><i6><i6><i6<U14>><i6><i6>(l6u1u2v3w
0w1x1x2x3x4w2w3w4w5w2w3w4w5w6ve3u3u6u7u8)(l6u1u2v1FaultReboot###
```

kareltucek, 2026-10-07:
Thanks for report! Will look into it!

kareltucek, 2026-10-07:
Should be fixed by https://github.com/UltimateHackingKeyboard/firmware/pull/1637 :)

GrauLab, 2026-10-07:
Tested your `uhk-firmware-18.0.3-ff1c2c8.tar.gz` and it works reliably now, super! 😀 Thank you very much for the ultra quick fix!👍  (This really helps me to regain some config space by generically defining all primary actions of keys assigned to my custom secondaries resolver macro, and even elegantly so on just another layer via Agent UI, instead of hardcoding `hold/tabKeys` for each possible trigger `keyID`...)

kareltucek, 2026-10-07:
Thank *you* for the report, patience and help with reproduction 😉.

closed 2026-10-08 by kareltucek