UltimateHackingKeyboard/firmware #1327: [UHK80] Dongle (host1 (Dongle, f9:b3:cb:6a:51:54)) and right half run different dongle protocol versions

gffranco, 2025-08-04:
Started getting the following message logged by my UHK80 after upgrading to version 15.1.0:

``` 
Dongle (host1 (Dongle, f9:b3:cb:6a:51:54)) and right half run different dongle protocol versions
  (dongle: 0.0.0, right: 1.0.2)
  Please upgrade!
```

According to the Agent firmware page all versions are the same:

```
 UHK Dongle firmware: UltimateHackingKeyboard/firmware: 15.1.0
 Right keyboard half firmware: UltimateHackingKeyboard/firmware: 15.1.0
 Left keyboard half firmware: UltimateHackingKeyboard/firmware: 15.1.0
```

Let me know if you need any more info on my side.

kareltucek, 2025-08-05:
Thanks for report!

You can safely ignore it. I will fix the algorithm to ignore 0.0.0 version (which means that no information is available - it was meant mainly to catch the cases of firmwares that wete so old that they didn't implement the dongle protocol versioning).

kareltucek, 2025-09-16:
Should be fixed by https://github.com/UltimateHackingKeyboard/firmware/commit/724419dd1f2ef863bee64ec166c2b41d03149a8b 

Thanks for report!