UltimateHackingKeyboard/firmware #1329: Different firmware versions on left and right keyboard - or not?

Peter111160, 2025-08-13:
I just updated the firmware of my UHK 80. In uhk-agent I get the information, I have now 15.0.1 on both keyboard halves:

"UHK Dongle firmware: UltimateHackingKeyboard/firmware: 15.0.1
 Right keyboard half firmware: UltimateHackingKeyboard/firmware: 15.0.1
 Left keyboard half firmware: UltimateHackingKeyboard/firmware: 15.0.1
 Key cluster firmware: UltimateHackingKeyboard/firmware: 11.2.0
 Trackpoint firmware: UltimateHackingKeyboard/firmware: 11.2.0"

But I get also an error message that's telling me:

"Error: Left and right keyboard halves have different firmware versions (Left: 14.0.0, Right: 15.0.1)!
Error: Left and right keyboard halves have different git tags (Left: v14.0.0, Right: v15.0.1)!
Error: Left checksum differs from the expected! Expected '41ce0c320816a305af8ad9e3708e03c3', got 'b216030ef7d9aa0e141eed3a7a7580d6'!
    Please flash both halves to the same version!"

Who's right?

mondalaci, 2025-08-14:
It's almost certainly a non-issue, but let us look into it.

@kareltucek Assuming it's a false alarm, how can we prevent it from happening?

kareltucek, 2025-08-14:
Proper way: add a usb variable that allows agent to disable the warning during flashing.

Another way: don't even begin the flashing procedure unless both halves are connected via usb.

But for now feel free to comment it out in the source code.

kareltucek, 2025-08-14:
@Peter111160 the warning was probably emitted during flashing when in fact right was on 15.x while left was still on 14.x. 

We combat that by a simple timer, but I guess you have used a single usb cable and have left the flashing for a minute or two when the cable needed to be reconnected to the left half, thus exceeding the check timer. Or your usb transfer speed may simply be lower than what we have tested with...

closed 2025-09-22 by mondalaci