UltimateHackingKeyboard/firmware #1330: Smart Macro expression evaluation issue

firngrod, 2025-08-21:
There seems to be an issue to do with ending while loops with function blocks and then immediately evaluating expressions.  Running the following macro, for example, will make the keyboard type `y` rather than the expected `yy`

```
setVar test 0
while ($test == 0) {
  setVar test 1
}
if (1) {
 tapKey y
}
if (1) {
 tapKey y
}
```

Removing the block and just whiling one statement will work:

```
setVar test 0
while ($test == 0) setVar test 1
if (1) {
 tapKey y
}
if (1) {
 tapKey y
}
```

If the while evaluated false prior to the first loop, there is no issue.

Running UHK V1 firmware 15.1.0

kareltucek, 2025-08-21:
Cute!

Thanks for report! Will look into it, although I am afraid it will have to wait a bit.

mhantsch, 2025-08-22:
> I am deeply impressed with the macro engine! It's let me finally set up secondary key behavior how I want it for HRMs.

It's a great engine, isn't it? BTW, I am curious: for HRM, did you see my setup on the [UHK forum thread about HRMs](https://forum.ultimatehackingkeyboard.com/t/miryoku-and-homerow-mods/723/24?u=maexxx)?

firngrod, 2025-08-22:
I don't expect you to fix this one, `goTo` is always a path to horrible code anyway, but the following macro will postpone keys for the full 10 seconds:
```
postponeKeys while(1) {
 goTo jailbreak
}
jailbreak:
delayUntil 10000
```

kareltucek, 2025-09-04:
The first one should be fixed in #1342. 

The action state was not properly resetted when searching for a matching brace.

--------------------------

```
postponeKeys while(1) {
 goTo jailbreak
}
jailbreak:
delayUntil 10000
```

This is indeed a won't fix scenario.

I am not going to pretend that this is ideal, but given the way the engine works (specifically: not being compiled and not using ASTs), trying to maintain proper scoping with goTo would be expensive (w.r.t. both cpu-time and development-time).

Think of braces rather as of _commands_ that push and pop some kind of context rather than of fixed labels that can be jumped to. If you decide to maintain current context while jumping to some other place... ...well, that's your choice.