UltimateHackingKeyboard/firmware #1331: Smart Macro ifReleased/ifNotActive/ifPlaytime issues on postponed keys

firngrod, 2025-08-22:
With the following macro assigned to key `A`
```
while (1) {
 ifReleased final tapKey a
 ifNotKeyActive $thisKeyId final tapKey b
 ifPlaytime 1500 final tapKey c
}
```
and another key, just to demonstrate
```
postponeKeys delayUntilRelease
```
the following oddities can be observed:

- When `A` is pressed alone and on it's own, it will output `a` or `c` as expected
- When `A` is pressed during postponing, playTime is counted from key press, not from dequeue time.  This may me as intended, but it's unexpected to me.  Otherwise works as expected and outputs `a` or `c` as expected
- When `A` is pressed and released while postponed, `ifReleased` and `ifNotKeyActive $thisKeyId` will never evaluate true and the macro will only ever output `c`

Running UHK60 V1 with firmware 15.1.0

firngrod, 2025-08-22:
In an attempt to find a work-around for this using `delayUntilReleaseMax`, I have discovered that while it seems to work as a way of detecting release in that it exits immediately when the key is released in all cases that I have tested, the time-out appears to be from macro start, not from command start, every time, including the same issue with `ifPlaytime` here, such that the macro below exits after one second, not 3 as one might expect

```
delayUntil 1000
delayUntilReleaseMax 1000
delayUntilReleaseMax 1000
write 'done'
```

I had hoped to use it to detect releases by delaying max 5ms and comparing time before to time after.

firngrod, 2025-09-16:
> > 
> 
> I see. The bug is that the while doesn't yield at the end of the executed block.

Which while?  The following macro will never type out x, even if it has been released while postponed, as I would have expected.
```
ifReleased final tapKey x
tapKey y
```

firngrod, 2025-09-16:
> Sorry, the naming of the commands is confusing, but no. IfReleased by specification operates in the time-machine-modified time, that is intended.

I find that counter-intuitive and, worse, not clearly documented, but it does look true.  The following 
It adds to the confusion that commands like delayUntil will not 'make up for lost time', but I can see how that could be difficult to implement, and not worth the effort when the intention of the macro engine is not to execute for any extended amount of time.

By the way, the macro below will always type `x` if released while postponing, regardless of how long it was held and or what the delay value is.  It works as expected when not released under a postponeKey.
```
delayUntil 1000
ifReleased final tapKey x
tapKey y
```

kareltucek, 2025-09-16:
Indeed, you are completely right it should.

mhantsch, 2025-09-16:
> [@mhantsch](https://github.com/mhantsch) any opinion on this?

Lots of opinions, but I'm not sure if they help ;)

I am happy to give you my thoughts on the topics you seem to have touched here.
I think a lot of the commands are named quite counter-intuitively.

**Concise naming 1:**

Anything named "...Until..." I would expect to be until a certain point in time, e.g. "wait until 15:00 o'clock". On the other hand, to address a time span, I would suspect commands to be named "...For...", e.g. "wait for 300 seconds".

`delayUntilRelease` makes sense, because it waits until a certain point in time - the release event.

`delayUntilReleaseMax` is probably OK, although this could just be an option for `delayUntilRelease`, similar to the `timeoutIn` parameter for `ifGesture`. `delayUntilReleaseMax 300` would become `delayUntilRelease timeoutIn 300`.

`delayUntil` should be renamed to `delayFor`, because it waits for (at least) a certain amount of time. There could be a general `delay timeoutIn 300`, but without `timeoutIn` it would never end, and such a macro command does not make any sense. So `delayFor` it is.

**Concise naming 2:**

Not sure what to do about the different time axis when things get postponed. I must admit I do not fully understand the postponer. I think a macro, or let's call it an activity has a few events that relate to it:

- Time of the event that starts the activity. I guess this is usually the time when a key is pressed, which in turn starts the macro.
- Time when the activity gets postponed.
- Time when the activity is taken off the postponer queue and continues.
- Time when the activity ends.

This is probably not quite correct, but here's my thought process following this model:

We have to differentiate between *real time* (wall clock time) and *run time* (play time of the activity). The difference between *real time* and *run time* is the *queue time*, the interval elapsed between putting the activity on the postponer queue and taking it from the postponer queue. The commands and system variables should make it clear whether they are referring to *real time* or *run time*. Maybe $activity.queuetime could provide the *queue time*, so that *real time* and *run time* can be converted into each other.

Would that help in any way?

firngrod, 2025-09-17:
> We have to differentiate between _real time_ (wall clock time) and _run time_ (play time of the activity). The difference between _real time_ and _run time_ is the _queue time_, the interval elapsed [...]

This is already getting quite complex, even without considering every other command which may be affected by the queue time shift, such as `ifKeyActive $thisKeyId`

I do wonder:  Was there any practical problem being solved when the decision to make the postponed keys play out in the past rather than just have them start when they were popped from the queue, or was it a "because I can" decision in what was maybe at the time a hobby project?  I completely respect if it was "because I can", but it is causing a bunch of confusion.  Would it be a big issue to either just have a command within the current paradigm to sync to wall time, `delayUntilUnpostponed` or `delayUntilNotQueued`, or maybe an alternative to `postponeKeys` which unrolls keys without time shift?  The latter would cause confusion if both postponer commands were in effect at the same time, though.
Alternatively, maybe just not try to do the whole playing out in the past?  I have a suspicion that it might also simplify a bunch of code.

kareltucek, 2025-09-25:
Also there are some curious usecases of nested time machine commands on alphabetic keys in https://forum.ultimatehackingkeyboard.com/t/how-to-create-a-macro-to-press-two-modifiers/2658/7 .