UltimateHackingKeyboard/firmware #1333: v15.0.1 Keyboard crashes

i661, 2025-08-24:
I've seen a couple of these crashes lately on my UHK 60 v2.   Have others encountered this issue?

Printing trace buffer because: Looks like your uhk60 crashed.
EV: 0
Tag: v15.0.1
=== K22P121M120SF7 Reset Cause Debug ===
RCM_SRS0: 0x00
RCM_SRS1: 0x04
RCM_SSRS0: 0x82
RCM_SSRS1: 0x04
PMC_LVDSC1: 0x10
PMC_LVDSC2: 0x00
PMC_REGSC: 0x04
SCB_HFSR: 0x00000000
SCB_CFSR: 0x00000000
SCB_BFAR: 0xE000ED38
SCB_MMFAR: 0xE000ED34

=== Reset Cause Analysis ===
Trace:
>{}{}{}<i1>{}{}{}{}{}<i4>{}<i1>{}{}{}{}{}{}{}{}{}<i1>{}<i4>{}{}{
}{}<i1>{}{}{}{}{}{}{}<i4>{}{}{}<i1>{}<i1>{}<i1>{}{}{}<i1>{}{}{}<
i4>{}{}{}<i1>{}{}{}{}{}{}{}{}<i4>{}<i1>{}{}{}{}{}<i1>{}{}{}{}{}<
i4>{}{}{}{}{}<i1>{}<i1>{}<i1>{}{}{}<i4>{}<i1>{}{}{}{}{}<i1>{}###

mondalaci, 2025-08-26:
@kareltucek Can you look into this?

mondalaci, 2025-08-26:
@i661 It's worth trying the 15.1.0 firmware release in the meantime.

Art3mK, 2025-09-02:
```
Printing trace buffer because: Looks like your uhk60 crashed.
EV: 0
Tag: v15.1.0
=== K22P121M120SF7 Reset Cause Debug ===
RCM_SRS0:  0x00
RCM_SRS1:  0x04
RCM_SSRS0: 0x82
RCM_SSRS1: 0x04
PMC_LVDSC1: 0x10
PMC_LVDSC2: 0x00
PMC_REGSC:  0x04
SCB_HFSR:  0x00000000
SCB_CFSR:  0x00000000
SCB_BFAR:  0xE000ED38
SCB_MMFAR: 0xE000ED34

=== Reset Cause Analysis ===
Trace:
{(l3)}<i1>{(l3)}{(l3)}{(l3)}{(l3)}<i4>{(l3)}{(l3)}{(l3)}{(l3)}{(
l3)}{(l3)}<i1>{(l3)}<i1>{(l3)}<i1>{(l3)}<i1>{(l3)}<i4>{(l3)<i1>}
{(l3)}{(l3)}{(l3)}{(l3)}<i1>{(l3)}{(l3)}{(l3)}{(l3)}{(l3)}{(l3)}
<i4>{(l3)}{(l3)}{(l3)}{(l3)}<i1>{(l3)}<i1>{(l3)}<i1>{(l3)}<i1###
Printing trace buffer because: Looks like your uhk60 crashed.
EV: 0
Tag: v15.1.0
=== K22P121M120SF7 Reset Cause Debug ===
RCM_SRS0:  0x00
RCM_SRS1:  0x04
RCM_SSRS0: 0x82
RCM_SSRS1: 0x04
PMC_LVDSC1: 0x10
PMC_LVDSC2: 0x00
PMC_REGSC:  0x04
SCB_HFSR:  0x00000000
SCB_CFSR:  0x00000000
SCB_BFAR:  0xE000ED38
SCB_MMFAR: 0xE000ED34

=== Reset Cause Analysis ===
Trace:
3)<i4>}{(l3)}{(l3)}<i1>{(l3)}<i1>{(l3)}<i1>{(l3)}<i1>{(l3)}<i1>{
(l3)}{(l3)}{(l3)}<i4>{(l3)}{(l3)}{(l3)}<i1>{(l3)}{(l3)}{(l3)}{(l
3)}{(l3)}{(l3)}{(l3)}{(l3)}<i4>{(l3)}<i1>{(l3)}<i1>{(l3)}<i1>{(l
3)}<i1>{(l3)}<i1>{(l3)}{(l3)}{(l3)}{(l3)}{<i4>(l3)}<i1>{(l3)}###
Printing trace buffer because: Looks like your uhk60 crash
```

kareltucek, 2025-09-02:
Guys can you describe what has happened from user perspective? 

I mean, have this popped up when powering UHK on, or have it crashed during normal operation (e.g., while typing, causing a stuck key or missed keystrokes, or switch to (the default) keymap)?

kareltucek, 2025-09-02:
Following analysis doesn't indicate any serious issue, maybe except power issues: https://claude.ai/share/05fbc4a6-f438-4325-b0af-6ec3909fcd4d 

@mondalaci proposed solution: disable crash reporting entirelly, unless manually enabled. Either:
- Via a preprocessor directive. (Simple, easy to implement)
- Via a runtime variable placed inside a noinit section. (More complicated, but allows turning on crash logs via a macro set statement.)

mondalaci, 2025-09-02:
Please implement your second proposal.

Art3mK, 2025-09-10:
Looks like mine actually had some kind of hardware issue. I’m not sure what exactly - I just disassembled the right half, checked that everything looked fine, put it back together, and now it works 😁 No more crashes! Just an hour ago it was crashing every five minutes and power cycling. Maybe my UHK60 is just getting old - the sticker on the PCB says it was made in 2019.

i661, 2025-09-10:
I did not take mine apart, but oddly, it hasn't crashed in the past couple
of weeks.  I have no explanation.