UltimateHackingKeyboard/firmware #1334: UHK Agent is throwing logs.  Seems to be working but mouse will jump occasionally.

GregoryLand, 2025-08-29:
UHK Agent is throwing logs.  Seems to be working otherwise but mouse will jump occasionally. 

"Loops like the event loop is spinning quite a lot: ### events in the last second." where ### is a random number between [128 - 346] so far.

As well as "ZEPHYR FATAL ERROR 4: Kernel Panic on CPU 0"

[image]

[image]

Agent Version: 8.0.0

Firmware Versions
```
Right: 15.0.1
Left: 15.0.1
Key Cluster: 12.2.0
Trackball: 12.2.0
```

Zephyr Logging:
```
2025-08-29 05:30:34 | UHK 80 right    | Batteries: 100% 100% (4138 4186)
2025-08-29 05:31:20 | UHK 80 right    | Batteries: 100% 100% (4138 4185)
2025-08-29 05:31:35 | UHK 80 right    | Batteries: 100% 100% (4137 4185)
2025-08-29 05:32:21 | UHK 80 right    | Batteries: 100% 100% (4137 4186)
2025-08-29 05:32:35 | UHK 80 right    | Batteries: 100% 100% (4136 4186)
2025-08-29 05:34:22 | UHK 80 right    | Batteries: 100% 100% (4136 4184)
2025-08-29 05:35:37 | UHK 80 right    | Batteries: 100% 100% (4135 4184)
```

mondalaci, 2025-08-30:
@kareltucek Can you look into this?

GregoryLand, 2025-09-02:
Good catch on the firmware revision.  I missed that a new firmware was available.

New Firmware:
```
Right: 15.1.0
Left: 15.1.0
Key Cluster: 12.2.0
Trackball: 12.2.0
```

Here is my device config.  It is pretty mundane.  Nothing really weird.
[UserConfiguration-uhk-80-firmware-15.1.0.json](https://github.com/user-attachments/files/22090631/UserConfiguration-uhk-80-firmware-15.1.0.json)

On the previous firmware I was experiencing spurious mouse events.  Here you can see a very unusual mouse event sent from the UHK-80.  Now 255 in one axis is really suspicious to me (smells like firmware) but i took the time to clean the trackball anyway just in case.
```
 event11  POINTER_MOTION          904  +87.665s	 -1.00/  0.00 ( -1.00/ +0.00)
 event11  POINTER_MOTION          905  +87.670s	 -2.00/  0.00 ( -2.00/ +0.00)
 event11  POINTER_MOTION          906  +87.680s	467.50/  0.00 (+255.00/ +0.00)
 event11  POINTER_MOTION          907  +87.684s	 -3.70/  0.00 ( -2.00/ +0.00)
 event11  POINTER_MOTION          908  +87.690s	 -1.02/  0.00 ( -1.00/ +0.00)
 event11  POINTER_MOTION          909  +87.700s	 -1.00/ -1.00 ( -1.00/ -1.00)
```

I will monitor for now and see if the problem occurs with the new firmware.  

I haven't been able to link the behavior to anything yet.  The keyboard is just being used under heavy programming/gaming workloads and being on 24/7 is the only thing I have figured out so far.  It's connected to 2 machines via Bluetooth (desktop and laptop) but spends 99% of its time connected via USB C to the desktop.

Direct response:
```
    Are you using active wait macros? : No
    Are you using any unusual features, settings? : No but feel free to read the uploaded config.
    Is this happening during : Just normal use after the keyboard has been on for a little while.
```

Next steps:
  I will monitor 15.1.0 to see if the spurious input continues.

Should I close this ticket for now?  If I can reproduce on the new firmware I can reopen it.

kareltucek, 2025-09-02:
Sounds like a race condition *somewhere*. Can happen when a value is being copied from one variable into another by two single bytes instructions, when one of them is positive and the other negative.  

Let's rename the ticket and continue here. 

I am afraid I am quite busy these days, and won't be available until the second half of september though :-(.

GregoryLand, 2025-09-04:
Done. Logs if you wish to verify expected behavior.

[good-flash-of-15.1.0-jumphunt.txt](https://github.com/user-attachments/files/22138616/good-flash-of-15.1.0-jumphunt.txt)
[UserConfiguration-uhk-80-firmware-jumphunt.json](https://github.com/user-attachments/files/22138610/UserConfiguration-uhk-80-firmware-jumphunt.json)

Entering wait loop.

GregoryLand, 2025-09-04:
Cleaner screenshot.

[image]

```
Probable jump detected at site UhkModuleDriver1! 206, -28
os: ***** MPU FAULT *****
os:   Data Access Violation
os:   MMFAR Address: 0x20035670
os: r0/a1:  0x20002474  r1/a2:  0x200356b0  r2/a3:  0x20035738
os: r3/a4:  0x00082f4d r12/ip:  0x0000000a r14/lr:  0x00073aff
os:  xpsr:  0x61000000
os: r4/v1:  0x000937f7  r5/v2:  0x20035738  r6/v3:  0x00000001
os: r7/v4:  0x20002474  r8/v5:  0x000a3baf  r9/v6:  0x00000001
os: r10/v7: 0x000a3bb5  r11/v8: 0x00000000    psp:  0x20035680
os: EXC_RETURN: 0xfffffffd
os: Faulting instruction address (r15/pc): 0x00072b64
os: >>> ZEPHYR FATAL ERROR 19: Unknown error on CPU 0
os: Current thread: 0x2000fa70 (i2c_poller)
===== PANIC =====
Printing trace buffer because: crash/panic
EV: 6291968
Tag: jumphunt_v1.2
Trace:
u8)(<i1>l6u1u2u3u6u7u8)(l6u1u2u3u6u7u8)<i1><i1><i1><i1><i1>(l6u1
u2v5u3u6u7u8)(l6u1u2u3u6<i1>u7u8)(l6u1u2u3u6u7u8)<i1><i1>(l6u1u2
v5u3u6u7u8)(l6u1u2u3u6<i1>u7u8)(l6u1u2u3u6u7u8)<i1><i1>(l6u1u2v5
u3u6u7u8)(l6u1u2u3u6<i1>u7u8)(l6u1u2u3u6u7u8)<i1><i1><i1><i1><i1
fatal_error: Resetting system
Probable jump detected at site MouseController1! -237, -1299
Probable jump detected at site MouseController2! -237, -1299
```

kareltucek, 2025-09-04:
Those crashes are a new thing, meaning that I have "just" messed up something in the debugging code, right?

Those values indeed look more like false positives. 

I won't be able to look into it for another week or so now, so  please feel free to flash to whatever firmware worked best for you for now.

kareltucek, 2025-09-04:
Actually should suffice to disable the dev mode in order to get non-crashing experience again 🤔.

GregoryLand, 2025-09-04:
I am not sure to be honest.  I was never able to recreate it before in the old firmware. I think I will try flashing vanilla 15.1.0 and 15.0.1 and try this triggering method see what happens.

I will disable dev mode and see if i can still trigger a data access violation first. 

Gives me some things to try while you are busy.

Here is a libinput and its associated crash.  

```
-event8   POINTER_MOTION               +1.740s	  0.00/ -0.30 ( +0.00/ -1.00)
 event8   POINTER_MOTION            2  +1.759s	  0.00/ -0.57 ( +0.00/ -1.00)
 event8   POINTER_MOTION            3  +1.769s	  2.93/ -2.93 ( +3.00/ -3.00)
 event8   POINTER_MOTION            4  +1.774s	 17.81/ -3.56 (+10.00/ -2.00)
 event8   POINTER_MOTION            5  +1.779s	 18.00/ -4.00 ( +9.00/ -2.00)
 event8   POINTER_MOTION            6  +1.789s	  6.78/  0.00 ( +4.00/ +0.00)
 event8   POINTER_MOTION            7  +1.793s	 11.67/ -1.67 ( +7.00/ -1.00)
 event8   POINTER_MOTION            8  +2.541s	  2.83/  5.67 ( +2.00/ +4.00)
 event8   POINTER_MOTION            9  +2.550s	  0.89/  1.79 ( +1.00/ +2.00)
 event8   POINTER_MOTION           10  +2.556s	-1464.83/-1789.33 (-799.00/-976.00)
 event8   POINTER_MOTION           11  +2.637s	1014.00/-2414.00 (+507.00/-1207.00)
 event8   POINTER_MOTION           12  +2.646s	-1682.00/-1406.00 (-841.00/-703.00)
 event8   POINTER_MOTION           13  +2.651s	970.00/436.00 (+485.00/+218.00)
 event8   POINTER_MOTION           14  +2.660s	636.00/-2064.00 (+318.00/-1032.00)
 event8   POINTER_MOTION           15  +2.665s	2544.00/-460.00 (+1272.00/-230.00)
 event8   POINTER_MOTION           16  +2.674s	716.00/-716.00 (+358.00/-358.00)
 event8   POINTER_MOTION           17  +2.680s	3030.00/-2916.00 (+1515.00/-1458.00)
 event8   POINTER_MOTION           18  +2.689s	  0.00/-392.00 ( +0.00/-196.00)
 event8   POINTER_MOTION           19  +2.695s	1060.00/3460.00 (+530.00/+1730.00)
 event8   POINTER_MOTION           20  +2.700s	394.00/598.00 (+197.00/+299.00)
-event2   DEVICE_REMOVED               Ultimate Gadget Laboratories UHK 80 Right USB seat0 default group1  cap:k
-event8   DEVICE_REMOVED               Ultimate Gadget Laboratories UHK 80 Right USB seat0 default group1  cap:p
-event9   DEVICE_REMOVED               Ultimate Gadget Laboratories UHK 80 Right USB seat0 default group1  cap:kp
-event2   DEVICE_ADDED                 Ultimate Gadget Laboratories UHK 80 Right USB seat0 default group4  cap:k
-event9   DEVICE_ADDED                 Ultimate Gadget Laboratories UHK 80 Right USB seat0 default group4  cap:kp scroll-nat calib
-event8   DEVICE_ADDED                 Ultimate Gadget Laboratories UHK 80 Right USB seat0 default group4  cap:p left scroll-nat scroll-button
 event8   POINTER_MOTION               +8.988s	1024.22/-56.84 (+973.00/-54.00)
 event8   POINTER_MOTION            2  +9.006s	  2.42/  0.00 ( +2.00/ +0.00)
 event8   POINTER_MOTION            3  +9.010s	 47.67/  3.67 (+26.00/ +2.00)
 event8   POINTER_MOTION            4  +9.020s	 44.00/ 12.00 (+22.00/ +6.00)
 event8   POINTER_MOTION            5  +9.025s	 76.00/ 14.00 (+38.00/ +7.00)
 event8   POINTER_MOTION            6  +9.034s	 34.00/  4.00 (+17.00/ +2.00)
 event8   POINTER_MOTION            7  +9.038s	 40.00/  4.00 (+20.00/ +2.00)
 event8   POINTER_MOTION            8  +9.047s	 19.68/  3.94 (+10.00/ +2.00)
 event8   POINTER_MOTION            9  +9.052s	 11.27/  3.76 ( +6.00/ +2.00)
 event8   POINTER_MOTION           10  +9.062s	 11.48/  0.00 ( +7.00/ +0.00)
 event8   POINTER_MOTION           11  +9.066s	 17.00/  0.00 ( +9.00/ +0.00)
 event8   POINTER_MOTION           12  +9.076s	  7.33/  0.00 ( +4.00/ +0.00)
 event8   POINTER_MOTION           13  +9.081s	 10.77/  1.54 ( +7.00/ +1.00)
 event8   POINTER_MOTION           14  +9.101s	  1.38/  0.00 ( +1.00/ +0.00)
 event8   POINTER_MOTION           15  +9.115s	  0.90/  0.00 ( +1.00/ +0.00)
 event8   POINTER_MOTION           16  +9.142s	 -3.00/  2.00 ( -3.00/ +2.00)
 event8   POINTER_MOTION           17  +9.151s	 -5.14/  0.00 ( -5.00/ +0.00)
 event8   POINTER_MOTION           18  +9.156s	-20.48/ -1.86 (-11.00/ -1.00)
 event8   POINTER_MOTION           19  +9.368s	  0.00/  1.58 ( +0.00/ +1.00)
 event8   POINTER_MOTION           20  +9.388s	  0.00/  0.57 ( +0.00/ +1.00)
 event8   POINTER_MOTION           21  +9.903s	 -0.56/  0.00 ( -1.00/ +0.00)
```

```
20035738  r6/v3:  0x00000001
os: r7/v4:  0x20002474  r8/v5:  0x000a3baf  r9/v6:  0x00000001
os: r10/v7: 0x000a3bb5  r11/v8: 0x00000000    psp:  0x20035680
os: EXC_RETURN: 0xfffffffd
os: Faulting instruction address (r15/pc): 0x00072b64
os: >>> ZEPHYR FATAL ERROR 19: Unknown error on CPU 0
os: Current thread: 0x2000fa70 (i2c_poller)
===== PANIC =====
Printing trace buffer because: crash/panic
EV: 6291968
Tag: jumphunt_v1.2
[... 81 lines cut ...]
16:03:36.316 › [DeviceOperation] status buffer segment: ****
16:03:36.324 › [DeviceOperation] status buffer segment: 70
16:03:36.332 › [DeviceOperation] status buffer segment: 738
16:03:36.340 › [DeviceOperation] status buffer segment: 3aff
16:03:38.028 › [DeviceOperation] Read "leftHalf" version information
16:03:38.036 › [DeviceOperation] Read "leftHalf" repo information
16:03:38.052 › [DeviceOperation] Read "leftHalf" firmware checksum
16:03:38.060 › [DeviceOperation] Read "leftModule" version information
16:03:38.068 › [DeviceOperation] Read "leftModule" repo information
16:03:38.084 › [DeviceOperation] Read "leftModule" firmware checksum
16:03:38.092 › [DeviceOperation] Read "rightModule" version information
[... cut ...]
16:03:38.204 › [DeviceOperation] status buffer segment: 0x2000fa70 (i2c_poller)
16:03:38.212 › [DeviceOperation] status buffer segment: r because: crash/panic
16:03:38.220 › [DeviceOperation] status buffer segment: )(l6u1u2u3u6<i1>u7u8)(l6u1u2u3u6u7u8)<i1><i1><i1><i1><i1>(l6u1
16:03:38.228 › [DeviceOperation] status buffer segment: u
16:03:38.236 › [DeviceOperation] status buffer segment: 1u2v
16:03:38.244 › [DeviceOperation] status buffer segment: 6u1u2v5
16:03:38.252 › [DeviceOperation] status buffer segment: i1><i1><i1

kareltucek, 2025-09-04:
> That said calling a logging function and it faulting with a data access violation is still really sus one way or another.

Not really. I had called those jump checks and logging from within an irq-disabled region, which by itself is sufficient to create this kind of issues on any peripheral-related or thread-sync-related operations.

> Interestingly enough flashing to 15.0.1 and flicking the mouse at stupidly high speeds (like really hammer it) will cause the event loop spinning logs.

That's fine. That problem has been fixed since - namely that log should not have been triggered by mouse-related functionality.

GregoryLand, 2025-09-14:
Video displaying the issue occurring.

https://youtu.be/9wgp_tOckos

GregoryLand, 2025-09-14:
```
So today I started getting mouse jumps very few seconds.  After digging around to figure out what has changed I realized I turned my laptop on.  The desktop that has the keyboard attached is getting flooded with messages.

```
[CHG] Device D6:58:58:C0:BC:81 Connected: yes
[SIGNAL] org.bluez.Device1.Disconnected org.bluez.Reason.Authentication Connection terminated due to authentication failure
[CHG] Device D6:58:58:C0:BC:81 Connected: no
[CHG] Device D6:58:58:C0:BC:81 Connected: yes
[SIGNAL] org.bluez.Device1.Disconnected org.bluez.Reason.Authentication Connection terminated due to authentication failure
[CHG] Device D6:58:58:C0:BC:81 Connected: no
[CHG] Device D6:58:58:C0:BC:81 Connected: yes
[SIGNAL] org.bluez.Device1.Disconnected org.bluez.Reason.Authentication Connection terminated due to authentication failure
[CHG] Device D6:58:58:C0:BC:81 Connected: no
[CHG] Device D6:58:58:C0:BC:81 Connected: yes
[SIGNAL] org.bluez.Device1.Disconnected org.bluez.Reason.Authentication Connection terminated due to authentication failure
[CHG] Device D6:58:58:C0:BC:81 Connected: no
[CHG] Device D6:58:58:C0:BC:81 Connected: yes
[SIGNAL] org.bluez.Device1.Disconnected org.bluez.Reason.Authentication Connection terminated due to authentication failure
[CHG] Device D6:58:58:C0:BC:81 Connected: no
[CHG] Device D6:58:58:C0:BC:81 Connected: yes
[SIGNAL] org.bluez.Device1.Disconnected org.bluez.Reason.Authentication Connection terminated due to authentication failure
[CHG] Device D6:58:58:C0:BC:81 Connected: no
[CHG] Device D6:58:58:C0:BC:81 Connected: yes
[SIGNAL] org.bluez.Device1.Disconnected org.bluez.Reason.Authentication Connection terminated due to authentication failure
[CHG] Device D6:58:58:C0:BC:81 Connected: no
[CHG] Device D6:58:58:C0:BC:81 Connected: yes
[SIGNAL] org.bluez.Device1.Disconnected org.bluez.Reason.Authentication Connection terminated due to authentication failure
[CHG] Device D6:58:58:C0:BC:81 Connected: no
[CHG] Device D6:58:58:C0:BC:81 Connected: yes
[SIGNAL] org.bluez.Device1.Disconnected org.bluez.Reason.Authentication Connection terminated due to authentication failure
[CHG] Device D6:58:58:C0:BC:81 Connected: no
[CHG] Device D6:58:58:C0:BC:81 Connected: yes
[SIGNAL] org.bluez.Device1.Disconnected org.bluez.Reason.Authentication Connection terminated due to authentication failure
[CHG] Device D6:58:58:C0:BC:81 Connected: no
[CHG] Device D6:58:58:C0:BC:81 Connected: yes
[SIGNAL] org.bluez.Device1.Disconnected org.bluez.Reason.Authentication Connection terminated due to authentication failure
[CHG] Device D6:58:58:C0:BC:81 Connected: no
[CHG] Device D6:58:58:C0:BC:81 Connected: yes
[SIGNAL] org.bluez.Device1.Disconnected org.bluez.Reason.Authentication Connection terminated due to authentication failure
[CHG] Device D6:58:58:C0:BC:81 Connected: no
[CHG] Device D6:58:58:C0:BC:81 Connected: yes
[SIGNAL] org.bluez.Device1.Disconnected org.bluez.Reason.Authentication Connection terminated due to authentication failure
[CHG] Device D6:58:58:C0:BC:81 Connected: no
[CHG] Device D6:58:58:C0:BC:81 Connected: yes
[SIGNAL] org.bluez.Device1.Disconnected org.bluez.Reason.Authentication Connection terminated due to authentication failure
[CHG] Device D6:58:58:C0:BC:81 Connected: no
```

GregoryLand, 2025-10-20:
Yea but what I don't understand is why the UHK is trying to connect at all?  It was paired with both the laptop and desktop but connected through the physical cable to the desktop and between the keyboards.  The active connection mode was set to the cable.  In this mode I would expect it to ignore connection attempts or maybe even have the bluetooth disabled entirely?

kareltucek, 2025-10-20:
Well, it is advertising in case someone wants to connect to it. Allowing multiple connected devices allows quick switching between multiple hosts.

But I guess adding a single host mode in which any host connection would have to be switched to via a corresponding switchHost action would make sense. 

@mondalaci your take on this?

mhantsch, 2025-10-21:
> * Many devices allow one connection only. I actually don't know about a bluetooth keyboard that would allow multiple connections at once.

My two ZMK keyboards connect to up to 5 BT devices simultaneously. They only send key reports to one of them at a time (user can select which is the *active* profile), but they keep all of them connected.