UltimateHackingKeyboard/firmware #1339: c2usb assert fail, and crash loop

kareltucek, 2025-09-04:
@benedekkupper I am getting this on master.

Right side spins on this loop:
```
----------
UHK 80 Right USB started
[00:00:00.054,565] <inf> fs_nvs: 3 Sectors of 4096 bytes
[00:00:00.054,565] <inf> fs_nvs: alloc wra: 1, c60
[00:00:00.054,595] <inf> fs_nvs: data wra: 1, 5f4
[00:00:00.055,847] <inf> bt_sdc_hci_driver: SoftDevice Controller build revision:
                                            fe 2c f9 6a 7f 36 22 2e  a0 79 c0 40 be 2c 03 20 |.,.j.6". .y.@.,.
                                            40 c2 f3 32                                      |@..2
[00:00:00.060,821] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.060,852] <inf> bt_hci_core: HW Variant: nRF52x (0x0002)
[00:00:00.060,882] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 254.63788 Build 573996906
[00:00:00.061,950] <inf> bt_hci_core: No ID address. App must call settings_load()

[00:00:00.099,761] <inf> bt_hci_core: Identity: F4:5E:F7:7F:E6:B2 (random)
[00:00:00.099,792] <inf> bt_hci_core: HCI: version 6.0 (0x0e) revision 0x104e, manufacturer 0x0059
[00:00:00.099,822] <inf> bt_hci_core: LMP: version 6.0 (0x0e) subver 0x104e
Reading hardware config
Reading user config
Applying user config
Flashed User Config version: 12.0.0 (native version: 12.0.0., at 56d35a30 / 00000000000000000000000000000000)
Setting battery mode to 0
Clearing bonds
 - Keeping a known bond.
Resetting left right link! Unidirectional
User config applied
abort()
[00:00:00.222,167] <err> os: r0/a1:  0x00000004  r1/a2:  0x00000000  r2/a3:  0x00000021
[00:00:00.222,167] <err> os: r3/a4:  0x00000004 r12/ip:  0x00082e83 r14/lr:  0x00037111
[00:00:00.222,198] <err> os:  xpsr:  0x21000000
[00:00:00.222,198] <err> os: r4/v1:  0x000a31d7  r5/v2:  0x000a2d25  r6/v3:  0x00000000
[00:00:00.222,229] <err> os: r7/v4:  0x00000000  r8/v5:  0x00000000  r9/v6:  0x00000000
[00:00:00.222,259] <err> os: r10/v7: 0x00000000  r11/v8: 0x00000000    psp:  0x200372d8
[00:00:00.222,259] <err> os: EXC_RETURN: 0x0
[00:00:00.222,259] <err> os: Faulting instruction address (r15/pc): 0x00037120
[00:00:00.222,320] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[00:00:00.222,351] <err> os: Current thread: 0x20010e78 (c2usb)
===== PANIC =====
Printing trace buffer because: crash/panic
EV: 67174981
Tag: 56d35a30
Trace:
><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1
><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1
><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1><i1
*** Booting nRF Connect SDK v2.8.0-7b0d38c3be66 ***
><i1><i1><i1><i1><i1><i1><i1>bp9<i1><i1><i1><i1><i1><i1><i1><i1>
[00:00:02.196,685] <err> fatal_error: Resetting system
uhk80:right$�
ssertion "power_state() == power::state::L3_OFF" failed: file "WEST_TOPDIR/c2usb/c2usb/port/zephyr/udc_mac.cpp", line 508, functio
n: int usb::zephyr::udc_mac::process_event(const udc_event&)
----------
UHK 80 Right USB started
```

Above log instruction points to:
```
$ ./build.sh uhk-80-right addrline 0x00037120
addrline for 0x00037120:
    /opt/west4/zephyr/lib/libc/common/source/stdlib/abort.c:14
```

Left side steadily fails on the same assert: 
```
assertion "power_state() == power::state::L3_OFF" failed: file "WEST_TOPDIR/c2usb/c2usb/port/zephyr/udc_mac.cpp", line 508, functi
on: int usb::zephyr::udc_mac::process_event(const udc_event&)
assertion "power_state() == power::state::L3_OFF" failed: file "WEST_TOPDIR/c2usb/c2usb/port/zephyr/udc_mac.cpp", line 508, functi
on: int usb::zephyr::udc_mac::process_event(const udc_event&)
assertion "power_state() == power::state::L3_OFF" failed: file "WEST_TOPDIR/c2usb/c2usb/port/zephyr/udc_mac.cpp", line 508, functi
on: int usb::zephyr::udc_mac::process_event(const udc_event&)
assertion "power_state() == power::state::L3_OFF" failed: file "WEST_TOPDIR/c2usb/c2usb/port/zephyr/udc_mac.cpp", line 508, functi
on: int usb::zephyr::udc_mac::process_event(const udc_event&)
assertion "power_state() == power::state::L3_OFF" failed: file "WEST_TOPDIR/c2usb/c2usb/port/zephyr/udc_mac.cpp", line 508, functi
on: int usb::zephyr::udc_mac::process_event(const udc_event&)
```

I am going away for another week tomorrow, so this isn't a high priority from my side.

benedekkupper, 2025-09-04:
Can you check the build from #1328? There is a fix for that there.

closed 2025-09-04 by kareltucek

benedekkupper, 2025-09-04:
I merged that PR so master is usable again.